MichaIng/DietPi #1675: Lost support of wifi USB adapter

Fray2000, 2018-04-01:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version 6.5
- SBC Device: RPi 2 B+
- Power supply used 5v 3A
- SD card used 8GB from Microdrive
- Distro Dietpi 6.5

#### Additional Information (if applicable):
The key is detected by ''lsusb'' command 

#### Output :

#### Bus 001 Device 042: ID 07d1:3c03 D-Link System AirPlus G DWL-G122 Wireless Adapter(rev.C1) [Ralink RT2571W]
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

#### Expected behaviour:
To be able to scan and detect any available wifi's.

#### Actual behaviour:
Is not able to scan for any WIFI

Fourdee, 2018-04-01:
@Fray2000 

Thanks for the report, hopefully its just the firmware that required. This is no longer installed by default for ralink, as ```firmware-misc-nonfree``` contains mostly intel/nvidia firmware.

Please try installing the package, reboot system then retry:
```
G_AGI firmware-misc-nonfree
reboot
```

--

NB: i'll make ```firmware-misc-nonfree``` default install for future images.

SuBLiNeR, 2018-04-01:
i guess it should be the package called firmware-ralink ... isn't it?

Fray2000, 2018-04-01:
i m installing it right now, wait a sec

Fray2000, 2018-04-01:
it does not work

2018-04-01 20:29 GMT+02:00 Francesco Leo <francescomaria.leo@gmail.com>:

SuBLiNeR, 2018-04-01:
@Fray2000 in console try to install via this:
`apt-get install firmware-misc-nonfree`

Fourdee, 2018-04-01:
@SuBLiNeR 

> this was for Jessie

Yep, Stretch:
```firmware-ralink``` is a virtual package for ```firmware-misc-nonfree```
https://packages.debian.org/stretch/firmware-ralink

Fray2000, 2018-04-01:
Perfect, it works thanks ! and is it possible to insert a command in the dietpi.txt file after burning the image  to install this package on first boot?

Fourdee, 2018-04-01:
@Fray2000 

We'll update the image to v6.6 and include the package by default.

Fray2000, 2018-04-01:
@Fourdee Thanks

Fourdee, 2018-04-01:
I'll mark this as closed.

New ticket for image update: https://github.com/Fourdee/DietPi/issues/1676

closed 2018-04-01 by Fourdee