MichaIng/DietPi #1678: Error running dietpi-update

yonesmit, 2018-04-02:
#### Required Information:
- DietPi Version | v6.5
- SBC Device     | RPi 2 Model B (armv7l) (index=2)
- Distro         | stretch (index=4)
- Command        | G_AGI: firmware-misc-nonfree
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Patchfile

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-misc-nonfree_20161130-3+rpt3_all.deb  Operation timed out after 120000 milliseconds with 0 out of 0 bytes received
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Fourdee, 2018-04-02:
@yonesmit 

Thanks for the report 👍 

Appears to be a issue with the RPi repo server during download (timeout). Please retry ```dietpi-update```.

URL is working here: https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-misc-nonfree_20161130-3+rpt3_all.deb

yonesmit, 2018-04-02:
I tried to run dietpi-update again. And the result is:

Fourdee, 2018-04-03:
@yonesmit 

Set the version to 6.5, then redo the update:
```
echo -e "6\n5\n" > /DietPi/dietpi/.version
dietpi-update
```

yonesmit, 2018-04-03:
I set the version to 6.5 and ran again the update procedure. The process went correctly without errors.
Thanks for your support.
Best Regards,

MichaIng, 2018-04-03:
@yonesmit 
Great, will mark as closed. Just for the record: If something went wrong during dietpi-update process but it still runs through (adjusting version string), we just implemented an option to rerun the current patch, so no need to manually adjust `.version` anymore 😉: `dietpi-update -1`

closed 2018-04-03 by MichaIng

Fourdee, 2018-04-03:
@MichaIng 

> dietpi-update -1

Forgot about this, nice one 👍