MichaIng/DietPi #1680: Unable to generate a working swap

mgherghi, 2018-04-03:
## Creating a bug report/issue:
I cant seem to create a swap , i have tried manually creating one using fallocate and then mkswap and swapon, but when i enter the swapon /var/swap command it gives me an error saying no such file. I have already made the swap and used chmod 600 to make it r/w . The swapfile is there , but the system will not allow me to use it as a swap..  What am I doing wrong ? 

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- RPI 3b+
-3.5 amp usb (Dell branded)
- SanDisk Extreme Plus
- Stretch

#### Expected behaviour:
trying to create a swap , but it keeps on failing.  

#### Actual behaviour:
DietPi-Swapfile: swapon /var/swap
 - exit_code = 255
 - VERSION:v6.6 | HW_MODEL:3 | HW_ARCH:2 | DISTRO:4

#### Extra details:
Clean install, nothing has been installed yet.

mgherghi, 2018-04-03:
Nvm , i figured out it was because i formatted mmcblk0p2 to f2fs and that was causing the issue

MichaIng, 2018-04-03:
@mgherghi 
Just a quick search: https://wiki.archlinux.org/index.php/swap
```
# fallocate -l 512M /swapfile
Note: fallocate may cause problems with some file systems such as F2FS or XFS.[1] As an alternative, using dd is more reliable, but slower:
# dd if=/dev/zero of=/swapfile bs=1M count=512
```
We use fallocate within our script: https://github.com/Fourdee/DietPi/blob/master/dietpi/func/dietpi-set_dphys-swapfile#L114

@Fourdee 
Do we want to provide wider filesystem support by adding fs check and use `dd` in case, as suggested within link above?

Fourdee, 2018-04-03:
@MichaIng 

> Do we want to provide wider filesystem support by adding fs check and use dd in case, as suggested within link above?

Yep 👍 

We can use following to detect filesystem type for the target swap location:
```
root@DietPi:~# stat --file-system --format=%T /tmp
tmpfs
root@DietPi:~# stat --file-system --format=%T /var
ext2/ext3
```

Fourdee, 2018-04-03:
https://github.com/Fourdee/DietPi/commit/1c4b7d94fa5b7d95f331ca875ba237b6274b5aed should do it. Once i've setup a test system i'll verify.

Fourdee, 2018-04-03:
🈯️ 
```
[ INFO ] DietPi-Swapfile | Setting /tmp tmpfs size: 1990M
[  OK  ] DietPi-Swapfile | mount -o remount,size=1990M tmpfs /tmp
[  OK  ] DietPi-Swapfile | swapoff -a
[ INFO ] DietPi-Swapfile | Deleting existing swapfile (/mnt/63864066-5558-408d-be95-1af911c94bb9/.swapfile)
[ SUB1 ] DietPi-Swapfile > Generating new swapfile
[ INFO ] DietPi-Swapfile | Size     = 99 MB
[ INFO ] DietPi-Swapfile | Location = /mnt/63864066-5558-408d-be95-1af911c94bb9/.swapfile
[  OK  ] DietPi-Swapfile | dd if=/dev/zero of=/mnt/63864066-5558-408d-be95-1af911c94bb9/.swapfile bs=4K count=25344
         DietPi-Swapfile | mkswap /mnt/63864066-5558-408d-be95-1af911c94bb9/.swa[  OK  ] DietPi-Swapfile | mkswap /mnt/63864066-5558-408d-be95-1af911c94bb9/.swapfile
         DietPi-Swapfile | swapon /mnt/63864066-5558-408d-be95-1af911c94bb9/.swa[  OK  ] DietPi-Swapfile | swapon /mnt/63864066-5558-408d-be95-1af911c94bb9/.swapfile
[ INFO ] DietPi-Swapfile | Setting /tmp tmpfs size: 1990M
[  OK  ] DietPi-Swapfile | mount -o remount,size=1990M tmpfs /tmp
```

Fourdee, 2018-04-03:
Resolved for v6.7.

@MichaIng Great job on finding the cause 👍

closed 2018-04-03 by Fourdee

d4f2, 2018-04-16:
I have the same issue with an **ntfs** filessystem

DietPi-Swapfile: fallocate -l 1072M /mnt/2C..CC/dietpi_userdate/.swapfile
Exit code: 1
DietPi version: v6.7 | HW_MODEL:3 | HW_ARCH:2 | DISTRO: 4
fallocate: fallocate failed: Operations not supported

`root@DietPi:/# findmnt -n -o FSTYPE -T /mnt/2C..CC/`
`fuseblk`
`root@DietPi:/#`

un-commenting the if clause for fallocate and just use **dd** in **dietpi-set_dphys-swapfile**
appears to work fine

`#Pre-allocate for filesystems which do no support quick-allocate with fallocate`
`#if [ "$fs_type" = 'f2fs' ] || [ "$fs_type" = 'xfs' ]; then`
`G_RUN_CMD dd if=/dev/zero of=$SWAP_LOCATION bs=4K count=$(( $SWAP_SIZE * 1024 / 4 ))`
`#else`
`#G_RUN_CMD fallocate -l $SWAP_SIZE'M' $SWAP_LOCATION`
`#fi`