MichaIng/DietPi #1681: OpenJDK 8 error

symbios24, 2018-04-03:
Hello

Upon installation of Nukkit i got an OpenJDK 8 file not found error

can you replicate the install to verify the problem?

Thanks

Fourdee, 2018-04-03:
@symbios24 

Which device is this running on?

SuBLiNeR, 2018-04-03:
JRE or JDK version? For myself i tried this, as this is always working for me if i had to setup the OpenJDK JRE - https://tecadmin.net/install-java-8-on-debian/

symbios24, 2018-04-04:
Its running in a Raspberry pi 3.

symbios24, 2018-04-04:
Its running in a Raspberry pi 3 not 3b+ on latest DietPi v6.6. Its OpenJDK 8 + JRE Library i uploaded the picture with the error in the link under:
https://ibb.co/g69ASH

Fourdee, 2018-04-04:
@symbios24 

Thanks 👍 , same issue as https://github.com/Fourdee/DietPi/issues/1682. I'll mark this as closed in favor of the other ticket.

closed 2018-04-04 by Fourdee