MichaIng/DietPi #1682: Nukkit not working on Raspberry pi 3

royspeed, 2018-04-04:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device RPi 3
- Power supply used 5v 5A 
- SD card used Sandisk ultra32 Gb & 2Gb
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title Nukkit
- Can this issue be replicated on a fresh installation of DietPi, Yes

#### Expected behaviour:
 Error on install

#### Actual behaviour:

DietPi Error Handler: ├─────────────────────────────┐
          │                                                                                   │ 
          │ DietPi-Software: G_AGI: openjdk-8-jdk                                               
          │  - exit_code = 100                                                                  
          │  - VERSION:v6.6 | HW_MODEL:3 | HW_ARCH:2 | DISTRO:4                                 
          │                                                                                     
          │ Log file contents:                                                                  
          │ Setting up libgtk-3-0:armhf (3.22.11-1+rpi3) ...^M                                  
          │ Setting up libatk-wrapper-java-jni:armhf (0.33.3-13+deb9u1) ...^M                   
          │ Setting up ca-certificates-java (20170531+nmu1) ...^M                               
          │ Error: missing `server' JVM at                                                      
          │ `/usr/lib/jvm/java-8-openjdk-armhf/jre/lib/arm/server/libjvm.so'.^M                 
          │ Please install or use the JRE or JDK that contains these missing components.^M      
          │ dpkg: error processing package ca-certificates-java (--configure):^M                
          │  subprocess installed post-installation script returned error exit status 4^M       
          │ dpkg: dependency problems prevent configuration of openjdk-8-jre-headless:armhf:^M  
          │  openjdk-8-jre-headless:armhf depends on ca-certificates-java; however:^M           
          │   Package ca-certificates-java is not configured yet.^M      ^M                                                                                  
          │ dpkg: error processing package openjdk-8-jre-headless:armhf (--configure):^M        
          │  dependency problems - leaving unconfigured^M                                       
          │ dpkg: dependency problems prevent configuration of openjdk-8-jdk-headless:armhf:^M  
          │  openjdk-8-jdk-headless:armhf depends on openjdk-8-jre-headless (=                  
          │ 8u162-b12-1~deb9u1); however:^M                                                     
          │   Package openjdk-8-jre-headless:armhf is not configured yet.^M                     
          │ ^M                                                                                  
          │ dpkg: error processing package openjdk-8-jdk-headless:armhf (--configure):^M        
          │  dependency problems - leaving unconfigured^M                                       
          │ dpkg: dependency problems prevent configuration of openjdk-8-jdk:armhf:^M           
          │  openjdk-8-jdk:armhf depends on openjdk-8-jdk-headless (= 8u162-b12-1~deb9u1);      
          │ however:^M                                                                          
          │   Package openjdk-8-jdk-headless:armhf is not configured yet.^M                     
          │ ^M                                                                                  
          │ dpkg: error processing package openjdk-8-jdk:armhf (--configure):^M  dependency problems - leaving unconfigured^M                                       
          │ dpkg: dependency problems prevent configuration of openjdk-8-jre:armhf:^M           
          │  openjdk-8-jre:armhf depends on openjdk-8-jre-headless (= 8u162-b12-1~deb9u1);      
          │ however:^M                                                                          
          │   Package openjdk-8-jre-headless:armhf is not configured yet.^M                     
          │ ^M                                                                                  
          │ dpkg: error processing package openjdk-8-jre:armhf (--configure):^M                 
          │  dependency problems - leaving unconfigured^M                                       
          │ Processing triggers for libc-bin (2.24-11+deb9u3) ...^M                             
          │ Processing triggers for systemd (232-25+deb9u2) ...^M                               
          │ Processing triggers for libgdk-pixbuf2.0-0:armhf (2.36.5-2+deb9u2) ...^M            
          │ Processing triggers for ca-certificates (20161130+nmu1) ...^M                       
          │ Updating certificates in /etc/ssl/certs...^M                                        
          │ 0 added, 0 removed; done.^M                                                         
          │ Running hooks in /etc/ca-certificates/update.d...^M                                 
          │ ^M                                                           Error: missing `server' JVM at                                                      
          │ `/usr/lib/jvm/java-8-openjdk-armhf/jre/lib/arm/server/libjvm.so'.^M                 
          │ Please install or use the JRE or JDK that contains these missing components.^M      
          │ E: /etc/ca-certificates/update.d/jks-keystore exited with code 1.^M                 
          │ done.^M                                                                             
          │ Errors were encountered while processing:^M                                         
          │  ca-certificates-java^M                                                             
          │  openjdk-8-jre-headless:armhf^M                                                     
          │  openjdk-8-jdk-headless:armhf^M                                                     
          │  openjdk-8-jdk:armhf^M                                                              
          │  openjdk-8-jre:armhf^M                                                              
          │ E: Sub-process /usr/bin/dpkg returned an error code (1)                             
          │                                                                                     
          │ If problems persist, please report this to DietPi for investigation, including a    
          │ screenshot of this error! (https://github.com/Fourdee/DietPi/issues).

#### Steps to reproduce:
Just install via dietpi-software

#### Did you submit a ```dietpi-bugreport```?
 241b6064-2af7-4102-a050-b235d3ca5b7b-0

#### Extra details:

Fourdee, 2018-04-04:
@royspeed 

Thanks for the report 👍 

We'll investigate.

Fourdee, 2018-04-04:
notes:
```
root@DietPi:~# find / -name libjvm.so
/usr/lib/jvm/java-8-openjdk-armhf/jre/lib/arm/client/libjvm.so
```
- This is an external bug, straight forward installation of ```openjdk-8-jdk``` fails. Seems to have happened in the past: https://www.raspberrypi.org/forums/viewtopic.php?t=197824

Fourdee, 2018-04-04:
## Workaround:
#### Run the following commands before installing Nukkit, or, OpenJDK:
This will download the required file, which is broken in the ```openjdk-8-jdk``` APT package
```
mkdir -p /usr/lib/jvm/java-8-openjdk-armhf/jre/lib/arm/server
wget http://dietpi.com/downloads/testing/libjvm.so -O /usr/lib/jvm/java-8-openjdk-armhf/jre/lib/arm/server/libjvm.so
```

Fourdee, 2018-04-04:
## Another method/workaround:
```
G_AGI ca-certificates-java
G_AGI openjdk-8-jre-headless
G_AGI openjdk-8-jdk
```

Confirmed RPi repo is only affected by this issue.

Fourdee, 2018-04-04:
As workarounds go, not ideal as ```openjdk-9-jre-headless``` is installed, however, this fixes the installation error with the ```openjdk``` package in Raspbian repo, and, will be applied to v6.7.

Completed.

closed 2018-04-04 by Fourdee

ravikv123, 2019-11-02:
**apt-get upgrade** may solve the issue...

MichaIng, 2019-11-02:
Jep `apt-get -f install` is the workaround we've implemented as well.