MichaIng/DietPi #1683: no support for native ext4 encryption in rpi dietpi 6.6

finasfinas, 2018-04-04:
## Creating a bug report/issue:

Support for native ext4 encryption has been added to the kernel since version 4.1. It is enabled with CONFIG_EXT4_ENCRYPTION=1

A running kernel can be checked for support by looking at the existence of the following file:
/sys/fs/ext4/features/encryption
If this file does not exist, then the kernel has been compiled without the CONFIG_EXT4_ENCRYPTION option and thus has no native ext4 encryption support.

This option does not cause overhead and brings significative gains in ease of use and performance when compared with other options like luks+ dm-crypt, etc. 

Userland tools to create and mount an encrypted ext4 filesystem is already present in DietPI as the version shipped of efsutils already has support for this.

#### Required Information:
- DietPi Version | 6.6
- SBC Device RPi3+
- Power supply used: genuine rpi power supply
- SD card used: sandisk ultra
- DietPI 6.6 4.14.31-v7+ #1104 SMP Thu Mar 29 16:52:18 BST 2018 armv7l GNU/Linux

#### Expected behaviour:
I should be able to create an encrypted ext4 filesystem with the command:
mkfs.ext4 -O encrypt /dev/mmcblk0pX
and then encrypt it with:
 e4crypt add_key "key"

#### Actual behaviour:
e4crypt fails because the kernel does not have support for encrypted ext4 filesystems

#### Steps to reproduce:
please read above

#### Did you submit a ```dietpi-bugreport```?
I did not, I thought that this was the place to report bugs.

#### Extra details:
nothing really, just please enable the option in kernel and issue a kernel update. :)

SuBLiNeR, 2018-04-04:
Hmm... ever tried to look for this via google? - https://github.com/NicoHood/NicoHood.github.io/wiki/Raspberry-Pi-Encrypt-Root-Partition-Tutorial

finasfinas, 2018-04-04:
There are many ways to encrypt a filesystem or a directory. There is no way that is best in all aspects than all the others.
Native ext4 encryption has two strengths: 
- it is faster
- is has a lower memory footprint
- it is easy to implement
- it will probably be default on next version of ubuntu

Regardless of this, the fact is that there is no support on dietpi/raspian because of the lack of a kernel module that has no good reason to not be enabled.
Another thing is: all the userland tools that are needed to implement this are already part of dietpi.

Finally, in my particular use case, native ext4 encryption is the way to go, all others suck. :)

SuBLiNeR, 2018-04-04:
@finasfinas I suggest you to closes this Issue because it isn't a general issue of DIetPi itself. It is a more relevant issue and task which you may ask better at the ppl of the raspbian building team.

This is nor a problem of DietPi nor a problem which i guess could the team resolve. Best is, as said, to open a new issue at the raspbian site. :)

finasfinas, 2018-04-04:
I already did. I opened a ticket here because I don't know what kernel dietpi uses. Does it use the one from raspbian or is it custom made by dietpi team?

SuBLiNeR, 2018-04-04:
@finasfinas it uses the one which is being delivered by raspbian/debian as standard - 4.14.30 is the actual one. The DietPi-team is not self-compiling kernels for theirs distro.

finasfinas, 2018-04-04:
cool, thanks for the info. I will close this bug report and proceed in raspbian.

closed 2018-04-04 by finasfinas