MichaIng/DietPi #1690: Cannot change the "CPU Min Freq Limit"

Fray2000, 2018-04-06:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version 6.6
- SBC Device RPi 2 B+
- Power supply used 5v 3A
- SD card used Microdrive 8GB [Classe 10]
- Distro Dietpi 6.6

#### Expected behaviour:
Ability to select the CPU Min Freq Limit

#### Actual behaviour:
It crashes with this error  ;

/DietPi/dietpi/dietpi-config: array assign: line 1916: syntax error near unexpected token `('
/DietPi/dietpi/dietpi-config: array assign: line 1916: `=("$(( ${available_frequency_array[$i]} / 1000 ))" "MHz"'

#### Extra details:
This error comes on a freshly installed Dietpi 6.6 with the Nextcloud package also installed and running
Link to the configutation https://imgur.com/a/WUf1K

MichaIng, 2018-04-06:
@Fray2000 
Oh yeah, this looks like a typo, thanks for reporting this, I will send commit to fix it for v6.7. You can fix it manually until then.
- Find these lines within your `/DietPi/dietpi/dietpi-config`: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-config#L1916-L1923
- Replace the `=+=` that occurs two times there with `+=`, so the first equality character is wrong there, causing the syntax error

Fourdee, 2018-04-07:
Confirmed working on RPi 3.

@MichaIng 

Legend, many thanks for fixing this 👍

closed 2018-04-07 by Fourdee