MichaIng/DietPi #1691: Kernel panic on boot

gpioneer90, 2018-04-07:
Not sure what happened, my RPi 3 was working dandy yesterday. Today it was unresponsive and a reboot produces the following error messages and hangs. I was on 6.6 before

https://imgur.com/a/htt5w

Fourdee, 2018-04-07:
@gpioneer90 

Looks like a filesystem corruption, usually due to:
- Power loss
- Insufficient PSU
- Unstable SD card

You'll probably want to reinstall the system from scratch. Please use a well branded SD card and official PSU for stability.

I'll mark this as closed due to it being a hardware/power issue.

closed 2018-04-07 by Fourdee

gpioneer90, 2018-04-07:
It all came in a pack from Fry's (a well known electronics retailer in the US). Bummer. Worth posting though in the rare case something is up with 6.6 and more people have this problem (hopefully not).