MichaIng/DietPi #1695: Error in DietPi-Config, settings low freq to min. manually

SuBLiNeR, 2018-04-10:
## Creating a bug report/issue:
dietpi-config error while trying to set minimum frequency to device!

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```? 6.6
- SBC Device (EG: RPi 3)? OrangePi / Raspberry Pi 2 / BananaPi 
- Power supply used (EG: 5v 1A RAVpower)? Original PSU with 5V 2.1A
- SD card used (EG: Sandisk ultra)? SanDisk EXTREME
- Distro (EG: Jessie) | ```uname -a```? latest stable on every built

#### Additional Information (if applicable):
- DietPi-config
- Can this issue be replicated on a fresh installation of DietPi? YES 

#### Expected behaviour:
If changing frequency to minimum available scaled down frequency, it could be set!

#### Actual behaviour:
dietpi-config crashes while starting tool to set minium frequency with console output.

#### Steps to reproduce:
start dietpi-config -> Performance Options -> set CPU Min Freq Limit - crash to console!

#### Extra details:
Console output:

> /DietPi/dietpi/dietpi-config:

 array assign: line 1916: syntax error near unexpected token `('

> /DietPi/dietpi/dietpi-config:

 array assign: line 1916: `=("$(( ${available_frequency_array[$i]} / 1000 ))" "MHz"'

MichaIng, 2018-04-10:
@SuBLiNeR 
Thanks for your report. Jep this accidental typo is known: https://github.com/Fourdee/DietPi/issues/1690
Fixed for v6.7 and until then you can fix by yourself: see link ⬆️

SuBLiNeR, 2018-04-10:
Totally forget that i've read this two days ago.... Shame on me!

closed 2018-04-10 by SuBLiNeR