MichaIng/DietPi #1696: DietPi-Software | Pi-hole: Solve issues with internal installer/updater

MichaIng, 2018-04-10:
Let's focus on this and get it solved for v6.7:
New issue to merge: https://github.com/Fourdee/DietPi/issues/1313 + https://github.com/Fourdee/DietPi/issues/1375 + https://github.com/Fourdee/DietPi/issues/1623

Pi-hole internal installer and updater breaks DietPi with the following:
- Installs `lighttpd`, even if other webserver is already present, causing one to fail.
- Overwrites `lighttpd.conf`, reverting Pi-hole to be the only served web page.
- Installs `dhcpcd5`, possibly causing issues in combination with DietPi default `isc-dhcp-client`.

As we cannot know, when user starts `pihole -up` / `-r`, thus also cannot directly edit Pi-hole scripts, a solution worth testing is to adjust `pihole-FTP.service` start/stop scripts via `/etc/systemd/system/pihole-FTP.service.d/dietpi.conf`.

So far I have this:
```
root@DietPi:/var/www/pihole# cat /etc/systemd/system/pihole-FTL.service.d/dietpi.conf
[Service]
ExecStartPre=/bin/bash -c '/var/lib/dietpi/dietpi-software/services/pihole-startpre.sh'
ExecStopPost=/bin/bash -c '/var/lib/dietpi/dietpi-software/services/pihole-stoppost.sh'
root@DietPi:/var/www/pihole# cat /var/lib/dietpi/dietpi-software/services/pihole-stoppost.sh
#!/bin/bash
# As PiHole overwrites our lighttpd.conf, we need to do a backup in case, using pihole-FTL ExecStopPost.

# Do lighttpd.conf backup, if installed via dietpi-software:
grep -q "aSOFTWARE_INSTALL_STATE\[84\]=2" /DietPi/dietpi/.installed && [ ! -f /mnt/dietpi_userdata/lighttpd.conf.bak ] && cp /etc/lighttpd/lighttpd.conf /mnt/dietpi_userdata/lighttpd.conf.bak

exit 0
root@DietPi:/var/www/pihole# cat /var/lib/dietpi/dietpi-software/services/pihole-startpre.sh
#!/bin/bash
# As PiHole installs lighttpd, dhcpcd5 and overwrites our lighttpd.conf,
# we need to purge those packages in case or recover our lighttpd.conf.
# To assure this on installation, reconfiguration and update, we do this on pihole-FTL ExecStartPre.

dpkg --get-selections > /tmp/dpkg-selections

# Purge dhcpcd5, if our default isc-dhcp-client is installed:
grep -q 'dhcpcd5' /tmp/dpkg-selections && grep -q 'isc-dhcp-client' /tmp/dpkg-selections && DEBIAN_FRONTEND=noninteractive apt-get purge -y dhcpcd5

# Purge lighttpd, if another webserver is (to be) installed via dietpi-software:
grep -q "aSOFTWARE_INSTALL_STATE\[8[35]\]=[12]" /DietPi/dietpi/.installed && grep -q 'lighttpd' /tmp/dpkg-selections && DEBIAN_FRONTEND=noninteractive apt-get purge -y lighttpd

# Recover lighttpd.conf, if backup was made:
grep -q "aSOFTWARE_INSTALL_STATE\[84\]=2" /DietPi/dietpi/.installed && mv /mnt/dietpi_userdata/lighttpd.conf.bak /etc/lighttpd/lighttpd.conf

exit 0
root@DietPi:/var/www/pihole# systemctl status pihole-FTL.service
● pihole-FTL.service - LSB: pihole-FTL daemon
   Loaded: loaded (/etc/init.d/pihole-FTL; generated; vendor preset: enabled)
  Drop-In: /etc/systemd/system/pihole-FTL.service.d
           └─dietpi.conf
   Active: active (running) since Tue 2018-04-10 18:17:38 CEST; 10s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 26269 ExecStopPost=/bin/bash -c /var/lib/dietpi/dietpi-software/services/pihole-stoppost.sh (code=exited, status=0/SUCCESS)
  Process: 26238 ExecStop=/etc/init.d/pihole-FTL stop (code=exited, status=0/SUCCESS)
  Process: 26281 ExecStart=/etc/init.d/pihole-FTL start (code=exited, status=0/SUCCESS)
  Process: 26273 ExecStartPre=/bin/bash -c /var/lib/dietpi/dietpi-software/services/pihole-startpre.sh (code=exited, status=0/SUCCESS)
    Tasks: 5 (limit: 4915)
   CGroup: /system.slice/pihole-FTL.service
           └─26328 /usr/bin/pihole-FTL

Apr 10 18:17:38 DietPi systemd[1]: Stopped LSB: pihole-FTL daemon.
Apr 10 18:17:38 DietPi systemd[1]: Starting LSB: pihole-FTL daemon...
Apr 10 18:17:38 DietPi pihole-FTL[26281]: Not running
Apr 10 18:17:38 DietPi pihole-FTL[26281]: chown: cannot access '/etc/pihole/dhcp.leases': No such file or directory
Apr 10 18:17:38 DietPi su[26322]: Successful su for pihole by root
Apr 10 18:17:38 DietPi su[26322]: + ??? root:pihole
Apr 10 18:17:38 DietPi su[26322]: pam_unix(su:session): session opened for user pihole by (uid=0)
Apr 10 18:17:38 DietPi pihole-FTL[26281]: FTL started!
Apr 10 18:17:38 DietPi systemd[1]: Started LSB: pihole-FTL daemon.
```
- Now forcing reinstall:
```
root@DietPi:/var/www/pihole# pihole -r
...
  [i] Checking for lighttpd (will be installed)
...
  [✓] Starting lighttpd service
  [✓] Enabling lighttpd service to start on reboot
...
  [✓] Pi-hole blocking is Enabled

  [i] The install log is located at: /etc/pihole/install.log
  Update Complete!
...
root@DietPi:/var/www/pihole# journalctl -r
Apr 10 18:01:18 DietPi lighttpd[23242]: 2018-04-10 18:01:18: (network.c.473) listen failed:  Address already in use
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Main process exited, code=exited, status=255/n/a
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Apr 10 18:01:18 DietPi bash[23188]: The following package was automatically installed and is no longer required:
Apr 10 18:01:18 DietPi bash[23188]:   libfam0
Apr 10 18:01:18 DietPi bash[23188]: Use 'apt autoremove' to remove it.
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
Apr 10 18:01:18 DietPi bash[23188]: The following packages will be REMOVED:
Apr 10 18:01:18 DietPi bash[23188]:   lighttpd*
Apr 10 18:01:18 DietPi systemd[1]: Stopped Lighttpd Daemon.
Apr 10 18:01:18 DietPi systemd[1]: Starting Lighttpd Daemon...
Apr 10 18:01:18 DietPi systemd[1]: Started Lighttpd Daemon.
Apr 10 18:01:18 DietPi lighttpd[23265]: 2018-04-10 18:01:18: (network.c.473) listen failed:  Address already in use
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Main process exited, code=exited, status=255/n/a
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
Apr 10 18:01:18 DietPi systemd[1]: Stopped Lighttpd Daemon.
Apr 10 18:01:18 DietPi systemd[1]: Starting Lighttpd Daemon...
Apr 10 18:01:18 DietPi systemd[1]: Started Lighttpd Daemon.
Apr 10 18:01:18 DietPi bash[23188]: 0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
Apr 10 18:01:18 DietPi bash[23188]: After this operation, 974 kB disk space will be freed.
Apr 10 18:01:18 DietPi lighttpd[23294]: 2018-04-10 18:01:18: (network.c.473) listen failed:  Address already in use
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Main process exited, code=exited, status=255/n/a
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Apr 10 18:01:18 DietPi bash[23188]: [614B blob data]
Apr 10 18:01:18 DietPi bash[23188]: Removing lighttpd (1.4.45-1) ...
Apr 10 18:01:18 DietPi systemd[1]: Reloading.
Apr 10 18:01:19 DietPi systemd[1]: Stopped Lighttpd Daemon.
Apr 10 18:01:19 DietPi systemd[1]: Reloading.
Apr 10 18:01:19 DietPi systemd[1]: Reloading.
Apr 10 18:01:19 DietPi bash[23188]: [614B blob data]
Apr 10 18:01:19 DietPi bash[23188]: Purging configuration files for lighttpd (1.4.45-1) ...
...
```
- As can be seen, lighttpd fails due to port already used by apache and startup attempt is repeated until our purge is done.
- Pi-hole installation went through without error, but I also already saw errors there and I am not sure, if the installation/update goes through fully in every case. Needs some testing on fresh system again.
- We could stop/start apache/nginx with pihole-FTL to assure Lighttpd is working correctly during update/install/reconfigure, but I don't like this done "silently", where users might want to stop pihole but keep webserver running...
- The missing /etc/pihole/dhcp.leases btw. has nothing to do with dhcpcd5, as on my VM I test systemd-networkd, thus do not have isc-dhcp-client installed, thus dhcpcd5 is not purged 😉.
_________________________
Pi-hole btw. will merge dnsmasq into pihole-FTP: https://pi-hole.net/2018/03/24/help-us-beta-test-ftldns/
- As on current state the service name did not change, this doesn't affect the solution above.

SuBLiNeR, 2018-04-10:
btw... i can't simulate this behaviour, I've installed pihole since v6.1 and than again with 6.3 freshly after i messed stuff badly.... 
I've never had any problems on mine 2 RPi3 upon this. Never seen any errors...
Would you mind to explain it a bit more?

Fourdee, 2018-04-12:
Yet more workarounds, endless cycle.

Ideally, we need to work on the Pi-Hole install script itself, add command line option support for disabling setup of any webserver. Also uninstall ```isc-dhcp-client``` on debian if required.

In the perfect world, add support for selecting a choice webserver to use, however, I don't currently have the time for this and i feel i'd only end up completely rewriting it from scratch https://raw.githubusercontent.com/pi-hole/pi-hole/master/automated%20install/basic-install.sh

A quick solution to prevent Lighttpd install could be, sed this out:
```
https://github.com/pi-hole/pi-hole/blob/master/automated%20install/basic-install.sh#L168-L169
```

Interesting unattended install:
https://github.com/pi-hole/pi-hole/blob/master/automated%20install/basic-install.sh#L1966

MichaIng, 2018-04-12:
@SuBLiNeR 
As long as you install it besides Lighttpd (DietPi default as well) and do not use `pihole -up` / `pihole -r` to update or reconfigure it, you will not face any issues.
Our installation script already handles the initial replacement of `lighttpd.conf` to make pi-hole web ui available at `my.domain.org/pihole` instead of `my.domain.org`, allowing other web services besides. But pi-hole internal installer/updater scripts will revert it again, if executed.

You will also face issues now with the latest pi-hole, if you choose another webserver preference, since pi-hole now installs Lighttpd besides and two webservers, trying to listen at same port 80, will break each other. This is new as far as I know, so was not yet the case when DietPi v6.3 was current.

k-plan, 2018-04-12:
@MichaIng 

> - Install DHCP client to use it's configuration file to assure disabled DHCP lease on all systems :rofl: 

Hmm ... no. They need install and need `dhcpcd5` to set a static ip address on Rasbian. But correct, only one time. After this, no need any more.

It will change some lines in  `/etc/dhcpcd.conf`, which doesn't exist on DietPi, because here` isc-dhcp-client` is used. Config file is here in : 
```
root@RPi-Zero-W:~# cat /etc/dhcp/dhclient.conf
# Configuration file for /sbin/dhclient.
#
.....
```

https://github.com/pi-hole/pi-hole/blob/master/automated%20install/basic-install.sh#L645-L667

https://discourse.pi-hole.net/t/hardware-software-requirements/273 => see: `IP Addressing`

-----------------------------------

> - that an update, and whether we want or not users will run pihole -up, will replace our adjustments, so that we anyway need to track those executions to revert our adjustments, that then might break the update etc...

Logic of `pihole -up` : 

https://github.com/pi-hole/pi-hole/blob/master/advanced/Scripts/update.sh#L151-L159

-----------------------------------

```
Pi-hole btw. will merge dnsmasq into pihole-FTP: https://pi-hole.net/2018/03/24/help-us-beta-test-ftldns/

- As on current state the service name did not change, this doesn't affect the solution above.

```

That is correct. But if `pihole-FTL` with FTLDNS™ will start to rollout, we must keep a eye on it.
We are not allowed to control `pihole-FTL` by `dietpi-services`  any more.

https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-services#L160

Or we will run into the same issue as with `dnsmasq` before.

https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-services#L175-L184

And, at the moment, if `dietpi-services`  control `pihole-FTL` , what it does, your  new service `/etc/systemd/system/pihole-FTL.service.d/dietpi.conf` will force a reinstall, if you do a 

 `root@RPi-Zero-W:~# dietpi-services restart restart`

Correct? Can't check it at the moment.

--------------------------

MichaIng, 2018-04-12:
@k-plan 
> They install and use dhcpcd5 to set a static ip address on Raspbian
Hmm, it tells dhcpcd to use a static IP / not lease IP via DHCP, doesn't it? But nothing else reads dhcpcd.conf... Would be also possible to do this via /etc/network/interfaces (ifupdown) without any DHCP server installed and even if, they would be inactive (dhclient needs `...inet dhcp`, dhcpcd need `...inet manual`, whereas `inet static` disables all DHCP leases AFAIK).
But using /etc/network/interfaces on the other hand doesn't work with systemd-networkd and it's internal DHCP server. It ignore interfaces file completely and has its own. But it leaves DHCP lease to dhclient/dhcpcd, if present.
As resolvconf is installed, this leads to auto disabled systemd-resolved as well.

Thus yeah for me it's still "installing DHCP to disable DHCP" moreless, but indeed it seems to be the way that assures disabled DHCP on most systems.

The only question is, if it is actually wanted to non-optional provide a damn always, in case with hacking methods, out-of-the-box running system, if this means that you limit users in doing their own network and webserver choices. A clear "Assign a static IP and for web ui allow your webserver accessing /var/www/pihole" would do the job from my point of view. But this is a question of taste.

> And dhcpcd will fail as lighttpd as well
Ah yeah, dhcpcd needs `...inet manual` configured interfaces, forgot that. However, it doesn't break something eather.

> But if pihole-FTL with FTLDNS™ will start to rollout, we must keep a eye on it.
We are not allowed to control pihole-FTL by dietpi-services any more.
Good point 👍. Maybe we should already exclude it, or is there a good reason to control it currently?

> your new service /etc/systemd/system/pihole-FTL.service.d/dietpi.conf will force a reinstall
Nope, it will in no case install Lighttpd. It will just check, if Lighttpd was installed via dietpi-software, otherwise purge it, if installed. So besides some checks, the only real action that is currently unnecessarily done on `dietpi-services restart` is, if Lighttpd was installed via dietpi-software, backup lighttpd.conf and recover it again on service start.

Fourdee, 2018-04-13:
This prevents lighttpd install (DietPi pre-installs webservers anyway): https://github.com/Fourdee/DietPi/commit/3e4ba42d3f832cfde5ed83439795ae6b65912e47

🈯️ Tested working with Nginx.

MichaIng, 2018-04-13:
@Fourdee 
Jep, fixes the most urgent issue for now. Still update of Pi-hole will break it again.

Should I create PR for my solution so you can do some testing about it? I am still more satisfied it, as it allows using of pihole commands and for my impression directly edit the pihole install script is a bid dangerous as well, and might need more attention/regular adjustments as the service hooks. But that's indeed a guess 😉.

mariusrugan, 2018-04-13:
@MichaIng,
did you experience the following issue: https://github.com/pi-hole/pi-hole/issues/2109
trying to investigate what could lead to this, zig-zagging in the DietPI+pihole issues, the only thing that is prone to generating such behaviour is the update system via cron. (pihole cron)

MichaIng, 2018-04-13:
@mariusrugan 
Check my answer: https://github.com/pi-hole/pi-hole/issues/2109#issuecomment-381176999

I will check out pihole cron as well, so see if there are potential issues.

MichaIng, 2018-04-13:
@Fourdee @k-plan 
What do you think, shall we already exclude pihole-FTL service from DietPi control, as there are already users that help testing FTLDNS branch? I think there is no big risk that something essential breaks/get corrupted during updates/backups/software installations, as it is for databases/file servers etc.?

k-plan, 2018-04-13:
@MichaIng 

> Check my answer:

Let's check `pihole-FTL` : 

```
root@RPi-Zero-W:~# service pihole-FTL status
● pihole-FTL.service - LSB: pihole-FTL daemon
   Loaded: loaded (/etc/init.d/pihole-FTL; generated; vendor preset: enabled)
   Active: active (exited) since Fri 2018-04-13 00:17:05 CEST; 20h ago
     Docs: man:systemd-sysv-generator(8)
  Process: 15107 ExecStop=/etc/init.d/pihole-FTL stop (code=exited, status=0/SUCCESS)
  Process: 15206 ExecStart=/etc/init.d/pihole-FTL start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/pihole-FTL.service

Apr 13 00:17:05 RPi-Zero-W systemd[1]: Starting LSB: pihole-FTL daemon...
Apr 13 00:17:05 RPi-Zero-W pihole-FTL[15206]: Not running
Apr 13 00:17:05 RPi-Zero-W pihole-FTL[15206]: chown: cannot access '/etc/pihole/dhcp.leases': No such file or directory
Apr 13 00:17:05 RPi-Zero-W su[15229]: Successful su for pihole by root
Apr 13 00:17:05 RPi-Zero-W su[15229]: + ??? root:pihole
Apr 13 00:17:05 RPi-Zero-W su[15229]: pam_unix(su:session): session opened for user pihole by (uid=0)
Apr 13 00:17:05 RPi-Zero-W pihole-FTL[15206]: dnsmasq: failed to create listening socket for port 53: Address already in use
Apr 13 00:17:05 RPi-Zero-W su[15229]: pam_unix(su:session): session closed for user pihole
Apr 13 00:17:05 RPi-Zero-W systemd[1]: Started LSB: pihole-FTL daemon.
```

Why is `port 53: Address already in use` ??

Let's check  `dnsmasq` : 

```
root@RPi-Zero-W:~# service dnsmasq status
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; disabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-04-13 00:17:04 CEST; 20h ago
  Process: 15165 ExecStartPost=/etc/init.d/dnsmasq systemd-start-resolvconf (code=exited, status=0/SUCCESS)
  Process: 15155 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=0/SUCCESS)
  Process: 15152 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCCESS)
 Main PID: 15164 (dnsmasq)
   CGroup: /system.slice/dnsmasq.service
           └─15164 /usr/sbin/dnsmasq -x /run/dnsmasq/dnsmasq.pid -u dnsmasq -r /run/dnsmasq/resolv.conf -7 /etc/dnsmasq.d,
.dpkg-dist,.dpkg-old,.dpkg-new --local-service --trust-anchor=.,19036,8,2,49aac11d7b6f6446702e54a1607371607a1a41855200fd2c
e1cdde32f24e8fb5 --trust-anchor=.,20326,8,2,e06d44b80b8f1d39a95c0b0d7c65d08458e880409bbc683457104237c7f8ec8d

Apr 13 00:17:03 RPi-Zero-W systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Apr 13 00:17:03 RPi-Zero-W dnsmasq[15152]: dnsmasq: syntax check OK.
Apr 13 00:17:04 RPi-Zero-W systemd[1]: Started dnsmasq - A lightweight DHCP and caching DNS server.
```

[image]

If the new `pihole-FTL` with `FTL DNS` doesn't need  `dnsmasq` any more, why does switching git branch and  updater  not uninstall   `dnsmasq` :question: 

Let's check what's will happen:

```
root@RPi-Zero-W:~# apt-get purge dnsmasq
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  dnsmasq-base libnetfilter-conntrack3 libnfnetlink0
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  dnsmasq*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 73.7 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 30883 files and directories currently installed.)
Removing dnsmasq (2.76-5+rpt1+deb9u1) ...
(Reading database ... 30875 files and directories currently installed.)
Purging configuration files for dnsmasq (2.76-5+rpt1+deb9u1) ...
dpkg: warning: while removing dnsmasq, directory '/etc/dnsmasq.d' not empty so not removed
Processing triggers for systemd (232-25+deb9u2) ...
```

`root@RPi-Zero-W:~# service pihole-FTL restart`

```
root@RPi-Zero-W:~# pihole status
  [✓] DNS service is running
  [✓] Pi-hole blocking is Enabled

```

```
root@RPi-Zero-W:~# service pihole-FTL status
● pihole-FTL.service - LSB: pihole-FTL daemon
   Loaded: loaded (/etc/init.d/pihole-FTL; generated; vendor preset: enabled)
   Active: active (running) since Fri 2018-04-13 20:58:16 CEST; 1min 35s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 29465 ExecStop=/etc/init.d/pihole-FTL stop (code=exited, status=0/SUCCESS)
  Process: 29472 ExecStart=/etc/init.d/pihole-FTL start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/pihole-FTL.service
           └─29504 /usr/bin/pihole-FTL

Apr 13 20:58:16 RPi-Zero-W su[29495]: + ??? root:pihole
Apr 13 20:58:16 RPi-Zero-W su[29495]: pam_unix(su:session): session opened for user pihole by (uid=0)
Apr 13 20:58:16 RPi-Zero-W pihole-FTL[29472]: FTL started!
Apr 13 20:58:16 RPi-Zero-W su[29495]: pam_unix(su:session): session closed for user pihole
Apr 13 20:58:16 RPi-Zero-W dnsmasq[29504]: started, version pi-hole-2.79 cachesize 150
Apr 13 20:58:16 RPi-Zero-W dnsmasq[29504]: compile time options: IPv6 GNU-getopt no-MMU no-DBus no-i18n no-IDN DHCP DHCPv6 no-
scripts TFTP no-conntrack ipset auth DNSSEC loop-detect inotify
Apr 13 20:58:16 RPi-Zero-W dnsmasq[29504]: reading /etc/resolv.conf
Apr 13 20:58:16 RPi-Zero-W dnsmasq[29504]: ignoring nameserver 127.0.0.1 - local interface
Apr 13 20:58:16 RPi-Zero-W dnsmasq[29504]: read /etc/hosts - 2 addresses
Apr 13 20:58:16 RPi-Zero-W systemd[1]: Started LSB: pihole-FTL daemon.
```

[image]

Now let's see, if it will run until tomorrow morning.

MichaIng, 2018-04-13:
@k-plan 
Jep, nice one, I wanted to do the same, but can't let my VM run through the night. But will also install it and let it run until I go sleep.

To test FTLDNS there are some instructions from Pi-hole that dnsmasq needs to be disabled. It is still testing, so I assume the final update will include purging or at least disabling dnsmasq.

Port blocking makes sense, as two dnsmasq solutions try to use it.

Fourdee, 2018-04-14:
@MichaIng 

> Maybe we can do some attempt to work together with Pi-hole devs about our other issues.

👍 

Made a start:
https://github.com/Fourdee/pi-hole/commit/dc2d221f0b39017b8e93dc4a504ba531c1878a78

k-plan, 2018-04-14:
@MichaIng 

> Now let's see, if it will run until tomorrow morning.

```
root@RPi-Zero-W:~# pihole status
  [✓] DNS service is running
  [✓] Pi-hole blocking is Enabled
```

```
root@RPi-Zero-W:~# service pihole-FTL status
● pihole-FTL.service - LSB: pihole-FTL daemon
   Loaded: loaded (/etc/init.d/pihole-FTL; generated; vendor preset: enabled)
   Active: active (running) since Sat 2018-04-14 00:17:04 CEST; 13h ago
     Docs: man:systemd-sysv-generator(8)
  Process: 32760 ExecStop=/etc/init.d/pihole-FTL stop (code=exited, status=0/SUCCESS)
  Process: 330 ExecStart=/etc/init.d/pihole-FTL start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/pihole-FTL.service
           └─379 /usr/bin/pihole-FTL

Apr 14 00:17:04 RPi-Zero-W su[370]: + ??? root:pihole
Apr 14 00:17:04 RPi-Zero-W su[370]: pam_unix(su:session): session opened for user pihole by (uid=0)
Apr 14 00:17:04 RPi-Zero-W pihole-FTL[330]: FTL started!
Apr 14 00:17:04 RPi-Zero-W su[370]: pam_unix(su:session): session closed for user pihole
Apr 14 00:17:04 RPi-Zero-W dnsmasq[379]: started, version pi-hole-2.79 cachesize 150
Apr 14 00:17:04 RPi-Zero-W dnsmasq[379]: compile time options: IPv6 GNU-getopt no-MMU no-DBus no-i18n no-IDN DHCP DHCPv6 no-scripts TFTP no-conntrack ipset auth DN
SSEC loop-detect inotify
Apr 14 00:17:04 RPi-Zero-W dnsmasq[379]: reading /etc/resolv.conf
Apr 14 00:17:04 RPi-Zero-W dnsmasq[379]: ignoring nameserver 127.0.0.1 - local interface
Apr 14 00:17:04 RPi-Zero-W dnsmasq[379]: read /etc/hosts - 2 addresses
Apr 14 00:17:04 RPi-Zero-W systemd[1]: Started LSB: pihole-FTL daemon.
```

Alles sieht gut aus und sollte funktionieren. Tut es aber nicht.

```
root@RPi-Zero-W:~# ping dietpi.com
ping: dietpi.com: Temporary failure in name resolution
```

```
root@RPi-Zero-W:~# netstat -ln | grep :53
tcp        0      0 0.0.0.0:53              0.0.0.0:*               LISTEN
tcp6       0      0 :::53                   :::*                    LISTEN
udp        0      0 0.0.0.0:53              0.0.0.0:*
udp6       0      0 :::53                   :::*

```

[image]

`ntp` läuft übrigens, dank fehlender Namensauflösung,  völlig Amok.

[image]
[image]

Aber alles dies hat nichts mit dem eigentlich von dir gestarteten Issue zu tun.

So, muss nun los. Mehr später.

MichaIng, 2018-04-14:
@k-plan 
Interesting, mine ran flawlessly until ~2 am, but to be honest, I am not sure, if I tested name resolution, just SSH, web ui and service status I think.

Today when I started up the VM again, everything worked, besides name resolution, the same issue as you, same enraging NTP 😆.
I tried to resolve it by restarting resolvconf, networking, ifdown/up eth0 and of course pihole-FTL and reboot. Nothing helped. Finally completely disabling pihole, reverting to default gateway DNS and updating pihole helped.

I tested if dietpi-services (starting pihole-FTL after most other services compared to systemd) is the issue, but seems not.

Now I found the following: `service networking restart` instantly breaks name resolution. I am not sure what is done by this, but as it basically does `ifup -a`, I think `/etc/network/interfaces` should be read? There `dns-nameservers` entry still shows my gateway IP, where `/etc/resolv.conf` shows localhost (127.0.0.1), not sure what should be, actually for my impression, it makes sense that the local DNS server itself does not resolve for himself, just for external clients, but I do not understand too much about this anyway 🤣.

So I tried changing the dns-nameservers entry, commenting it etc. but nothing helped. Instead after a restart it doesn't matter, everything works again. Seems the entry is ignored until networking restart breaks it. Also:
```
root@VM-Stretch:~# systemctl status networking
● networking.service - Raise network interfaces
   Loaded: loaded (/lib/systemd/system/networking.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sat 2018-04-14 14:35:37 CEST; 4min 33s ago
     Docs: man:interfaces(5)
  Process: 317 ExecStart=/sbin/ifup -a --read-environment (code=exited, status=0/SUCCESS)
  Process: 309 ExecStartPre=/bin/sh -c [ "$CONFIGURE_INTERFACES" != "no" ] && [ -n "$(ifquery --read-environment --list --exclude=lo)" ] && udevadm settle
(code=exited, status=1/FAILURE)
 Main PID: 317 (code=exited, status=0/SUCCESS)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/networking.service

Apr 14 14:35:36 VM-Stretch systemd[1]: Starting Raise network interfaces...
Apr 14 14:35:37 VM-Stretch systemd[1]: Started Raise network interfaces.
```
Success everywhere, but overall FAILURE. Not sure what actually failed, no more logs

Okay, run ifdown/up in verbose mode. Hmm, however just lo interface starts up, not eth0. `ifup eth0` manually fixes name resolution. Verifying:
[image]

Hmm, funny that at first restart didn't help, not sure if this is it and what's the reason why ifup -a/networking fails/stops at starting up eth0...

At least can you verify this `ifup -a` issue and/or have an idea why this could happen?

MichaIng, 2018-04-14:
@Fourdee 
PR on this: https://github.com/Fourdee/pi-hole/pull/1
- Write webserver/cron installation decisions to setup vars file as well, so they will be handled by `pihole -up` automatically (runs by default as unattended). Just `pihole -r` would still break it.
- I think it would make sense then to implement a separate whiptail --yesno after web interface decision to ask for webserver as well.
- Btw. pi-hole cron jobs do not create any issues for me, they make totally sense, see no reason to disable.

k-plan, 2018-04-14:
@MichaIng 

> At least can you verify this ifup -a issue and/or have an idea why this could happen?

Das gleiche, wie bei dir. 

```
root@RPi-Zero-W:~# ifup -a --verbose
/bin/run-parts --exit-on-error --verbose /etc/network/if-pre-up.d
run-parts: executing /etc/network/if-pre-up.d/ethtool
run-parts: executing /etc/network/if-pre-up.d/wireless-tools
run-parts: executing /etc/network/if-pre-up.d/wpasupplicant
/bin/run-parts --exit-on-error --verbose /etc/network/if-up.d
run-parts: executing /etc/network/if-up.d/000resolvconf
run-parts: executing /etc/network/if-up.d/ethtool
run-parts: executing /etc/network/if-up.d/upstart
run-parts: executing /etc/network/if-up.d/wpasupplicant

```
Weder mit `/etc/init.d/networking restart` noch mit `service networking restart` oder `systemctl restart networking.service` es endet immer mit:

```
root@RPi-Zero-W:~# /etc/init.d/networking restart
[....] Restarting networking (via systemctl): networking.service

```
... und dann hilft nur noch den Stecker ziehen und Hard-Reset.

Auch nach einem `reboot` funktioniert die Namensauflösung nicht.

Aber das lässt sich fixen mit:

```
root@RPi-Zero-W:~# ping dietpi.com
ping: dietpi.com: Temporary failure in name resolution

root@RPi-Zero-W:~# pihole restartdns
  [✓] Restarting DNS service

root@RPi-Zero-W:~# ping dietpi.com
PING dietpi.com (185.101.92.145) 56(84) bytes of data.
64 bytes from 185.101.92.145 (185.101.92.145): icmp_seq=1 ttl=54 time=14.1 ms
64 bytes from 185.101.92.145 (185.101.92.145): icmp_seq=2 ttl=54 time=13.9 ms

```

Okay?! Schauen wir nach, wie das in der Beta FTLDNS Version funktioniert:

https://github.com/pi-hole/pi-hole/blob/FTLDNS/pihole#L333-L349

Ahhh, `resolver`

```
root@RPi-Zero-W:~# service resolvconf status
● resolvconf.service - Nameserver information manager
   Loaded: loaded (/lib/systemd/system/resolvconf.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sat 2018-04-14 19:17:03 CEST; 35min ago
     Docs: man:resolvconf(8)
 Main PID: 99 (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/resolvconf.service

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

```

In der jetzigen Version ist das noch anders:

https://github.com/pi-hole/pi-hole/blob/master/pihole#L331-L347

Aha, `dnsmasq` alles klar?

-------------------

Das "DNS-Einfrier-Problem" wird aber trotzdem exakt um 00:17  wieder auftreten.
Warum? Eigentlich ist da die Uhrzeit, wo standardmäßig `cron.hourly` läuft, nicht aber `cron.daily`.
Hmmm ... noch mal in der Vergangenheit gekramt. Ja, da war doch was mit den log Dateien und RAM-Log in Verbindung mit `pihole`.

https://github.com/Fourdee/DietPi/issues/918

https://github.com/Fourdee/DietPi/commit/db84245b5fb3d386f814f683fc4d362549830464

Und hier dürfte der Hund begraben sein, dass es immer einmal am Tag um 00:17 passiert:

https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-logclear#L164-L177

k-plan, 2018-04-14:
@Fourdee 

> Made a start, but ended up nearly rewriting everything, couldn't help myself lol:

Déjà-vu ?  lol

Remind me back to the days, when dietpi supports only 30 software titles and  have his own pihole fork.

Runs really cool but was a lot of work for you and a never ending story ... https://github.com/Fourdee/DietPi/issues/427

A lot of trouble for you and with pihole developers .... 
So I personally can understand the decision, to give up the own fork and goes with `install.pi-hole.net` installer.

Main problem with this sort of software is, they are only focus on some devices (see: `dhcpd5`) and it doesn't considerate of other software or distributions, to keep their code and work clean and simple.

> I wonder if this is something to do with how DietPi handles things, ...

If I read patter like this, the opposite party should give thought as well, why dietpi must do this?  It's really difficult and annoying to reversing every code changes and get a head on, what coders are want to score. 
It will be more easy to put this installation into wast bin.

MichaIng, 2018-04-14:
@k-plan 
> pihole restartdns

Interesting, this doesn't work for me, eth0 interface stays down, need to `ifup eth0` to resolve it.
I use FTLDNS branch btw 😉.

About the log issue, yeah might be a reason, that Pi-hole does not like DietPi-RamLog? On the other hand, no other software has issues with it (?). But easy to test by disabling DietPi-RamLog, I have it always disabled btw, use nearly only `journalctl`.

> https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-logclear#L164-L177

Ahhhh yes that makes totally sense. dnsmasq is started first, thus breaks pihole-FTL start afterwards 👍 👍 !

@mariusrugan 
Please try to just `apt purge dnsmasq`, if you use the new FTLDNS branch. This prevents any unwanted start of this now conflicting service.

MichaIng, 2018-04-14:
@k-plan 
Interesting, dietpi-logclear and/or restarting pihole-FTL doesn't cause any issue on my VM. Name resolving still works, even that resolvconf also shows the status about rotated logs:
```
root@VM-Stretch:~# dietpi-logclear 1
[  OK  ] Root access verified.
[ INFO ] DietPi-Logclear | Clearing log entries for PiHole, that are not from today, please wait...
[ INFO ] DietPi-Logclear | PiHole logs have been changed, restarting, please wait...
Failed to stop dnsmasq.service: Unit dnsmasq.service not loaded.
Failed to start dnsmasq.service: Unit dnsmasq.service not found.
[ SUB1 ] DietPi-Process_tool > Apply
[  OK  ] DietPi-Process_tool | Cron (1015) : Nice      0
[  OK  ] DietPi-Process_tool | Cron (1015) : Affinity  0
[  OK  ] DietPi-Process_tool | Cron (1015) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Dropbear (402) : Nice      0
[  OK  ] DietPi-Process_tool | Dropbear (402) : Affinity  0
[  OK  ] DietPi-Process_tool | Dropbear (402) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Dropbear (2054) : Nice      0
[  OK  ] DietPi-Process_tool | Dropbear (2054) : Affinity  0
[  OK  ] DietPi-Process_tool | Dropbear (2054) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Pi-hole FTL (3722) : Nice      0
[  OK  ] DietPi-Process_tool | Pi-hole FTL (3722) : Affinity  0
[  OK  ] DietPi-Process_tool | Pi-hole FTL (3722) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Webserver Lighttpd (986) : Nice      0
[  OK  ] DietPi-Process_tool | Webserver Lighttpd (986) : Affinity  0
[  OK  ] DietPi-Process_tool | Webserver Lighttpd (986) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Webserver Php-fpm (959) : Nice      0
[  OK  ] DietPi-Process_tool | Webserver Php-fpm (959) : Affinity  0
[  OK  ] DietPi-Process_tool | Webserver Php-fpm (959) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Webserver Php-fpm (964) : Nice      0
[  OK  ] DietPi-Process_tool | Webserver Php-fpm (964) : Affinity  0
[  OK  ] DietPi-Process_tool | Webserver Php-fpm (964) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Completed
[ INFO ] DietPi-Logclear |
 - Log file directory      | /var/log
 - Processed files         | 26
 - Excluded files          | 0
 - Cleared log files       | 24
 - Unsupported files       | 0
 - Deleted files           | 0
 - Space cleared           | 18554 KB

root@VM-Stretch:~# ping dietpi.com
PING dietpi.com (185.101.92.145) 56(84) bytes of data.
64 bytes from 185.101.92.145 (185.101.92.145): icmp_seq=1 ttl=55 time=21.2 ms
64 bytes from 185.101.92.145 (185.101.92.145): icmp_seq=2 ttl=55 time=21.2 ms
--- dietpi.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1002ms
rtt min/avg/max/mdev = 21.228/21.249/21.271/0.147 ms
root@VM-Stretch:~# systemctl status resolvconf
● resolvconf.service - Nameserver information manager
   Loaded: loaded (/lib/systemd/system/resolvconf.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sat 2018-04-14 19:47:13 CEST; 3h 42min ago
     Docs: man:resolvconf(8)
  Process: 171 ExecStart=/sbin/resolvconf --enable-updates (code=exited, status=0/SUCCESS)
  Process: 163 ExecStartPre=/bin/touch /run/resolvconf/postponed-update (code=exited, status=0/SUCCESS)
  Process: 131 ExecStartPre=/bin/mkdir -p /run/resolvconf/interface (code=exited, status=0/SUCCESS)
 Main PID: 171 (code=exited, status=0/SUCCESS)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/resolvconf.service

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
root@VM-Stretch:~# dietpi-logclear 2
[  OK  ] Root access verified.
[ INFO ] DietPi-Logclear |
 - Log file directory      | /var/log
 - Processed files         | 26
 - Excluded files          | 0
 - Cleared log files       | 0
 - Unsupported files       | 0
 - Deleted files           | 1
 - Space cleared           | 0 KB

root@VM-Stretch:~# ping dietpi.com
PING dietpi.com (185.101.92.145) 56(84) bytes of data.
64 bytes from 185.101.92.145 (185.101.92.145): icmp_seq=1 ttl=55 time=23.1 ms
64 bytes from 185.101.92.145 (185.101.92.145): icmp_seq=2 ttl=55 time=19.9 ms
--- dietpi.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 19.921/21.519/23.118/1.605 ms
```
However, the started dnsmasq service definitely causes problems as well with FTLDNS branch, thus purging it us necessary to prevent these from my point of view.

Lol, so you face just the resolvconf issue, me just the ifupdown/networking issue, I am confused, especially since I don't understand what is going on for both of them 🤔 😆.

> Will we now focus on the upcoming FTLDNS release?

Since the Lighttpd issue is at least on Pi-hole installation fixed for now, I think it is best if we focus on the future version. Hmm @Fourdee can you commit be applied to the FTLDNS branch as well, at least without too much rework?

Fourdee, 2018-04-15:
@MichaIng 

> Hmm @Fourdee can you commit be applied to the FTLDNS branch as well, at least without too much rework?

PR sent for dev branch:
https://github.com/pi-hole/pi-hole/pull/2115

Will nudge this to v6.8, see no reason to delay v6.7 any further.

k-plan, 2018-04-15:
@Fourdee 

> PR sent for dev branch: `--disable-install-webserver`

A nice one. Lets wait and see what will happend. Hope pihole dev team will successful review and merge your commit.

**Summary** (so far) : 

- 🈯️   - temporary "workaround" for `lighttpd` installation problem in v6.7 https://github.com/Fourdee/DietPi/commit/f30a855335a5b61448268ca4f17c75e809213590

- 🈷️   - work in process: PR sent for dev branch: `--disable-install-webserver` => result stay open

- 🈴  - can not find/see a fix/workaround for `dhcpd5` installation

- 🈴  - can not find/see a fix/workaround to use the updated block page  #1313 with any web server stack  supported by DietPi

- 🈴  - upcoming `ftldns` of `pihole` is not supported at the moment.

Correct so far? Or missed something?

MichaIng, 2018-04-15:
@k-plan 
Jep looks right so far.
- DHCP: Still I don't see any real issue with this, dhcpcd5 and isc-dhcp-client being installed side-by-side. We had this within our images as well for a long time, as dhcpcd5 is marked as "required" package, thus will never be autoremoved. You can easily check this on VM at least, I always need to actively purge it after pihole installation 😆, apt-mark auto + G_AGA does not do the job. dhcpcd5 just fails without hurting something,, as we configure interfaces with static or dhcp prefix, where dhcpcd need "manual".
But yeah, still would be nice to directly skip the installation 👍 .
- `pihole -up` fix is included with Fourdees PR, as disabled webserver will be remembered/saved to setup variables file.
- Ah that's true, also found the bid more beautiful block page, hmm should be simple to replace our old one?
- I think it's "okay" to not support FTLDNS until it's official release. But jep vote for support with v6.8, as it is quite easy to realize as well and will not break current pihole release: Just exclude pihole-FTL from dietpi-services.

MichaIng, 2018-04-17:
@k-plan @Fourdee 
FTLDNS support: https://github.com/Fourdee/DietPi/pull/1714
- Exclude pihole-FTL from dietpi-services control to prevent broken name resolution.
- DietPi-Logclear: Restart dnsmasq only, if found enabled, otherwise FTLDNS seems to be active, which conflicts dnsmasq.

MichaIng, 2018-04-23:
PR merged into Pi-hole development branch 🍾 👍. Not sure about the master branch schedule, but I will create some DietPi branch to adjust/test with Pi-hole dev branch.

k-plan, 2018-04-23:
@Fourdee 

> _jacobsalmela wrote:_
> @ Fourdee please contact me for a special gift for contributing to our repo.

wow ...  :+1:   nice trait. Have not reckon that this will happen. 

If we now get a update by pihole, we can check, if `pihole -up` will work for us or we have to fix something.
I'm curious about it.

MichaIng, 2018-04-29:
@Fourdee @k-plan 
I tried to test the Pi-hole development branch (the only one, which contains Fourdee's PR), but sadly it's kinda horrible to do and failed as well:
- Installation worked fine, Lighttpd was left out.
- FTLDNS is included with this branch by the way, at least with the pihole-FTL branch. Pi-hole core does `pihole-FTL --resolver` to check if it is FTLDNS or dnsmasq is needed.
- But the Pi-hole installer script installs the current release branch, at least for FTL, web interface and the scripts in `/opt/pihole`, thus manual update is necessary then. `/opt/pihole` are sadly not updated at all during my tests 🙁.
- Found the reason for not working checkout of web interface: https://github.com/pi-hole/pi-hole/pull/2153 **€: Ah nice already merged 😄**
- `echo 'development' > /etc/pihole/ftlbranch` allows to control FTL version. Sadly the current development pihole-FTL version creates some error on startup about failing permissions to bind to listening port 53 (definitely nothing else bound to this port!). Reverting everything to FTLDNS branch worked, but then our changes are not present (and variables not understood) again.

**€: Okay after some back and forth git checkouts, finally everything is on dev branch including the /opt/pihole scripts. And dev pihole-FTL also works now, perhaps according to the latest commit: https://github.com/pi-hole/FTL/pull/256/files

 🈯️ And yey as my PR (above) got merged, I could directly test pihole -up, with success!**

slim0287, 2018-05-10:
I'm sorry this thread is a little confusing for me to follow with regard to what it has fixed.  I'm running Pi-Hole with DietPi on my Raspberry Pi 2.  When I update Pi-hole, I've learned I need to follow the process I wrote on the Pi-Hole discourse page (https://discourse.pi-hole.net/t/on-update-error-with-local-repository/6283/3), which includes the following four steps: (a) performing a `git reset --hard` in `/var/www/html/admin` and `/etc/.pihole`, where `git status` shows it's necessary; (b) copying `/etc/lightpd/lighttpd.conf` to a backup file;  (c) executing `pihole -up`; (d) restoring `lighttpd.conf` from the backup file.  

What did v6.7 of DietPi fix with respect to the four steps above?  (b) and (d)?  Or the whole shot?  Or none of the above?

Thanks
Slimko

MichaIng, 2018-05-10:
@slim0287 
Hehe, sorry for confusion, jep indeed this thread has grown non-linearly due to some parallel appearing issues/tasks.

So far nothing changed about `pihole -up`.
We need to wait for the next PiHole release which contains a fix then. Also for existing installations we need to add/change a setup variable from our side, before `pihole-up` works without manual steps.

I think we will release a DietPi update shortly after PiHole got updated with Fourdees addition, to do the mentioned adjustment from our side and interactively inform user about this via whiptail dialog.

So stay tuned ;).

slim0287, 2018-06-08:
Does V6.9 fix the Pihole manual update process?  Or are we still waiting on that correction?

Thanks

MichaIng, 2018-06-08:
@slim0287 
We are still waiting for the next Pi-hole release, which should implement our solution. The PR (see above) is already merged into development branch and this means it will be implemented into next release, AFAIK.
Last release was in beginning of April, so according to usual release schedule it be soon. But no guarantee, especially since this release should include the new FTLDNS (merge of dnsmasq into pihole-FTL), which appears to be a core change and could take more testing time than usual.

It is always possible to remove and reinstall Pi-hole via dietpi-software to have latest release, but of course not to nice solution 😉.

slim0287, 2018-06-09:
@MichaIng 
Admittedly, the manual process for updating Pihole takes me perhaps 2 minutes if I do it leisurely so I shouldn't complain.  But given we live in a world now where I literally can talk to the lightbulbs in my house without me being a crazy person and they actually respond..... well, I can't help but ask to save those two minutes....  😉

MichaIng, 2018-06-09:
@slim0287 
Jep, also our interest to save those 2 minutes for all our users 😃. Especially thinks should work as expected. If users get known with `pihole -up` then it should also do it's job on our system.

But some software titles tend to install themselves a bid intrusive. Let's say in some cases the intention of the developers is to turn a system into in this case a "network adblock device", where our intention is to enable certain different features in parallel on one modular device. But yeah it's worth to working on this with Pi-hole, as the current v6.9 statistics show that it's the most installed software title on DietPi systems, besides dependencies like webserver, database and SSH server of course.

unclehowell, 2018-06-10:
I'm following this also. I'm trying to make a user friendly OS for the RPI (with web UI) and autonomous pihole install is important to my project.

dpantel, 2018-06-22:
For those of us who unwittingly ran pihole -up, what is the best method of restoring the dietpi settings (without resetting pihole settings)?

slim0287, 2018-06-22:
@dpantel I hope someone has a better answer that I do:  when I made this error, I ended up reinstalling everything.  Operating system and Pihole.

MichaIng, 2018-06-22:
@dpantel 
You will definitely need some manual work, depending on what webserver etc, you installed.
Can you list all software titles, that you installed via `dietpi-software`, before applying `pihole -up`?

dpantel, 2018-06-23:
just the dietpi base and pihole (lighttpd)

MichaIng, 2018-06-27:
@dpantel 
Sorry for the delay.
> just the dietpi base and pihole (lighttpd)

Ah this makes it easy, just restore the previous `/etc/lighttpd/lighttpd.conf`:
```
cat << _EOF_ > /etc/lighttpd/lighttpd.conf
server.modules = (
	"mod_access",
	"mod_alias",
	"mod_compress",
 	"mod_redirect",
)

server.document-root        = "/var/www"
server.upload-dirs          = ( "/var/cache/lighttpd/uploads" )
server.errorlog             = "/var/log/lighttpd/error.log"
server.pid-file             = "/var/run/lighttpd.pid"
server.username             = "www-data"
server.groupname            = "www-data"
server.port                 = 80

index-file.names            = ( "index.php", "index.html", "index.lighttpd.html" )
url.access-deny             = ( "~", ".inc" )
static-file.exclude-extensions = ( ".php", ".pl", ".fcgi" )

compress.cache-dir          = "/var/cache/lighttpd/compress/"
compress.filetype           = ( "application/javascript", "text/css", "text/html", "text/plain" )

# default listening port for IPv6 falls back to the IPv4 port
include_shell "/usr/share/lighttpd/use-ipv6.pl " + server.port
include_shell "/usr/share/lighttpd/create-mime.assign.pl"
include_shell "/usr/share/lighttpd/include-conf-enabled.pl"
_EOF_
systemctl restart lighttpd
```

This should be it, but please report if you face further issues, it also depends a bid on used update branch, initial version etc.

Fourdee, 2018-06-30:
Waiting for user reply = Waiting for Pi-Hole next release 😃

dpantel, 2018-06-30:
@MichaIng 
Sorry, I missed your reply.  I have already saved the pihole config and reinstall pihole with dietpi.

@Fourdee 
exactly :)

1activegeek, 2018-07-12:
@Fourdee don't know if this is specifically related - but I found an issue with logs currently from piHoles extra logging with dnsmasq. I know I obviously mentioned this on other thread - just coming here to keep in the loop. I had opened an issue with PH guys who helped me identify something that was majorly problematic it would seem. My /var/log directory has filled up completely 100%. 

See here: https://github.com/pi-hole/pi-hole/issues/2270#issuecomment-404602554

Is there something I should be concerned about here? I have a feeling this could be causing issues among other things running that I might not be aware of since even the syslog has no new entries since midnight last night.

1activegeek, 2018-08-08:
I'm assuming you guys already know FTLDNS was released officially ... but if not, PING!! Happy to test out the updated changes when its time! 😄

MichaIng, 2018-08-08:
@1activegeek @Fourdee 
Nicce, so we can finally test and finish this.

FTLDNS btw. is already supported with our current implementation and `pihole -up` can be used (at least it should, needs testing again as said), after adding `INSTALL_WEB_SERVER=false` to `/etc/pihole/setupVars.conf`, if it was not already chosen during initial Pi-hole install.

MichaIng, 2018-08-11:
PR up: https://github.com/Fourdee/DietPi/pull/2003
@Fourdee  I just added the Pi-hol testing branch you created as a start.

MichaIng, 2018-08-11:
@Fourdee 
The update vars indeed needs to exist: https://github.com/pi-hole/pi-hole/blob/master/automated%20install/basic-install.sh#L2343-L2355
`$useUpdateVars` is then used as interactive/non-interactive flag.

Even creating this with reasonable default values should leave to broken install. Some steps are intended to be done interactively one time: https://github.com/pi-hole/pi-hole/blob/master/automated%20install/basic-install.sh#L2377-L2396

I think for now, we can't make it non-interactive 🤔. We would need to do all the above steps within install script and create setupVars file, but Pi-hole updates could then always break our install and some settings are hard to choose anyway.

Fourdee, 2018-08-11:
@MichaIng 

> I think for now, we can't make it non-interactive 🤔. We would need to do all the above steps within install script and create setupVars file, but Pi-hole updates could then always break our install script and some settings are hard to choose anyway.

Yep agree 👍

MichaIng, 2018-08-13:
@Fourdee 
For blocking page, we could implement some of Pi-hole specific Lighttpd settings, but into separate config file of course: https://github.com/pi-hole/pi-hole/blob/master/advanced/lighttpd.conf.debian

`server.error-handler-404    = "pihole/index.php"` need to be `"html/pihole/index.php"` in our case.

The /admin/ specific config could/should be added as well, `"mod_setenv",` needed then.

But when I access a non-allowed web page, I still get redirected to the default index.php (webroot) via 200 (no 404). 

Checked blocking page for relative paths: https://github.com/pi-hole/pi-hole/blob/master/advanced/index.php#L21
But landing.php is never created, not within /var/www/html nor /var/www (webroot).

Not sure yet how blocking page redirection really works then. Needs some work to fully implement it with all webservers. Could try to just move it to webroot (including .css file).
VM just broke down 🤣, will go on later.

Fourdee, 2018-08-13:
Ok, I'll create a new ticket for adding blocking page support, will take some time and is holding v6.13 release up.

I'll try and resolve the service not starting during boot.

Fourdee, 2018-08-13:
Completed (for now).

closed 2018-08-13 by Fourdee