MichaIng/DietPi #1698: Odroid C2 / Kodi crash

dnalor, 2018-04-12:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version ? 6
- SBC Device (EG: RPi 3)? Odroid C2
- Power supply used (EG: 5v 1A RAVpower)? Standard
- SD card used (EG: Sandisk ultra)? emmc
- Distro (EG: Jessie) | ```uname -a```? Linux DietPi 3.14.79+

#### Additional Information (if applicable):
- Software title? Kodi
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Expected behaviour:
Kodi starts

#### Actual behaviour:
Kodi crashes

#### Steps to reproduce:
start Kodi 

#### Did you submit a ```dietpi-bugreport```?
No

#### Extra details:
Symptoms:
Kodi crashes immediately after starting it (both with startkodi and automatic start)
Issue looks similar to #399

Error message:
```
/usr/local/bin/kodi: 18: /usr/local/bin/kodi: pulseaudio: not found
/usr/local/lib/kodi/kodi-xrandr: Failed to get size of gamma for output default
*** Error in `/usr/local/lib/kodi/kodi.bin': munmap_chunk(): invalid pointer: 0x0000000002b78a28 ***
```
General information:
- New installation (DietPi v6.0, Linux DietPi 3.14.79+), up to date according to DietPi-Update
- Only basic config (timezone, locale, network)
- Kodi is the only additional package installed
- Also tested with LibreELEC Kodi, which runs ok, so HW issue unlikely

Logs
- nothing in Xorg.log
- kernel (note: 1m sleep added to startkodi):
```
Apr 12 19:08:15 DietPi kernel: kodi.bin[2787]: unhandled level 1 translation fault (11) at 0xfffffffffffff8, esr 0x92000005
Apr 12 19:08:15 DietPi kernel: pgd = ffffffc0594aa000
Apr 12 19:08:15 DietPi kernel: [fffffffffffff8] *pgd=00000000596d0003, *pmd=0000000000000000
Apr 12 19:08:15 DietPi kernel: 
Apr 12 19:08:15 DietPi kernel: CPU: 1 PID: 2787 Comm: kodi.bin Not tainted 3.14.79+ #1
Apr 12 19:08:15 DietPi kernel: task: ffffffc059422000 ti: ffffffc059828000 task.ti: ffffffc059828000
Apr 12 19:08:15 DietPi kernel: PC is at 0x115001c
Apr 12 19:08:15 DietPi kernel: LR is at 0x114ffd4
Apr 12 19:08:15 DietPi kernel: pc : [<000000000115001c>] lr : [<000000000114ffd4>] pstate: a0000000
Apr 12 19:08:15 DietPi kernel: sp : 0000007fc21aa430
Apr 12 19:08:15 DietPi kernel: x29: 0000007fc21aa430 x28: 000000000229c0c0 
Apr 12 19:08:15 DietPi kernel: x27: 000000000229c060 x26: 0000000002215000 
Apr 12 19:08:15 DietPi kernel: x25: 000000000229c088 x24: 0000000000000000 
Apr 12 19:08:15 DietPi kernel: x23: 0000000002bd8648 x22: 0000000002bd8600 
Apr 12 19:08:15 DietPi kernel: x21: 0000000002210000 x20: 0000000002b44fe8 
Apr 12 19:08:15 DietPi kernel: x19: 0000000002b44fd8 x18: 0000007fc21aa270 
Apr 12 19:08:15 DietPi kernel: x17: 0000007f8049ede8 x16: 000000000221d9d0 
Apr 12 19:08:15 DietPi kernel: x15: 0000007f7ee0d588 x14: 0000007f7f715000 
Apr 12 19:08:15 DietPi kernel: x13: ff00000000000000 x12: 0000000000000000 
Apr 12 19:08:15 DietPi kernel: x11: 0000000000000038 x10: 0000000000000000 
Apr 12 19:08:15 DietPi kernel: x9 : 1999999999999999 x8 : 0000000000000000 
Apr 12 19:08:15 DietPi kernel: x7 : ffffffffffffffff x6 : 00000000000000ff 
Apr 12 19:08:15 DietPi kernel: x5 : 0000000000000000 x4 : 0000007f7ef4b150 
Apr 12 19:08:15 DietPi kernel: x3 : ffffffffffffffe8 x2 : fffffffffffffff8 
Apr 12 19:08:15 DietPi kernel: x1 : 00000000ffffffff x0 : 000000007ee0b5c0 
Apr 12 19:08:15 DietPi kernel: 
Apr 12 19:09:15 DietPi kernel: fb: osd[0] canvas.idx =0x40
Apr 12 19:09:15 DietPi kernel: fb: osd[0] canvas.addr=0x78000000
Apr 12 19:09:15 DietPi kernel: fb: osd[0] canvas.width=7680
Apr 12 19:09:15 DietPi kernel: fb: osd[0] canvas.height=2160
Apr 12 19:09:36 DietPi kernel: fb: osd[0] canvas.idx =0x40
Apr 12 19:09:36 DietPi kernel: fb: osd[0] canvas.addr=0x78000000
Apr 12 19:09:36 DietPi kernel: fb: osd[0] canvas.width=7680
Apr 12 19:09:36 DietPi kernel: fb: osd[0] canvas.height=1080
Apr 12 19:09:36 DietPi kernel: BUG: Bad page state in process Xorg  pfn:5b5c8
Apr 12 19:09:36 DietPi kernel: page:ffffff80419c2040 count:-1 mapcount:0 mapping:          (null) index:0x0
Apr 12 19:09:36 DietPi kernel: page flags: 0x0()
Apr 12 19:09:36 DietPi kernel: page dumped because: nonzero _count
Apr 12 19:09:36 DietPi kernel: Modules linked in: cfg80211 8192cu spidev spi_gpio spi_bitbang w1_gpio wire meson_gpiomem gxbb_wdt fuse ipv6 autofs4
Apr 12 19:09:36 DietPi kernel: CPU: 2 PID: 2823 Comm: Xorg Not tainted 3.14.79+ #1
Apr 12 19:09:36 DietPi kernel: Call trace:
Apr 12 19:09:36 DietPi kernel: [<ffffffc001088fac>] dump_backtrace+0x0/0x144
Apr 12 19:09:36 DietPi kernel: [<ffffffc001089110>] show_stack+0x20/0x28
Apr 12 19:09:36 DietPi kernel: [<ffffffc00183b930>] dump_stack+0x80/0xa4
Apr 12 19:09:36 DietPi kernel: [<ffffffc00117f928>] bad_page+0xe8/0x130
Apr 12 19:09:36 DietPi kernel: [<ffffffc00117fe44>] get_page_from_freelist+0x304/0x7f4
Apr 12 19:09:36 DietPi kernel: [<ffffffc001181338>] __alloc_pages_nodemask+0x164/0x9dc
Apr 12 19:09:36 DietPi kernel: [<ffffffc0011c95a4>] new_slab+0x90/0x318
Apr 12 19:09:36 DietPi kernel: [<ffffffc0011cbab4>] __slab_alloc.isra.57.constprop.63+0x294/0x350
Apr 12 19:09:36 DietPi kernel: [<ffffffc0011cbf64>] kmem_cache_alloc+0x140/0x250
Apr 12 19:09:36 DietPi kernel: [<ffffffc0010ac1e8>] copy_process.part.43+0x1178/0x1508
Apr 12 19:09:36 DietPi kernel: [<ffffffc0010ac754>] do_fork+0xd8/0x3dc
Apr 12 19:09:36 DietPi kernel: [<ffffffc0010acb48>] SyS_clone+0x40/0x4c
```

k-plan, 2018-04-12:
@dnalor 

> General information:
>
> - New installation (DietPi v6.0, Linux DietPi 3.14.79+), up to date according to DietPi-Update

Can you please provide output of:

`root@Odroid-C2-Stretch:~# cat /etc/os-release`

dnalor, 2018-04-12:
@k-plan 
```
root@DietPi:~#  cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 8 (jessie)"
NAME="Debian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=debian
HOME_URL="http://www.debian.org/"
SUPPORT_URL="http://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

Fourdee, 2018-04-13:
@dnalor 

Thanks for the report, we'll try to replicate.

dnalor, 2018-04-13:
@Fourdee 
> Did you install Kodi from dietpi-software? Please confirm kodi-odroid installed:
> dpkg -l | grep kodi
```
root@DietPi:~#  dpkg -l|grep kodi
ii  kodi-odroid                    16.1-6+deb8                             arm64        Kodi - award-winning free and open source (GPL) software media player and entertainment hub (ODROID build)
```
> DietPi uses 720p by default, try changing to 1080p in dietpi-config, then retest.

Tried both, no difference (C2 is connected to a computer monitor)
Here's the relevant xorg.log excerpt:
```
[    35.615] (WW) Falling back to old probe method for fbturbo
[    35.615] (II) Loading sub module "fbdevhw"
[    35.615] (II) LoadModule: "fbdevhw"
[    35.615] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[    35.616] (II) Module fbdevhw: vendor="X.Org Foundation"
[    35.616]    compiled for 1.16.4, module version = 0.0.2
[    35.616]    ABI class: X.Org Video Driver, version 18.0
[    35.617] (II) FBTURBO(0): using /dev/fb0
[    35.617] (WW) VGA arbiter: cannot open kernel arbiter, no multi-card support
[    35.617] (II) FBTURBO(0): Creating default Display subsection in Screen section
        "Default Screen Section" for depth/fbbpp 24/32
[    35.617] (==) FBTURBO(0): Depth 24, (==) framebuffer bpp 32
[    35.617] (==) FBTURBO(0): RGB weight 888
[    35.617] (==) FBTURBO(0): Default visual is TrueColor
[    35.617] (==) FBTURBO(0): Using gamma correction (1.0, 1.0, 1.0)
[    35.617] (II) FBTURBO(0): hardware: OSD FB (video memory: 98304kB)
[    35.617] (**) FBTURBO(0): Option "fbdev" "/dev/fb0"
[    35.617] (**) FBTURBO(0): Option "SwapbuffersWait" "true"
[    35.617] (II) FBTURBO(0): processor: ARM Cortex-A53
[    35.617] (II) FBTURBO(0): checking modes against framebuffer device...
[    35.617] (II) FBTURBO(0): checking modes against monitor...
[    35.617] (--) FBTURBO(0): Virtual size is 1920x1080 (pitch 1920)
[    35.617] (**) FBTURBO(0):  Built-in mode "current"
[    35.617] (==) FBTURBO(0): DPI set to (96, 96)
[    35.617] (II) Loading sub module "fb"
[    35.617] (II) LoadModule: "fb"
[    35.618] (II) Loading /usr/lib/xorg/modules/libfb.so
[    35.620] (II) Module fb: vendor="X.Org Foundation"
[    35.620]    compiled for 1.16.4, module version = 1.0.0
[    35.620]    ABI class: X.Org ANSI C Emulation, version 0.4
[    35.620] (==) Depth 24 pixmap format is 32 bpp
[    35.641] (II) FBTURBO(0): using backing store heuristics
[    35.641] (II) FBTURBO(0): enabled VFP/NEON optimizations
[    35.641] (==) FBTURBO(0): Backing store enabled
[    35.643] (==) FBTURBO(0): DPMS enabled
[    35.656] (II) Loading sub module "dri2"
[    35.656] (II) LoadModule: "dri2"
[    35.656] (II) Module "dri2" already built-in
[    35.657] (II) FBTURBO(0): display controller hardware overlays are unsupported on meson8/gxbb.
[    35.657] (II) FBTURBO(0): Wait on SwapBuffers? enabled
[    35.657] (II) FBTURBO(0): [DRI2] Setup complete
[    35.657] (II) FBTURBO(0): [DRI2]   DRI driver: lima
[    35.657] (II) FBTURBO(0): using DRI2 integration for Mali GPU (UMP buffers)
[    35.657] (II) FBTURBO(0): Mali binary drivers can only accelerate EGL/GLES
[    35.657] (II) FBTURBO(0): so AIGLX/GLX is expected to fail or fallback to software
[    35.657] (==) RandR enabled
[    35.676] (II) SELinux: Disabled on system
[    35.679] (EE) AIGLX error: dlopen of /usr/lib/aarch64-linux-gnu/dri/lima_dri.so failed (/usr/lib/aarch64-linux-gnu/dri/lima_dri.so: cannot open shared object file$
[    35.679] (EE) AIGLX: reverting to software rendering
[    35.710] (II) AIGLX: Loaded and initialized swrast
[    35.710] (II) GLX: Initialized DRISWRAST GL provider for screen 0
```

SuBLiNeR, 2018-04-13:
@dnalor Tried to open an issue at the Kodi-Devs? It seems like it is either a problem of KODI itself or your kernel which are you using isn't supported.

See lines where it tries to load modules... and fails somehow.... that's what i see so far.

On several pages, they suggest to use LibreELEC for KODI on C2 (e.g. https://kodi.wiki/view/ODROID) 

@Fourdee Nothing against DietPi, as it fits almost everything to the user's xp. But myself think at this point sometimes ppl have to use another solution and not holding on something which can't be easy solved.

Fourdee, 2018-04-14:
Appears fine, same results here running GLES under Mali:
```
[    35.657] (II) FBTURBO(0): using DRI2 integration for Mali GPU (UMP buffers)
[    35.657] (II) FBTURBO(0): Mali binary drivers can only accelerate EGL/GLES
[    35.657] (II) FBTURBO(0): so AIGLX/GLX is expected to fail or fallback to software
```

Fourdee, 2018-04-24:
Marking as closed due to:
- Unable to replicate on fresh install
- Possible hardware issue, please verify official PSU and well branded SD/EMMC used.
- Possible incompatible TV in some form

closed 2018-04-24 by Fourdee