MichaIng/DietPi #1701: DietPi-Software | MycroftAI

LexiconCode, 2018-04-13:
MycroftAI

Mycroft is a open source voice assistant. It can run anywhere – on a desktop computer, single board computers like the Raspberry Pi. It is open so it can be remixed, extended, improved. It can be used in anything from a science project to an enterprise software application.
[Overview](https://mycroft.ai/)
[Documentation](https://github.com/MycroftAI/mycroft-core/blob/dev/README.md)

SuBLiNeR, 2018-04-13:
@LexiconCode This sound really nice, but as i rode about it, i'm not really aware, that it will run on every SBC as of now. They mention it only as a "ready to install" for RPi 3 and upper.

If you want it to run on other devices you've get to go the other way.... -> https://github.com/MycroftAI/mycroft-core/blob/dev/README.md

If it is supposed to run on every ever released device, the question should be than, why it isn't tested for more than RPi3 upper..... yeah, all may support "Debian Stretch", but which kernel at least and which modules does error if installed? 

Did you installed 'MycroftAI' on other devices than Raspberry? 

Not to be toxic but, if you want this, you may install it first and than open an issue which we all could get an efford at least, depending on its "errors"....

I'm sure lot of ppl will be impressed if 'DietPi' could solve the problem and implement it somehow, aren't they?

LexiconCode, 2018-04-30:
Ubuntu 16.04. [Odroid C2, C1, XU4](https://forum.odroid.com/viewtopic.php?t=27661)

MichaIng, 2018-12-26:
Added to FeatHub, feel free to vote for it: https://feathub.com/Fourdee/DietPi/+5

MichaIng, 2019-01-23:
Who the hell pushed this FeatHub request so hard the last days/week 😄? Although cheating should be prevented via GitHub account login, nearly 20 votes in this short time makes me think 🤔 😄.

However, maybe Mycroft devs or community found it and voted all together. I already had a look into it:
- Install looks pretty easy via install script: https://github.com/MycroftAI/mycroft-core/blob/dev/dev_setup.sh
- It installs the environment in `python-virtualenv`, isolated/secured but slow... See our HA install which does the same. Every reinstall takes up to 40 minutes just to (re)install python into the virtualenv. However testing, before rating and isolation/security/compatibility vs performance trade.
- Installs into `/opt/mycroft`, log to `/var/log/mycroft/` etc. If we want, we could move it via symlinks into dietpi_userdata to reduce SDcard I/O.
- The install script contains interactive config steps, so this is nothing we can/should work around without having much maintenance trouble. If we find something urgent that could be optimized or made more flexible (via env variables and/or command arguments), we can still commit to the script directly.

- [ ] Next step is test install on Stretch VM...

MichaIng, 2019-01-23:
@LexiconCode 
You mean someone particular? 😉

Jep interesting project indeed, and spreading better earlier than later gives it better chance to stay against the commercial alternatives in voice assistant market.

LexiconCode, 2019-01-23:
I couldn't help myself but post on [mycroft forums](https://community.mycroft.ai/t/vote-for-mycroft-integration-into-dietpi-os/5484). So that's where some of the votes come from at least.

Mycroft has been running it on my raspberry pi for some time and has integrations for Home Assistant. It's great to see voice assistant assistant that's built for privacy by design.

MichaIng, 2019-01-23:
> It's great to see voice assistant assistant that's built for privacy by design.

Jep, I totally agree, way better than having one of these commercial devices, where suddenly users find their private conversations stored on an external server, and then reading reports about data abuse, security breaches and such.

LexiconCode, 2019-05-11:
Just checking in @MichaIng as I know you're busy with other issues. Let me know when this is testable. I will gather support from the community abroad and myself to work out the bugs

MichaIng, 2019-05-11:
@LexiconCode 
Thanks for your offer and sorry for the delay. External causes forced us to concentrate on especially two other large topics. With v6.24 this will be implemented definitely 🤞.

MichaIng, 2019-06-18:
Test install passed:
```
root@VM-Stretch:/mnt/dietpi_userdata# G_TREESIZE mycroft-core/
2.5 GB mycroft-core/
2.2 GB mycroft-core/mimic
183.5 MB mycroft-core/.venv
128.2 MB mycroft-core/.git
15.7 MB mycroft-core/mycroft
1.4 MB mycroft-core/test
72.0 KB mycroft-core/doc
48.0 KB mycroft-core/scripts
48.0 KB mycroft-core/bin
24.0 KB mycroft-core/.github
```
- Mimic build memory consumption peaks 2.7G and this can be higher since core count build processes run concurrently.
- Mimic build on VM took ~20 minutes, on SBC this will be much longer.
- Mimic offline is optional to allow offline usage as fallback if device is online, so I think we should skip it on default install.

By default Mycroft asks you to register at `home.mycroft.ai` to pair your device. It can be skipped but for several skills (weather info, Speach-to-Text, Wolfram-Alpha, ...) API keys are required. So you would need to register manually on each provider to get those keys. Mycroft Home manages this internally. However many things can be done OOTB as well:
```
 tell me a joke
 >> All arrays Chuck Norris declares are of infinite size, because Chuck Norris knows no bounds.
```
- Many skills are installed on first start, others need to be installed manually, I guess possibly through Mycroft Home as well.
```
 Search for noodles
 >> Let me look up noodles
 >> Noodles are made from unleavened dough which is rolled flat and cut, or extruded, into one of a variety of
    shapes. Noodle shapes include long, thin strips , helices, tubes, strings, shells, or other shapes.
```
```
 Set an alarm for today 18:00
 >> For what time?
 18:00
 >> For what time?
 >> You can register me by going to Devices, selecting Add Device then use the code the number Four. 'X' as in X-Ray.
    'K' as in Kilogram. 'A' as in Apple. 'W' as in Whiskey. 'V' as in Victor
 6 pm
 >> Alarm set for tomorrow at six p.m., in 22 hours
```
- I didn't get yet how to tell it to not ask for registration for now or e.g. to skip Wolfram Alpha. Many sentences I enter are attempted to resolve via Wolfram Alpha but since no API key is added, it fails.

Ah here, Mycroft Skills Manager: https://mycroft.ai/documentation/msm/
```
source /mnt/dietpi_userdata/mycroft-code/.venv/bin/activate
msm list
msm remove mycroft-pairing
msm install ...
```
- Lol on next start it reinstalls the pairing skill automatically, as well as the Wolfram Alpha fallback skill that I removed as it produces errors in my case 😆.
- However generally works. Will test on real machine with microphone tomorrow.

Next would be a systemd unit to start the background services and an additional `dietpi-autostart` option to start into Mycroft CLI?

Another thing:
- The installer cannot be automated, without pre-creating a setupVar file and doing some steps manually.
- I am thinking about opening a PR to add installer options to pre-answer all questions.
__________
Note to self:
- Test with `python3-setuptools` and `python3-gi` instead of their Python2 counterparts as deps in installer, since Mycroft runs with Python3 and this would allow to skip Python2 install completely.
- In case contribute change.

MichaIng, 2019-06-18:
Procedure:
```
G_AGI git
cd /mnt/dietpi_userdata
git clone -b master --depth 1 https://github.com/MycroftAI/mycroft-core.git
cd mycroft-core
./dev_setup.sh -r -sm # -r=allow to run as root; -sm=skip building Mimic
./start-mycroft.sh debug
```
- Build/install works with 1G RAM
- Initial start requires more, we should assure RAM + swap is at least 2G

Using the above `python3-setuptools` + `python3-gi` instead of `python-setuptools` + `python-gobject-2-dev` works well, although I skipped Mimic build this time. So no Python 2 installed now, Python 3 only.

forslund, 2019-06-18:
If you have a repo I can provide a mimic deb for you so you don't need to build it.

MichaIng, 2019-06-18:
@forslund 
Many thanks for the offer. This is where I don't know enough about Python builds: Would that work on all architectures or would a separate package be required for each architecture? DietPi currently supports ARMv6 (armv6hf as with RPi1/Zero), ARMv7, ARMv8 and x86_64.

We do not have an own APT repo but could host the packages on our server for `dpkg -i` install.

However nothing urgent, since the potential of an AI assistant is limited anyway without network access 😉. Any everyone is free to build Mimic afterwards, when memory (+swap) size and time is there. I would add some instructions to your online docs then.

Btw another question about this. I build Mimic on top of a non-Mimic install by simply re-running the `dev_setup.sh` with `-fm`. While settings, skills etc are preserved, it doubles the other install steps (some like APT installs of course are skipped/finish directly as long as no updates are available). Is there a way to build Mimic only on top of an existing install?

forslund, 2019-06-18:
Well mimic is written in C. I didn't wrongly assumed the Pi in the name meant Raspberry Pi hardware. Sorry for the confuzzlement :)

There is an `install-mimic.sh` script in the `scripts` folder which should be usable

MichaIng, 2019-06-19:
@forslund 
Ah, indeed DietPi started as Raspberry Pi OS at first, but quickly was extended to Odroids, Sparky, NanoPi, Pine64 and x86_64, limited explicit support for Orange Pi, BananaPi and some others SBCs. Basically it can be installed on any Debian based base image with the above architectures. So yeah, then I think the effort is too high for now with Mimic, found `install-mimic.sh`, which makes a manual build easy, if required.
___________
Did test install on real machine with microphone, found few things we need to take care:
- Mycroft expects PulseAudio by default, but on DietPi ALSA is installed only by sound options. I am not keen to change this, as well it generally works, but...
- Our default `/etc/asound.conf` does not seem to include proper microphone config. I am not too experienced with this, but it works when simply moving the conf out which makes ALSA revert to system defaults and the microphone magically works. But of course with multiple sound devices (onboard + USB DAC typically) the config is required, so we need a microphone configuration. `arecord -l/-L` to scan for devices to select and add correctly to `asound.conf` as default recording device. I just need to check out the correct syntax/directives for this 😆.
- Mycroft calls `paplay` to play wav files by default, but this can be changed to ALSA by setting `"play_wav_cmdline": "aplay %1",` in either `/etc/mycroft/mycroft.conf` (is not created by default) or `/root/.mycroft/mycroft.conf`.
- To play `.ogg` files, we need to manually install [vorbis-tools](https://packages.debian.org/stretch/vorbis-tools) for `ogg123` playback binary.

We could further make the systemd unit run as an own user `mycroft`. It might require to be added to the `audio` group, but other skills I tested so far do not require any special permissions.

forslund, 2019-06-19:
Security-wise it's a good idea to run mycroft as a non-root user since skills are python modules allowed to do anything python is allowed to do... It expects that it'll have a $HOME directory available for the user it's running as.

We recommend using the `/etc/mycroft/mycroft.conf` location to setup "platform" or os related settings. like using aplay instead of paplay see [picroft](https://github.com/MycroftAI/enclosure-picroft/blob/stretch/etc/mycroft/mycroft.conf) for example. mycroft doesn't write to this config at all, only reads it while the config under $HOME/.mycroft may be updated from the software.

MichaIng, 2019-06-19:
Would be easiest to set the users home dir simply to the mycroft-core dir, as this makes as well backup etc. easier, same for skills dir. Also since DietPi allows to easily move the special `/mnt/dietpi_userdata/`(where mycroft-core will be created) to an external drive, this would be preferrable to reduce disk I/O on SDcard, which are in use on most SBCs.

Skills dir is already in `.gitignore` (aimed for symlink but includes a dir in case as well), the above would require to add `.mycroft`, no big deal.

@forslund 
Does Mycroft expect the skills to be in /opt or does it look inside the core dir and just follows the symlink in case? In case, we can simply switch symlink and main dir, no big deal.

forslund, 2019-06-19:
You can relocate the main mycroft data dir using the config parameter `"data_dir"`

The skills dir will then be `{data_dir}/skills`...That said the setup scripts (`prepare_msm.sh` for example) may need hand editing to make sure it looks at the correct folder.

MichaIng, 2019-06-19:
Okay I think, since the datadir might contain further dirs/files by time, we should:
- Create `mycroft` user with `/mnt/dietpi_userdata/mycroft-data` being home dir.
- Place `ln -s /mnt/dietpi_userdata/mycroft-data /opt/mycroft` to make installer write skills directly to the right location.
- Set `"data_dir`" to `/mnt/dietpi_userdata/mycroft-data` as well, after install, via `/etc/mycroft/mycroft.conf` (besides other mention DietPi specific needs).

This way variable data and the core install are in separete directories to avoid inteferences, but both easily moveable to external drive via DietPi tools.

MichaIng, 2019-06-22:
PR up: https://github.com/MichaIng/DietPi/pull/2931

MichaIng, 2019-06-27:
@LexiconCode 
How did you configure the microphone (/etc/asound.conf)? Or did you install PulseAudio?

I found the default sample rate (16000) does not work with DietPi default asound.conf, that is set when choosing a sound card with equaliser disabled. What worked was adding the following to mycroft.conf:
```
  "listener": {
    "sample_rate": 48000
  }
```

Enabling equaliser things are broken again with an error in Mycroft audio service.

LexiconCode, 2019-06-29:
> 
> 
> @LexiconCode
> How did you configure the microphone (`/etc/asound.conf`)? Or did you install PulseAudio?
> 
> I found the default sample rate (16000) does not work with DietPi default asound.conf, that is set when choosing a sound card with equaliser disabled. What worked was adding the following to mycroft.conf:
> 
> ```
>   "listener": {
>     "sample_rate": 48000
>   }
> ```
> 
> I am thinking to add this to our `/etc/mycroft/mycroft.conf` which can then be overridden by user in `/mnt/dietpi_userdata/mycroft-data/.mycroft/mycroft.conf` if wanted/required.
> 
> Enabling equaliser broke it again with an error in Mycroft audio service.
> 
> But all this might depend on the hardware, although I tested on all VMs with different emulated sound cards as well as another notebook (no VM). All required 48000 as sample rate when using default `hw:n,m` PCM.
 
I'd look at my current that up and I'll get back to you tomorrow. If memory serves me correct I believe I used PulseAudio.

LexiconCode, 2019-07-03:
My current hardware is a bit unusual for the microphone which means a strange config. I need to use something fully supported like [PlayStation Eye](https://www.amazon.com/PlayStation-Eye-3/dp/B000VTQ3LU?ref_=fsclp_pl_dp_11) via USB versus the add-on board I have now for the Raspberry pi.

closed 2019-07-03 by MichaIng