MichaIng/DietPi #1702: HDD Spindown

bokiroki, 2018-04-13:
Hi, i read that dietpi put external hdd to sleep after 10 minutes.. but my hdd spindown after 30 minutes, i checked several times..
This is my hdd
https://www.wdc.com/products/portable-storage/wd-elements-portable.html#WDBUZG0010BBK-NESN
also i entered this command to see spindown status 
hdparm -C /dev/sda1
and output is strange.. what this mean?
/dev/sda1:
SG_IO: bad/missing sense data, sb[]:  f0 00 01 00 50 40 ff 0a 00 00 00 00 00 1d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 drive state is:  unknown

bokiroki, 2018-04-14:
@Fourdee thanks.. one more question.. is there a way to limit program from usin all cpu power.. so i tried to play a movie trough plex, and i got error message, and raspberry pi become unresponsive... i can't enter ssh and plex web.. yellow led light is turned on.. i think that that movie file i got error is transcoding..but i can't to anything now to stop it..only option is to turn the power off..

Fourdee, 2018-04-14:
@bokiroki 

> i think that that movie file i got error is transcoding.

We highly recommend disabling transcoding on the RPi device and PLEX interface (under settings). CPU transcoding under performs.

Usually we'd mention ```dietpi-process_tool``` to lower the priority of PLEX processes, however, PLEX spawns a new transcoding process on the fly, which we cant background check for. However, their should be some settings in the PLEX web interface to set a lower transcode priority.

closed 2018-04-15 by bokiroki