MichaIng/DietPi #1707: AmiBerry | Update to 2.19

Fourdee, 2018-04-15:
https://github.com/midwan/amiberry/releases/tag/v2.19

k-plan, 2018-04-15:
BTW: 
Perhaps you can as well have a look into this: 
[Amiberry input "leaks" to the console (SDL2 bug?)](http://dietpi.com/phpbb/viewtopic.php?f=11&t=3049)

Fourdee, 2018-04-16:
Binaries/code update, just need to test + replicate https://github.com/Fourdee/DietPi/issues/1707#issuecomment-381403586

Fourdee, 2018-04-18:
> https://github.com/midwan/amiberry/issues/304#issuecomment-382222815

Fourdee, 2018-04-18:
🈯️ http://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Stretch_AmiBerry.7z Updated to include support for RPi 3B+:
https://github.com/midwan/amiberry/issues/304#issuecomment-382222815

Fourdee, 2018-04-18:
Confirmed SDL2 terminal leak.
- confirmed FAST boot quit breaks keyboard input,  term switch required.
- SDL2 version is 2.0.7, should have the reported fix for terminal leak
- ? https://discourse.libsdl.org/t/workaround-rpi2-keystrokes-leaking-into-x11-terminal/21750
- ? https://github.com/inoremap/shut-term-keys
- 🈯️ Unable to replicate illegal instruction on 2.18

Fourdee, 2018-04-24:
🈵 Has no effect:
```
G_AGI build-essential
wget https://github.com/inoremap/shut-term-keys/archive/master.zip 
unzip master.zip
cd shut-term-keys-master
make
```

Fourdee, 2018-04-29:
Resolved standard boot SDL2 keyboard link.
Fast boot is still affected, unable to resolve currently, will mark it as a known issue.

closed 2018-04-29 by Fourdee