MichaIng/DietPi #1710: Not able to update to 6.7

magnusoverli, 2018-04-15:
#### Required Information:
- DietPi Version | v6.6
- SBC Device     | RPi 3 Model B (armv7l) (index=3)
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Update

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Ign:2 https://archive.raspberrypi.org/debian stretch InRelease
Get:3 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Get:4 https://archive.raspberrypi.org/debian stretch Release [24.8 kB]
Ign:5 https://archive.raspberrypi.org/debian stretch Release.gpg
Reading package lists...
E: The repository 'https://archive.raspberrypi.org/debian stretch Release' is no longer signed.
```

MichaIng, 2018-04-15:
@magnusoverli 
Thanks for your report. Just updated my RPi as well an had not issues with exactly the same repository. Perhaps a temporary issue?

Otherwise could you try to clear your APT cache and manually run apt update: `apt clean && apt update`
Possibly even:
```
apt clean
mv /var/lib/apt/lists /tmp
mkdir -p /var/lib/apt/lists/partial
apt update
```

Fourdee, 2018-04-16:
Confirmed working on RPi for me aswell, think @MichaIng is correct with it being  temporary hosting issue.

However, @magnusoverli please let us know if the above solution resolves.

Fourdee, 2018-04-16:
Interesting, seems Debian repo having some issues aswell with some of our users:
https://community.roonlabs.com/t/dietpi-roon-server-with-additional-hdd-ssd/41440/4?u=dan_knight

magnusoverli, 2018-04-16:
Hi! I was able to update this morning. Did a reboot and diet dietpi-update without any errors.

k-plan, 2018-04-16:
@magnusoverli 

Thanks for your response. 👍 

I'll mark this as closed, please reopen if required.

closed 2018-04-16 by k-plan