MichaIng/DietPi #1715: Swapfile on btrfs ?

mgherghi, 2018-04-18:
## Creating a bug report/issue:
Not sure if this is possible, but may i create a swap on my btrfs disk ? 

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

elRadix, 2018-04-18:
maybe this will help or have it integrated with dietpi as an option?
https://github.com/sebastian-philipp/btrfs-swapon

mgherghi, 2018-04-18:
I am having a hard time understanding why i cant swap on btrfs.  Its a fairly modern filesystem for Linux, is the os moving away from swap ?

MichaIng, 2018-04-18:
@mgherghi 
"Linux" is not moving towards btrfs, thus it's not moving away from swap. Just few distributions use it as default, e.g. SUSE Linux and MeGoo, and Fedora has the plan to use it as default, but it did not yet happen since several versions. And if Fedora does not, than this means something 😉.
On the other hand RHEL e.g. want to drop support for btrfs as they (their users) seem to face some critical issues by times.

Overall btrfs still seems to have some issues, that prevent most distributions from using it as default. I am not sure about the performance issues with swap on btrfs, but I think there are good reasons why this is not possible without workarounds. Thus I would still name ext4 as default Linux fs for good reasons and as this is fact for most Linux distros 😉.

@elRadix 
Thanks for the link. This seems quite easy and we could add this to our scripts with reference, no additional packages are required to run the commands. Maybe then at least show some hint to user, that this is "hack" not recommended for performance and stability reasons. Also note the warnings within script readme:

> Keep in mind, that a copy-on-write file system is not the best choice to use a swap file. Don't expect high performance.
> ...
> Don't balance your file system as long as you use this swap file.

Fourdee, 2018-04-19:
@MichaIng 

> this "hack" is not recommended for performance and stability reasons

I'am not personally keen on running a loopback to achieve a swapfile on btrfs. As hacks goes, its long winded and the performance impact negates its purpose.

We do offer multiple filesystem support in DietPi, allowing users to experiment outside of the standard ext4, however, there is always downsides (EG: FAT doesn't support permissions).

Regardless, if our users want the ability to run a swapfile on btrfs, and Sebastian Philipp's workaround offers a solution, we can try to integrate it at the very least. But, this will be a low priority task.

MichaIng, 2018-04-19:
Further reference:
https://btrfs.wiki.kernel.org/index.php/FAQ#Does_btrfs_support_swap_files.3F
https://btrfs.wiki.kernel.org/index.php/Project_ideas#Swap_file_support

leggewie, 2018-08-31:
Why bother?  Anybody who prefers btrfs for the files (like me) should simply set up a separate swap partition.  Anything else is insanity.

MichaIng, 2019-01-15:
@leggewie 
You are absolutely right. We should not enable a feature that is known to be risky and has performance disadvantages. No big deal to setup a separate partition.

This is actually something we could offer instead from within DietPi-Drive_Manager, if a `btrfs` formatted drive/partition is selected: Creating an ext4 formatted swap partition on the same drive.
- Shrinking current partition to free disk space should generally work, however it would be best if the swap partition was the first on drive (relevant for HDDs). Not sure how to achieve that without reformatting the whole drive?

MichaIng, 2019-04-17:
Linux 5.0 supports swap file on BTRFS: https://kernelnewbies.org/LinuxChanges/#Linux_5.0.Btrfs_swap_file_support

Alternative reasonable request: Allow using an existing partition as swap partition. So it's then up to the user to create the partition manually. Our script will just toggle swap on it.

MichaIng, 2019-05-06:
I mark this as closed. Swap partition support is a separate topic and generally not related to BTRFS.

closed 2019-05-06 by MichaIng