MichaIng/DietPi #1716: "fbset" used with RPi only; necessary at all?

MichaIng, 2018-04-18:
I found the command/package `fbset` just used a single time with RPi devices at `DietPi-Boot`, if HDMI is chosen to be disabled: https://github.com/Fourdee/DietPi/blob/testing/dietpi/boot#L269

If didn't oversee something, then all other devices ignore the related `dietpi.txt` setting and use other methods, e.g. `config.txt` for resolution changed, same as RPi for all other choices than `headless`.

At least this means we should be able to install `fbset` only on RPi devices. Possibly we can leave setting framebuffer geometrie to 16 16 16 16 8 to `config.txt` as well and can completely skip `fbset`, or was there some reason to go with the command instead?

Ref: https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

Fourdee, 2018-04-19:
@MichaIng 

> At least this means we should be able to install fbset only on RPi devices. Possibly we can leave setting framebuffer geometrie to 16 16 16 16 8 to config.txt as well and can completely skip fbset, or was there some reason to go with the command instead?

Its been a few years, can't honestly remember why we didn't just config.txt 16x16. I'll try it and send commit.

Fourdee, 2018-04-19:
Yep works fine without ```fbset```. Completed.

MichaIng, 2018-04-19:
@Fourdee 
Nice, that was fast 😃.

There are also max value settings available, but honestly I have no idea what they are for:
```
FRAMEBUFFER_WIDTH
The framebuffer_width command specifies the console framebuffer width in pixels. The default is the display width minus the total horizontal overscan.

FRAMEBUFFER_HEIGHT
The framebuffer_height command specifies the console framebuffer height in pixels. The default is the display height minus the total vertical overscan.

MAX_FRAMEBUFFER_HEIGHT, MAX_FRAMEBUFFER_WIDTH
Specifies the maximum dimensions that the internal frame buffer is allowed to be.

FRAMEBUFFER_DEPTH
Use framebuffer_depth to specify the console framebuffer depth in bits per pixel. The default value is 16.
```
- "console framebuffer" vs "internal framebuffer", or just boot value vs max. value that can be set afterwards (e.g. by plugging another screen?) or are those the related settings for fbset -vxres -vyres virtual resolutions?

Fourdee, 2018-04-20:
> MAX_FRAMEBUFFER_HEIGHT, MAX_FRAMEBUFFER_WIDTH
Specifies the maximum dimensions that the internal frame buffer is allowed to be.

Maybe this changes the output resolution. As ```FRAMEBUFFER_HEIGHT``` still runs at 1080p, regardless of setting.
I'll have to run a test later 👍

k-plan, 2018-04-23:
Hmm ... can be wrong, a long time ago, but it was something with` fbset` and LCD panels support like this:  

[Added support for more LCD panels in v.123](http://dietpi.com/phpbb/viewtopic.php?f=15&t=541)

MichaIng, 2018-04-23:
@k-plan @Fourdee 
Ohhh, important point. The package `fbset` does not only provide the command `fbset` to adjust the framebuffer geometry, but also `con2fbmap` to connect consoles to screens (?). And we use this on enabling/disabling the panels: https://github.com/Fourdee/DietPi/blob/testing/dietpi/func/dietpi-set_hardware#L671
Ref.: http://manpages.ubuntu.com/manpages/trusty/man1/con2fbmap.1.html

The same can be achieved by adding e.g. `fbcon=map:12` to RPi `cmdline.txt` or `boot.ini` bootargs on Odroids, to in this case map tty1 to /dev/fb2. For my impression this is also the preferred way compared to `con2fbmap` somewhere after/during boot. There might be other commands we currently run via systemd service, that would be better placed within kernel command line.
€: Ah this is also used here on disabling panel: https://github.com/Fourdee/DietPi/blob/200da7982072c37d0077b967b5dcaaf3c6ca2ae0/dietpi/func/dietpi-set_hardware#L736

But using this, a reboot is needed. To change current resolution and map consoles without reboot, `fbset` with related commands is needed, as far as I could see. So for my impression for permanent fb settings, we should use kernel/boot arguments, but we may want to provide direct changes? In this case we need to add `fbset` as pre-req to install on demand like `xinput-calibrator`: https://github.com/Fourdee/DietPi/blob/200da7982072c37d0077b967b5dcaaf3c6ca2ae0/dietpi/func/dietpi-set_hardware#L592

MichaIng, 2018-04-26:
Ah add to prereq already there :+1:

closed 2018-04-26 by MichaIng