MichaIng/DietPi #1717: Updating mono stuck

roelfoubert, 2018-04-18:
Today I did apt-get upgrade and it gets stuck setting up mono-roslyn (5.10.1.47-0xamarin4+raspbian9b1).
Is there a solution for this problem?

MichaIng, 2018-04-18:
@roelfoubert 
Hmm, as far as I could see, we do no real adjustment to mono, thus for now I would suggest this being and external issue, maybe temporarily faulty package.

There are no error messages, nothing showing up within `journalctl`?
__________________
I just tested installation on Debian, which worked fine, although `mono-roslyn` setup took longest compared to the other packages:
```
...
Unpacking mono-roslyn (5.10.1.47-0xamarin4+debian9b1) ...
Setting up libmono-system-transactions4.0-cil (5.10.1.47-0xamarin4+debian9b1) ...
Setting up libmono-posix4.0-cil (5.10.1.47-0xamarin4+debian9b1) ...
Setting up libmono-system-io-compression4.0-cil (5.10.1.47-0xamarin4+debian9b1) ...
Setting up libmono-system-numerics4.0-cil (5.10.1.47-0xamarin4+debian9b1) ...
Setting up libmono-system-enterpriseservices4.0-cil (5.10.1.47-0xamarin4+debian9b1) ...
Setting up libmono-xbuild-tasks4.0-cil (5.10.1.47-0xamarin4+debian9b1) ...
Setting up libmono-system-servicemodel-internals0.0-cil (5.10.1.47-0xamarin4+debian9b1) ...
Setting up libmono-microsoft-build-framework4.0-cil (5.10.1.47-0xamarin4+debian9b1) ...
Setting up libmono-system-core4.0-cil (5.10.1.47-0xamarin4+debian9b1) ...
Setting up libmono-system-runtime-serialization4.0-cil (5.10.1.47-0xamarin4+debian9b1) ...
Setting up libmono-microsoft-build-utilities-v4.0-4.0-cil (5.10.1.47-0xamarin4+debian9b1) ...
Setting up libmono-system-data4.0-cil (5.10.1.47-0xamarin4+debian9b1) ...
Setting up libmono-system-xml-linq4.0-cil (5.10.1.47-0xamarin4+debian9b1) ...
Setting up libmono-microsoft-build-engine4.0-cil (5.10.1.47-0xamarin4+debian9b1) ...
Setting up libmono-microsoft-build-tasks-v4.0-4.0-cil (5.10.1.47-0xamarin4+debian9b1) ...
Setting up mono-xbuild (5.10.1.47-0xamarin4+debian9b1) ...
Setting up mono-roslyn (5.10.1.47-0xamarin4+debian9b1) ...

```
Sadly I have no testing RPi to try raspbianstretch repo.

Fourdee, 2018-04-19:
Fresh install confirmed functional on RPi:
```
root@DietPi:~# dpkg -l | grep roslyn
ii  mono-roslyn                                                 5.10.1.47-0xamarin4+raspbian9b1 all          Microsoft C# compiler
```

Could be an issue with the repo, its patching, HW instability. Many factors.

@roelfoubert 

Please try reinstalling Mono:
```
G_AGP mono-*
G_AGA
G_AGI mono-runtime
```

And depending on the software title that uses mono, you may want to install these aswell, just to be covered:
```
G_AGI mono-devel mediainfo libmono-cil-dev
```