MichaIng/DietPi #1719: "tvservice" and "vcgencmd" | $PATH and $LD_LIBRARY_PATH

MichaIng, 2018-04-19:
A while ago I recognized that `DietPi-Boot` produces an error massage when trying to execute `tvservice -o` and me being unable to execute `vcgencmd` to get settings etc.: "command not found"
- So I added full path `/opt/vc/bin/vcgencmd|tvservice` and ran into next error: "library not found"
- So I added `LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/opt/vc/lib /opt/vc/bin/vcgencmd|tvservice` which worked then.

Since we use both commands within DietPi scripts, I guess this has something to do with me purging all `raspberrypi-firmware` (kernel, bootloader, vc library and libaries) packages, relying completely on `rpi-update`, which generally works fine. By this I have some additional free space, since documentation and stuff is skipped, compared with our solution to just set those packages on hold, when using `G_RPI_UPDATE`. But there seem to be other, more important things, missing as well 😅.
- I guess the packages also add vc library and binaries to PATH/LD_LIBRARY_PATH, or add related symlinks for the binaries. Could someone verify this: `echo $PATH` and/or check bin/sbin folders, how those binaries are added to shell?
- I learned that `LD_LIBRARY_PATH` should/cannot be set just as exported variable. Also it is empty for me, when I try echo, even that default libc and armhf libraries are obviously inside. The method to add those to system, is adding them via `/etc/ld.so.conf.d/`, where also the ones for libc and armhf are found. Could someone check if on RPi with firmware packages installed, there is an additional entry for RPi/videocore: `ls -al /etc/ld.so.conf.d/`?

I use bashrc/dietpi-globals to add aliases for the needed commands, but this excludes all the other binaries and aliases are not available from within scripts (needs additional edit there), so would be good to know, how this is officially implemented:
```
[ -f /opt/vc/bin/vcgencmd ] && alias vcgencmd="LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/opt/vc/lib /opt/vc/bin/vcgencmd"
[ -f /opt/vc/bin/tvservice ] && alias tvservice="LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/opt/vc/lib /opt/vc/bin/tvservice"
```

Fourdee, 2018-04-20:
Fresh RPi install:
```
root@DietPi:~# vcgencmd
root@DietPi:~# tvservice -o
Powering off HDMI
root@DietPi:~# tvservice -oecho $PATH^C
root@DietPi:~# echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
root@DietPi:~# echo $LD_LIBRARY_PATH

root@DietPi:~# which vcgencmd
/usr/bin/vcgencmd
root@DietPi:~# which tvservice
/usr/bin/tvservice
```

```libraspberrypi-bin``` is the package which contains these programs:
```
The following packages will be REMOVED:
  libraspberrypi-bin*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 1,264 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 16943 files and directories currently installed.)
Removing libraspberrypi-bin (1.20180417-1) ...
root@DietPi:~# which tvservice
root@DietPi:~# which vcgencmd
root@DietPi:~#
```

MichaIng, 2018-04-20:
@Fourdee 
Ahh, the APT package puts them into /usr/bin/, okay that explains it. Are those symlinks to /opt/vc/bin? Otherwise updated binaries by rpi-update would be ignored Oo. Did you check `ls -al /etc/ld.so.conf.d/` if there is an entry to add the library?

If everything is as above (library link + symlinks), then I think we do not need to adjust anything and I would leave the packages in place. There might be other things missing when just using rpi-update. Maybe I find some time to analyse those more closely or open issue at rpi-update for clarification.

Fourdee, 2018-04-21:
@MichaIng 

Seems its a direct symlink, interesting:
```
root@DietPi:~# ls -lha /usr/bin/vcgencmd
lrwxrwxrwx 1 root root 20 Apr 17 13:07 /usr/bin/vcgencmd -> /opt/vc/bin/vcgencmd

root@DietPi:~# ls -lha /usr/bin/tvservice
lrwxrwxrwx 1 root root 21 Apr 17 13:07 /usr/bin/tvservice -> /opt/vc/bin/tvservice
```

MichaIng, 2018-04-21:
@Fourdee 
Nice, thanks for checking this. So it's handled more less as expected and for my impression this is also the method that makes most sense: I guess all binaries from within `/opt/vc/bin` are linked then to allow global access.

I am already asking at an open issue on rpi-update if other important things are done by the APT packages that need to be done manually when just using rpi-update (and purging APT packages). I will experiment with this, but as said: On DietPi we should stay with the packages to be on the secure side. There are other ways to prevent most man/info pages from being installed 😉.