MichaIng/DietPi #1720: CPU governor setting defaults back to performance after reboot (Allo Sparky)

renebouwmeester, 2018-04-19:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version 6.7
- SBC Device Allo Sparky
- Power supply used SBooster 5V/3A
- SD card used (EG: Sandisk ultra)? Sandisk
- Distro: Stretch

#### Additional Information (if applicable):
- Software title? Setting governor via dietpi-config > performance options
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Expected behaviour:
When set to (f.e.) ondemand, it should stick to ondemand after reboot

#### Actual behaviour
Defaults back to perfomance

#### Steps to reproduce:
Change governor via dietpi-config and reboot

#### Additional info:
Changing governor and rebooting on RasPI 3B+ works fine.

renebouwmeester, 2018-04-19:
Weird. It's fine now. Rebooted twice, once removing all power (it's dual powered with a USBridge attached). Setting sticks. Both cpu and cat report ondemand.

Ondemand is stable here (running slightly less hot), but the Sparky/USBridge is only running as audio bridge.

Anyway – false, it appears. I'll keep an eye on it and report back if it happens again. Thanks for your attention and keep up the great work with DietPi!

Fourdee, 2018-04-20:
> We set CPU governor to "performance" on Sparky for stability reasons.

@renebouwmeester 

Yep, this is a recommendation directly from Allo. Although, i've personally never had any issues with ```ondemand``` myself, so should be fine to use it.

MichaIng, 2018-04-20:
@renebouwmeester
Okay great, keep keep is informed if this happens again. I will mark as closed here.

closed 2018-04-20 by MichaIng