MichaIng/DietPi #1722: Backup issue

labibllaca, 2018-04-20:
## Creating a bug report/issue:

#### Details:
- Date           | Fri 20 Apr 04:59:38 BST 2018
- DietPi version | v6.7
- Img creator    | 
- Pre-image      | 
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Distro         | stretch (index=4)
- Command        | ncc maintenance:mode --on
- Exit code      | 1
- Software title | DietPi-Services

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional Information:

#### Additional logs:
```
Log file contents:
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /var/www/nextcloud/lib/private/DB/Connection.php(151): Doctrine\DBAL\Connection->setTransactionIsolation(2)
#5 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(172): OC\DB\Connection->__construct(Array, Object(Doctrine\DBAL\Driver\PDOMySql\Driver), Object(Doctrine\DBAL\Configuration), Object(Doctrine\CommonventManager))
#6 /var/www/nextcloud/lib/private/DB/ConnectionFactory.php(152): Doctrine\DBAL\DriverManager::getConnection(Array, Object(Doctrine\DBAL\Configuration), Object(Doctrine\CommonventManager))
#7 /var/www/nextcloud/lib/private/Server.php(628): OC\DB\ConnectionFactory->getConnection('mysql', Array)
#8 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#9 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OCP\IDBConnecti...')
#10 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\IDBConnecti...')
#11 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(164): OC\ServerContainer->query('OCP\IDBConnecti...')
#12 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#13 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('DatabaseConnect...')
#14 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('DatabaseConnect...')
#15 /var/www/nextcloud/lib/private/Server.php(1511): OC\ServerContainer->query('DatabaseConnect...')
#16 /var/www/nextcloud/lib/private/Server.php(332): OC\Server->getDatabaseConnection()
#17 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#18 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OC\Authenticati...')
#19 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\Authenticati...')
#20 /var/www/nextcloud/lib/private/Server.php(336): OC\ServerContainer->query('OC\Authenticati...')
#21 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#22 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OC\Authenticati...')
#23 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\Authenticati...')
#24 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(164): OC\ServerContainer->query('OC\Authenticati...')
#25 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#26 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OC\Authenticati...')
#27 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\Authenticati...')
#28 /var/www/nextcloud/lib/private/Server.php(352): OC\ServerContainer->query('OC\Authenticati...')
#29 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#30 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OCP\IUserSessio...')
#31 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\IUserSessio...')
#32 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(164): OC\ServerContainer->query('OCP\IUserSessio...')
#33 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#34 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('UserSession')
#35 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('UserSession')
#36 /var/www/nextcloud/lib/private/Server.php(1380): OC\ServerContainer->query('UserSession')
#37 /var/www/nextcloud/lib/private/Server.php(688): OC\Server->getUserSession()
#38 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#39 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OC\App\AppManag...')
#40 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\App\AppManag...')
#41 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(164): OC\ServerContainer->query('OC\App\AppManag...')
#42 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#43 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('AppManager')
#44 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('AppManager')
#45 /var/www/nextcloud/lib/private/Server.php(1684): OC\ServerContainer->query('AppManager')
#46 /var/www/nextcloud/lib/private/legacy/app.php(332): OC\Server->getAppManager()
#47 /var/www/nextcloud/lib/private/legacy/app.php(113): OC_App::getEnabledApps()
#48 /var/www/nextcloud/lib/base.php(661): OC_App::loadApps(Array)
#49 /var/www/nextcloud/lib/base.php(1080): OC::init()
#50 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#51 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#52 {main}
```

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Fourdee, 2018-04-20:
@labibllaca 

Many thanks for the report 👍 

@MichaIng Any chance you can pick this one up if you have time?

MichaIng, 2018-04-20:
@labibllaca 
There seems to be an issue with your database/MariaDB. I guess you are not able to access your Nextcloud instance as well?

Please check the following and provide output:
`journalctl -t mysqld`

Also there may be more information helpful:
- Which webserver are you using?
- Was this a fresh install of DietPi+Nextcloud, or did you recover a backup?
- Do you remember any change (dietpi-update, apt update, Nextcloud update) that you did, before the error occurred the first time?

MichaIng, 2018-05-11:
I close due to missing reply. Please reopen if required.

closed 2018-05-11 by MichaIng