MichaIng/DietPi #1724: Owncloud

Piingliv, 2018-04-21:
report/issue: When pulling the own cloud files, i encounter the following error "Job for mysql.service failed because the control process exited with error code. See "systemctl status mysql.service" and "journalctl -xe" for details. 

Required Information:
- DietPi Version | v6.7
- SBC Device  | RPi 3 model B+ (armv71) (index=3)
- SD card used  | sands ultra 32GB
- Distro  | stretch (index4)

Additional Information (if applicable):
- Software title | own cloud
- Exit code | 1
- command systemctl restart mysql

Steps to reproduce:
Start up Dietpi, select configuration 47 to automatic installation/pull for own cloud files.

yumiris, 2018-04-21:
What's the output when you run `systemctl status mysql.service` right after you get the error?

MichaIng, 2018-04-21:
@Piingliv 
Thanks for your report. Jep as the service start/restart of mysql (MariaDB) failed, the suggestion of yumiris would be the first thing to check.

We execute the command right after adding 4-byte support settings to MariaDB: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L8298

I just tested installation on my VM and it worked well. But it's Debian, not Raspbian and there might be some differences.

Some additional information would be helpful:
- Which webserver did you choose?
- Did you recognize any other error/warning during the whole installation process?
- Was this on a fresh DietPi image? If not, was MariaDB installed the first time on this machine? If not, was ownCloud installed the first time?

`journalctl -t mysqld` or even parts of `journalctl` around the time or error occurrence might also give a more complete picture of what was going on.

Fourdee, 2018-04-22:
Fresh install test, RPi 3 + default lighttpd + Owncloud
- 🈯️ No errors, running fine.

Piingliv, 2018-04-22:
Hi, thank you all for your help. @yumiris the output when I run systemctl status mysql.service right after i get the error is as follows. 

```
root@dietpi:# systemctl status mysql.service
mysql.service - LSB: Start and stop the mysql database server deamon
loaded:loaded(/etc/init.d/mysql; generated; vendor preset: enabled)
active failed (Result: exit-code) since Sun 2018-04-22 13:49:59 BST; 53s ago
Docs: man systems-sysv-generator(8)
process: 9027 Execstart=/etc/init.d/mysql start (exitcode=exited, status=1/FAILURE

apr 22 13:49:27 diet mysql[9211] 2018-04-22 13:49:27 1995640832 [note] Recovering after a crash using tc.log
apr 22 13:49:27 diet mysql[9211] 2018-04-22 13:49:27 1995640832 [note] starting crash recovery...
apr 22 13:49:27 diet mysql[9211] 2018-04-22 13:49:27 1995640832 [note] Crash recovery finished.
apr 22 13:49:27 diet mysql[9211] 2018-04-22 13:49:27 1995640832 [note] server socket created on IP: '127.0.0.1'
apr 22 13:49:27 diet mysql[9211] 2018-04-22 13:49:27 1995640832 [ERROR] Fatal error: Can't open and lock privilege tables: Incorrect file format 'user'
apr 22 13:49:27 diet mysql[9211]: starting mariaDB database server mysql......... failed!
apr 22 13:49:27 diet system[1]: mysql.service: control process exited, code=exited status=1
apr 22 13:49:27 diet system[1]:Failed to start LSD: start and stop the mysql database server deamon.
apr 22 13:49:27 diet system[1]:mysql.service: unit entered failed state. 
apr 22 13:49:27 diet system[1]: mysql.service: failed with result 'exit code'_
```

Fourdee, 2018-05-07:
@MichaIng 

I'll mark this as closed due to:
- No response
- Unable to replicate on a fresh install https://github.com/Fourdee/DietPi/issues/1724#issuecomment-383362401

Please reopen if required.

closed 2018-05-07 by Fourdee