MichaIng/DietPi #1727: XRDP not showing Desktop Environment

SimoneBnc, 2018-04-22:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```6.7```?
- SBC Device (RPi 3)?
- Power supply used (5V 2A)?
- SD card used (Samsung 64GB UHS-3)?
- Distro Stretch

#### Additional Information (if applicable):
- Software title?  XRDP
- Can this issue be replicated on a fresh installation of DietPi?
Yes, just install MATE and XRDP 

#### Expected behaviour:

Like in earlier versions, like v150, you could open RDP connection, insert username and password and you had access to desktop
#### Actual behaviour:

On RDP client you are presented with login screen with Xorg as default, you type username and password and the login window disappears leaving the light blu background and nothing happens on the client side but if you look at the raspberry console you see the desktop environment being loaded.
#### Steps to reproduce:

Install dietpi, then MATE (or any other DE) and XRDP, then connect with any RDP client 
#### Did you submit a ```dietpi-bugreport```?

46b57397-f825-41eb-9e75-e1a274a45cc5-1
#### Extra details:

SimoneBnc, 2018-04-24:
Actually I did use Xorg but it is opening the DE on the RPI console, unless there is something different to do.

Fourdee, 2018-04-24:
Confirmed bug:
- 🈵 MATE
- 🈵 LXDE

REF: https://github.com/neutrinolabs/xrdp/issues/731#issuecomment-292903254

Fourdee, 2018-04-24:
Ok, appears its only functional with Xvnc + vncserver 4 installed. 
Could be an external bug with XRDP, limited to Xorg (although, unsure due to lack of any useful logs for debugging).
Either way, we'll make the changes to DietPi, to automatically install VNC4 server along with XRDP.
[image]

### @SimoneBnc Workaround:
Install VNC4 server
```
 dietpi-software install 28
reboot
```
Use any password, not used for XRDP.

Then select Xvnc as connection method

SimoneBnc, 2018-04-24:
> Use any password, not used for XRDP.

Do you mean to create a new user or what?

Ami011995, 2019-08-01:
I have also faced same issue with Cent OS 7-6 1810 (Core). I have installed Xrdp 0.9.7 on CentOS(64 Bit) in GUI mode. i am able to connect RDP through Xvnc but using Xorg is having blue screen problem.

OS: CentOS 7-6 1810
Architecture: 64 Bit
Xrdp version: 0.9.7
RDP Client: Windows 10 (RDP 10.2 Supported)

MichaIng, 2019-08-01:
@Ami011995 
With 0.9.9 its the same.

But to be honest, if you're already install a VNC server, why not using it directly and skip all this RDP hassle? Also it is MUCH faster!! I use the (old) tiny portable TightVNC viewer: https://www.tightvnc.com/download/1.3.10/tightvnc-1.3.10_x86_viewer.zip

But I found a new approach to use the Xorg login option via XRDP:
- https://github.com/neutrinolabs/xorgxrdp
- https://packages.debian.org/buster/xorgxrdp (perhaps CentOS ships is as well)

The Debian package page makes it actually quite clear that these drivers are required for direct X server connection: https://packages.debian.org/buster/xrdp
On Jessie they did not yet exist, so were obviously integrated into the xrdp package (obviously source-wise), which is why this issue here came up with Stretch.
________
But note this is the "DietPi" GitHub site, so you should consult the CentOS community in this.