MichaIng/DietPi #1729: AUTO_SETUP_INSTALL_SOFTWARE_ID=152 - crashes dietpi

unclehowell, 2018-04-25:
## Creating a bug report/issue:

when you install software ID 152 with dietpi.txt it works fine.
But try to run the install using a command on a running system (dietpi-software install) and it freezes

DietPi Version | 6.7 
RPi 3
5v 2a PSU
cheap & nasty 16GB SD Card 

I'm working on building something upon dietpi called WaveOS, so should have more feedback for you guys as the project develops

Fourdee, 2018-04-26:
@unclehowell 

> when you install software ID 152 with dietpi.txt it works fine.
But try to run the install using a command on a running system, later down the line (dietpi-software install) and it freezes mid install.

Thanks for the report 👍 

I'll run a install test to see if we can replicate.

Fourdee, 2018-04-26:
@unclehowell 

Appears to be working fine in local testing, unable to replicate.

If its a hard lock, could be hardware related (SD/PSU instability). Do you receive any error messages when it fails?

```
root@DietPi:~# dietpi-software install 152
.........
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Optimize and configure software

[ INFO ] DietPi-Software | Applying DietPi optimizations and configurations for RPi 3 Model B (armv7l), please wait...

[ SUB1 ] DietPi-Services > dietpi_controlled
[  OK  ] DietPi-Services | dietpi_controlled : avahi-daemon
[  OK  ] DietPi-Services | dietpi_controlled : cron
[  OK  ] DietPi-Software | DietPi-Software installation completed.
[ SUB1 ] DietPi-Services > start
[  OK  ] DietPi-Services | start : avahi-daemon
[  OK  ] DietPi-Services | start : cron
[ SUB2 ] DietPi-Process_tool > Apply
[  OK  ] DietPi-Process_tool | Completed
```

Fourdee, 2018-05-06:
No response, making as closed.
Please reopen if required.

closed 2018-05-06 by Fourdee

unclehowell, 2018-05-07:
Thanks. I just install it via dietpi.txt now. That works. No error message when it freezes, it just does. 
I have a few other raspberry pi's kicking around, 2 & 3. I'll keep you posted.