MichaIng/DietPi #1730: Dietpi update from 6.4 to 6.7 on nanopi neo2

GB59, 2018-04-26:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | v6.4
- SBC Device     | NanoPi NEO 2 (aarch64) (index=65)
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Update

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
Get:10 https://cdn-aws.deb.debian.org/debian stretch-updates/main Translation-en 2018-03-28-2029.04.pdiff [2,033 B]
Get:11 https://cdn-aws.deb.debian.org/debian-security stretch/updates/main arm64 Packages [353 kB]
Get:12 https://cdn-aws.deb.debian.org/debian-security stretch/updates/main Translation-en [164 kB]
Get:13 https://cdn-aws.deb.debian.org/debian stretch-backports/main arm64 Packages.diff/Index [27.8 kB]
Get:14 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en.diff/Index [27.8 kB]
Get:15 https://cdn-aws.deb.debian.org/debian stretch-backports/contrib arm64 Packages.diff/Index [9,100 B]
Get:17 https://cdn-aws.deb.debian.org/debian stretch-backports/contrib Translation-en.diff/Index [4,180 B]
Get:18 https://cdn-aws.deb.debian.org/debian stretch-backports/non-free arm64 Packages.diff/Index [4,180 B]
Get:19 https://cdn-aws.deb.debian.org/debian stretch-backports/non-free Translation-en.diff/Index [7,624 B]
Ign:13 https://deb.debian.org/debian stretch-backports/main arm64 Packages.diff/Index
Ign:13 https://cdn-aws.deb.debian.org/debian stretch-backports/main arm64 Packages.diff/Index
Get:20 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-03-20-2027.59.pdiff [913 B]
Get:21 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-03-21-2027.13.pdiff [12.4 kB]
Get:22 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-03-23-2029.39.pdiff [1,896 B]
Get:23 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-03-26-1427.05.pdiff [661 B]
Get:24 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-03-28-0228.33.pdiff [208 B]
Get:25 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-03-28-2029.04.pdiff [699 B]
Get:26 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-03-30-1423.56.pdiff [33 B]
Get:27 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-03-30-2026.46.pdiff [961 B]
Get:28 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-03-31-0221.15.pdiff [31 B]
Get:29 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-04-08-1456.41.pdiff [183 B]
Get:30 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-04-11-1432.45.pdiff [13.7 kB]
Get:31 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-04-11-2023.04.pdiff [1,521 B]
Get:32 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-04-12-0225.46.pdiff [886 B]
Get:33 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-04-12-0828.47.pdiff [821 B]
Get:34 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-04-13-0225.32.pdiff [550 B]
Get:35 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-04-13-2021.57.pdiff [3,318 B]
Get:36 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-04-14-0225.46.pdiff [89 B]
Get:37 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-04-18-0222.52.pdiff [130 B]
Get:38 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-04-20-1430.21.pdiff [2,743 B]
Get:39 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-04-20-2023.55.pdiff [1,980 B]
Get:40 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-04-22-2033.30.pdiff [99 B]
Get:41 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-04-23-1420.05.pdiff [433 B]
Get:41 https://cdn-aws.deb.debian.org/debian stretch-backports/main Translation-en 2018-04-23-1420.05.pdiff [433 B]
Get:42 https://cdn-aws.deb.debian.org/debian stretch-backports/contrib arm64 Packages 2018-03-29-0835.32.pdiff [545 B]
Get:43 https://cdn-aws.deb.debian.org/debian stretch-backports/contrib arm64 Packages 2018-04-25-0226.50.pdiff [641 B]
Get:43 https://cdn-aws.deb.debian.org/debian stretch-backports/contrib arm64 Packages 2018-04-25-0226.50.pdiff [641 B]
Get:44 https://cdn-aws.deb.debian.org/debian stretch-backports/contrib Translation-en 2018-03-28-2029.04.pdiff [713 B]
Get:44 https://cdn-aws.deb.debian.org/debian stretch-backports/contrib Translation-en 2018-03-28-2029.04.pdiff [713 B]
Get:45 https://cdn-aws.deb.debian.org/debian stretch-backports/non-free arm64 Packages 2018-04-20-1430.21.pdiff [26 B]
Get:45 https://cdn-aws.deb.debian.org/debian stretch-backports/non-free arm64 Packages 2018-04-20-1430.21.pdiff [26 B]
Get:46 https://cdn-aws.deb.debian.org/debian stretch-backports/non-free Translation-en 2018-04-20-1430.21.pdiff [26 B]
Get:47 https://cdn-aws.deb.debian.org/debian stretch-backports/non-free Translation-en 2018-04-20-2023.55.pdiff [42 B]
Get:48 https://cdn-aws.deb.debian.org/debian stretch-backports/non-free Translation-en 2018-04-21-2025.43.pdiff [1,013 B]
Get:48 https://cdn-aws.deb.debian.org/debian stretch-backports/non-free Translation-en 2018-04-21-2025.43.pdiff [1,013 B]
Get:49 https://cdn-aws.deb.debian.org/debian stretch-backports/main arm64 Packages [296 kB]
Fetched 1,231 kB in 6s (185 kB/s)
Reading package lists...
E: Failed to fetch https://deb.debian.org/debian/dists/stretch-backports/main/binary-arm64/PackagesIndex  Couldn't parse pdiff index
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

Fourdee, 2018-04-26:
@GB59 

Thanks for the report 👍 

Please try changing the APT mirror:
```dietpi-config``` > Networking options NAS/MISC > APT mirror

Please let us know if this resolves.

Fourdee, 2018-05-02:
Unable to replicate, no response, marking as closed. Please reopen if required.

closed 2018-05-02 by Fourdee