MichaIng/DietPi #1733: Fresh install stopped on Xserver

Gokujo, 2018-04-27:
## Creating a bug report/issue:
During a fresh install it stopped on Xserver
[image]

#### Required Information:
- DietPi Version_ 6.7
- SBC Device: RPi 3B
- Distro: Linux DietPi 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux
#### Steps to reproduce:
I've chosen this soft to install:
- MATE: desktop enviroment
- Chromium: (Optional) web browser
- GPIO: gpio interface library for rpi (python)
- OpenSSH
- Build-Essentials: common packages for compiling
- Git Client: git clone etc
Webserver ist Apache2 and SSH ist OpenSSH

Fourdee, 2018-04-27:
@Gokujo 

Thanks for the report 👍 

I was unable to replicate in local testing, most likely temporary internet/APT issue, or filesystem issue.

Please try re-running the installation again. 

If the same issue occurs, please paste results of (everything after/below ```[20.0]``` should be enough):
```
dmesg
```

It may also be worth trying another APT mirror:
```dietpi-config``` > Networking Options NAS/MISC > APT mirror

Gokujo, 2018-04-27:
`root@DietPi:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.34-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1110 SMP Mon Apr 16 15:18:51 BST 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3dc00000
[    0.000000] On node 0 totalpages: 255488
[    0.000000] free_area_init_node: node 0, pgdat 80c84e40, node_mem_map bd336000
[    0.000000]   Normal zone: 2246 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 255488 pages, LIFO batch:31
[    0.000000] random: fast init done
[    0.000000] percpu: Embedded 17 pages/cpu @be59f000 s38720 r8192 d22720 u69632
[    0.000000] pcpu-alloc: s38720 r8192 d22720 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 253242
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=720 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=27504eef-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 990968K/1021952K available (7168K kernel code, 575K rwdata, 2072K rodata, 1024K init, 706K bss, 22792K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbe800000 - 0xff800000   (1040 MB)
                   lowmem  : 0x80000000 - 0xbe600000   ( 998 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80800000   (8160 kB)
                     .init : 0x80b00000 - 0x80c00000   (1024 kB)
                     .data : 0x80c00000 - 0x80c8fd4c   ( 576 kB)
                      .bss : 0x80c96f4c - 0x80d478b4   ( 707 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 25231 entries in 74 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000004] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000012] Switching to timer-based delay loop, resolution 52ns
[    0.000153] Console: colour dummy device 80x30
[    0.000432] console [tty1] enabled
[    0.000452] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000472] pid_max: default: 32768 minimum: 301
[    0.000648] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000667] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001219] Disabling memory control group subsystem
[    0.001275] CPU: Testing write buffer coherency: ok
[    0.001525] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.001742] Setting up static identity map for 0x100000 - 0x10003c
[    0.001815] Hierarchical SRCU implementation.
[    0.002164] smp: Bringing up secondary CPUs ...
[    0.002566] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.002981] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.003377] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.003429] smp: Brought up 1 node, 4 CPUs
[    0.003465] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.003476] CPU: All CPU(s) started in HYP mode.
[    0.003485] CPU: Virtualization extensions available.
[    0.003996] devtmpfs: initialized
[    0.009407] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.009548] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.009571] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.009895] pinctrl core: initialized pinctrl subsystem
[    0.010355] NET: Registered protocol family 16
[    0.011809] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.014234] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.014251] hw-breakpoint: maximum watchpoint size is 8 bytes.
[... 190 lines cut ...]
[    4.420803] usb 1-1.4: reset high-speed USB device number 4 using dwc_otg
[    4.467829] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[    4.553428] mt7601u 1-1.4:1.0: ASIC revision: 76010001 MAC revision: 76010500
[    4.562431] mt7601u 1-1.4:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[    4.913069] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[... cut ...]
[   26.063972] input: HID 046a:0023 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:046A:0023.0001/input/input1
[   26.071102] input: HID 046a:0023 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.1/0003:046A:0023.0002/input/input2
[   26.131402] cherry 0003:046A:0023.0001: input,hidraw0: USB HID v1.11 Keyboard [HID 046a:0023] on usb-3f980000.usb-1.2/input0
[   26.200987] cherry 0003:046A:0023.0002: input,hidraw1: USB HID v1.11 Device [HID 046a:0023] on usb-3f980000.usb-1.2/input1
[   65.991454] Adding 1097724k swap on /var/swap.  Priority:-2 extents:7 across:1523708k SSFS
`

Fourdee, 2018-04-27:
@Gokujo 

Thanks, looks fine.

Which SDcard and PSU used on the device?

Gokujo, 2018-04-27:
I tell you this in monday

Fourdee, 2018-05-05:
No response, marking as closed. Please reopen if required.

closed 2018-05-05 by Fourdee

Gokujo, 2018-05-06:
@Fourdee sorry, I've forgotten to post here.
We use SD-Cards from SanDisk and ther original PSu that was delivered with the device.

I tried installation on 3 devices - same problem. we waited 4-7 hours to see what happens and on one device the installation continued after 4 hours and on second one after ca 6 or 7 hours. and this is strange because usually it takes ca 30 minutes.

Gokujo, 2018-05-07:
`6388b7b3-76e7-4531-a684-b0b132abee7e-0 `

Gokujo, 2018-05-08:
> Please verify the exact PSU used on the device (brand/model/voltage/AMP)

[image]

> Try repairing the filesystem fsck `-f -y /boot`, then reboot and see if the error is removed from `dmesg`

```[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.34-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1110 SMP Mon Apr 16 15:18:51 BST 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3b800000
[    0.000000] On node 0 totalpages: 245760
[    0.000000] free_area_init_node: node 0, pgdat 80c84e40, node_mem_map baf86000
[    0.000000]   Normal zone: 2160 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 245760 pages, LIFO batch:31
[    0.000000] random: fast init done
[    0.000000] percpu: Embedded 17 pages/cpu @baf2f000 s38720 r8192 d22720 u69632
[    0.000000] pcpu-alloc: s38720 r8192 d22720 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 243600
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=720 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=27504eef-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 952416K/983040K available (7168K kernel code, 575K rwdata, 2072K rodata, 1024K init, 706K bss, 22432K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbc800000 - 0xff800000   (1072 MB)
                   lowmem  : 0x80000000 - 0xbc000000   ( 960 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80800000   (8160 kB)
                     .init : 0x80b00000 - 0x80c00000   (1024 kB)
                     .data : 0x80c00000 - 0x80c8fd4c   ( 576 kB)
                      .bss : 0x80c96f4c - 0x80d478b4   ( 707 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 25231 entries in 74 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000004] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000012] Switching to timer-based delay loop, resolution 52ns
[    0.000154] Console: colour dummy device 80x30
[    0.000430] console [tty1] enabled
[    0.000450] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000470] pid_max: default: 32768 minimum: 301
[    0.000647] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000665] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001220] Disabling memory control group subsystem
[    0.001276] CPU: Testing write buffer coherency: ok
[    0.001528] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.001746] Setting up static identity map for 0x100000 - 0x10003c
[    0.001819] Hierarchical SRCU implementation.
[    0.002170] smp: Bringing up secondary CPUs ...
[    0.002571] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.002984] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.003381] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.003433] smp: Brought up 1 node, 4 CPUs
[    0.003470] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.003481] CPU: All CPU(s) started in HYP mode.
[    0.003490] CPU: Virtualization extensions available.
[    0.003997] devtmpfs: initialized
[    0.009391] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.009529] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.009553] futex hash table entries: 1024 (order: 4, 65536 bytes)
[... 212 lines cut ...]
[    4.652200] usb 1-1.5: pl2303 converter now attached to ttyUSB0
[    5.002766] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    5.994179] FS-Cache: Netfs 'cifs' registered for caching
[    5.994519] Key type cifs.spnego registered
[    5.994533] Key type cifs.idmap registered
[    6.058505] CIFS VFS: Error connecting to socket. Aborting operation.
[    6.058520] CIFS VFS: cifs_mount failed w/return code = -111
[    6.990773] mmc1: error -110 whilst initialising SDIO card
[    7.132090] wlan0: authenticate with 00:0f:bb:07:0c:2a
[    7.193436] wlan0: send auth to 00:0f:bb:07:0c:2a (try 1/3)
[    7.195071] wlan0: authenticated
[... cut ...]
[   10.110777] mmc1: error -110 whilst initialising SDIO card
[   13.210870] mmc1: error -110 whilst initialising SDIO card
[   15.360112] CIFS VFS: Error connecting to socket. Aborting operation.
[   15.360123] CIFS VFS: cifs_mount failed w/return code = -111
[   26.137333] random: crng init done
´´´

this message I got from fresh installed dietpi. this time it took 30-40 minutes with all software (automated install don't work yet? e.g. `AUTO_SETUP_INSTALL_SOFTWARE_ID=` in dietpi on first install)

Gokujo, 2018-05-09:
@Fourdee 

> I'd highly recommend official RPi PSU

This one was delivered with raspberry

> Should be working fine once set. 

I didn't saw it on first look

> I'll take a look.

```#Modifications to /boot/* files, will not be saved when powered on.
#Please ensure you edit from the DietPi-RamDisk location /DietPi/*

#NB: This is intended for advanced users, unless you know what you are doing, do not edit this file. Please use the DietPi programs instead.
#NB: Do not remove uncommented lines, as the items are scraped by DietPi programs, on demand.

#------------------------------------------------------------------------------------------------------
# DietPi-Automation
# All settings below will be applied on the 1st boot of DietPi, ONCE.
#------------------------------------------------------------------------------------------------------

##### Networking Options #####

#	If both Ethernet and Wifi are enabled, Wifi will take priority and Ethernet will be disabled.
#		1=enabled
AUTO_SETUP_NET_ETHERNET_ENABLED=1
AUTO_SETUP_NET_WIFI_ENABLED=1

#	Enter your Wifi details below, if applicable (Case Sensitive).
AUTO_SETUP_NET_WIFI_SSID=WLAN_GAST
AUTO_SETUP_NET_WIFI_KEY=PASS
#	available | WPA-PSK / WPA-EAP / NONE | WEP=No longer supported
AUTO_SETUP_NET_WIFI_KEYMGR=WPA-PSK

#  For WPA2 Enterprise uncomment the following entries and set accordingly
#AUTO_SETUP_NET_WIFI_PROTO=RSN
#AUTO_SETUP_NET_WIFI_PAIRWISE=CCMP
#AUTO_SETUP_NET_WIFI_AUTH_ALG=SHARED
#AUTO_SETUP_NET_WIFI_EAP=PEAP
#AUTO_SETUP_NET_WIFI_IDENTITY=User
#AUTO_SETUP_NET_WIFI_PASSWORD=Pass
#
# If your WPA2 Enterprise requries a certificate copy the cert to this /boot directory and enter the name on the following line
#AUTO_SETUP_NET_WIFI_CERT=wpa2ent_cert.cer
# Uncomment either or both lines below and set accordingly for your WPA2 Enterprise needs.
#AUTO_SETUP_NET_WIFI_PHASE1=
#AUTO_SETUP_NET_WIFI_PHASE2=auth=MSCHAPV2

#	Enter your Static Network details below, if applicable.
AUTO_SETUP_NET_USESTATIC=0
AUTO_SETUP_NET_STATIC_IP=192.168.13.34
AUTO_SETUP_NET_STATIC_MASK=255.255.255.0
AUTO_SETUP_NET_STATIC_GATEWAY=192.168.0.10
AUTO_SETUP_NET_STATIC_DNS=1.1.1.1

#	Hostname
AUTO_SETUP_NET_HOSTNAME=JAS-TT

#	Force ethernet speeds
#		NB: This is mainly aimed towards Pine A64's which may have a HW issue that causes unstable 1Gbit link.
#		0=automatic speed | 10 = 10mbit, 100 = 100mbit etc
AUTO_SETUP_NET_ETH_FORCE_SPEED=0

##### Misc Options #####

#	Size of swapfile to generate (MB)
#		0=Disabled | 1=auto (2GB-RAM = size) | 2+=manual
AUTO_SETUP_SWAPFILE_SIZE=500
[... 178 lines cut ...]

#Optional username for ownCloud/Nextcloud admin account, default is 'admin'. Applied during installation.
SOFTWARE_OWNCLOUD_NEXTCLOUD_USERNAME=admin

#Optional data directory for ownCloud, default is '/mnt/dietpi_userdata/owncloud_data'. Applied during installation.
#	This option is for advanced users. For full compatibility, please keep this options defaults, and, use dietpi-drive_manager to move the DietPi user data location.
SOFTWARE_OWNCLOUD_DATADIR=/mnt/dietpi_userdata/owncloud_data

#Optional data directory for Nextcloud, default is '/mnt/dietpi_userdata/nextcloud_data'. Applied during installation.
#	This option is for advanced users. For full compatibility, please keep this options defaults, and, use dietpi-drive_manager to move the DietPi user data location.
SOFTWARE_NEXTCLOUD_DATADIR=/mnt/dietpi_userdata/nextcloud_data

#Wifi Hotspot
SOFTWARE_WIFI_HOTSPOT_SSID=DietPi-HotSpot
#	minimum of 8 characters
SOFTWARE_WIFI_HOTSPOT_KEY=dietpihotspot
SOFTWARE_WIFI_HOTSPOT_CHANNEL=3

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# Dev settings
#------------------------------------------------------------------------------------------------------
DEV_GITBRANCH=master
DEV_GITOWNER=Fourdee
```

Fourdee, 2018-05-09:
@Gokujo 

Thanks for dietpi.txt, looks fine.

Lets check whats installed:
```
dietpi-software list | grep ' =2 '
```

And the automation/1st run log:
```
cat /var/tmp/dietpi/logs/dietpi-firstrun-setup.log
```

---

> This one was delivered with raspberry

Its still not official, stability cannot be guaranteed unfortunately.

Gokujo, 2018-05-11:
I can send it a little bit later. I tried to add rasppbery into Windows Active Directory but there went something wrong and no passwords were synchronized. And automated installation went well. Like I said I didn't noticed this line in config:  `AUTO_SETUP_AUTOMATED=`. But I couldn't connect to enterprise wi-fi network.

Gokujo, 2018-05-11:
@Fourdee 

> `dietpi-software list | grep ' =2 '`

[image]

> `cat /var/tmp/dietpi/logs/dietpi-firstrun-setup.log`

```
[ INFO ] DietPi-Drive_Manager | Checking RootFS R/W access. Please wait...

[  OK  ] DietPi-Drive_Manager | RootFS R/W access.

[ INFO ] DietPi-Software | Initializing database, please wait...

[ INFO ] DietPi-Software | Reading database, please wait...

[ INFO ] DietPi-Software | DietPi - GPLv2 License:

This program is free software: you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation, either version 2 of the License, or any later version.

This program is distributed in the hope that it will be useful, but WITHOUT ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU General Public License for more details.

You should have received a copy of the GNU General Public License along with this program.  If not, see http://www.gnu.org/licenses/
───────────────────────────────────────
DietPi     | 10:51 | Fri 11/05/18 
───────────────────────────────────────
v6.7       | RPi 3 Model B (armv7l)
───────────────────────────────────────
NULL       | 
───────────────────────────────────────

 Welcome to DietPi-Software 

DietPi-Software | (1/5) Testing connection to ftp://mirror.netcologne.de/raspbian/raspbian/, please wait...
[.     ] 
[..    ] 
[ ..   ] 
[  ..  ] 
[   .. ] 
[    ..] 
[    ..] 
[   .. ] 
[  ..  ] 
[ ..   ] 
[..    ] 
[..    ] 
[ ..   ] 
[  ..  ] 
[   .. ] 
DietPi-Software | Failed connection attempt (1/5), retrying...
DietPi-Software | (2/5) Testing connection to ftp://mirror.netcologne.de/raspbian/raspbian/, please wait...
[    ..] 
[    ..] 
[  OK  ] DietPi-Software | Connection test: ftp://mirror.netcologne.de/raspbian/raspbian/

[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of time sync (1/30)

[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of time sync (2/30)

[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of time sync (3/30)

[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of time sync (4/30)

[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of time sync (5/30)

[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of time sync (6/30)

[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of time sync (7/30)

[... 2509 lines cut ...]
[  ..  ] 
[   .. ] 
[    ..] 
[    ..] 
[   .. ] 
[  ..  ] 
[ ..   ] 
[..    ] 
[..    ] 
[ ..   ] 
[  ..  ] 
[  OK  ] DietPi-Set_software | dpkg-reconfigure -f noninteractive locales

[  OK  ] locale de_DE.UTF-8 | Completed

[ INFO ] DietPi-Software | Setting Keyboard de. Please wait...

DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installation completed

[  OK  ] DietPi-Software | The system will now reboot. 
 This completes the DietPi-Software installation.

```