MichaIng/DietPi #1734: Install owncloud

aterhorst, 2018-04-29:
## Creating a bug report/issue:

Connecting to download.owncloud.org (download.owncloud.org)|46.4.80.187|:443...
connected.
ERROR: The certificate of ‘download.owncloud.org’ is not trusted.

#### Required Information:
- DietPi Version 6 7
- SBC Device RPi 3
- Power supply used 5V 2.1 A 
- SD card used (EG: Sandisk ultra)?
- Distro Linux DietPi 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

aterhorst, 2018-04-29:
Managed to reproduce problem with fresh install. I did work around by creating  
    nano ~/.wgetrc
inserting check_certificate=off
and then source ~/.wgetrc

aterhorst, 2018-04-29:
This was not a problem last time I installed owncloud (Jessie).

Fourdee, 2018-04-29:
@aterhorst 

Thanks for the report 👍 

Unable to replicate on Odroid C2 Jessie:
```
[  OK  ] DietPi-Software | Connection test: https://download.owncloud.org/community/owncloud-latest.zip
[  OK  ] DietPi-Software | wget https://download.owncloud.org/community/owncloud-latest.zip -O package.zip
[  OK  ] DietPi-Software | unzip -o package.zip -d /var/www
```

Please check the following package is installed:
```
dpkg -l | grep ca-certificates
```

If not, install it
```
G_AGI ca-certificates
```

Fourdee, 2018-05-05:
Unable to replicate on a fresh install of RPi 3.
```
[  OK  ] DietPi-Software | Connection test: https://download.owncloud.org/community/owncloud-latest.zip
[  OK  ] DietPi-Software | wget https://download.owncloud.org/community/owncloud-latest.zip -O package.zip
[  OK  ] DietPi-Software | unzip -o package.zip -d /var/www
```

```ca-certificates``` is installed by default.

Marking as closed, please reopen if required.

closed 2018-05-05 by Fourdee