MichaIng/DietPi #1735: Dietpi software | sonarr - error on install

clarky0000, 2018-04-29:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version 159
- SBC Device RPi 3
- SD card used (EG: Sandisk ultra)?
- Distro Linux DietPi 4.9.62-v7+ #2 SMP Fri Nov 17 23:52:26 GMT 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
- Software sonarr 
- Can this issue be replicated on a fresh installation of DietPi? Dont know.

Got an error trying to install Sonarr. Error from Dietpi menu:

---------------------------------------
Apt-get has failed:
-Apt string = mono-runtime
-Error code = 100
-Software Title = Mono: runtime libraries and repo
-Apt log = /etc/dietpi/logs/dietpi-software_apt.log

DietPi-software will now exit.
--------------------------------------------

Error from log file above:

E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

I did try 'apt --fix-broken install' but it didnt seem to resolve the issue

#### Steps to reproduce:
Install sonarr from software menu

Fourdee, 2018-04-29:
@clarky0000 

Thanks for the report, we'll try to replicate 👍

Fourdee, 2018-04-29:
@clarky0000 

> DietPi Version 159

You are running an outdated version of DietPi, which is no longer supported. v6.0 or higher is required and contains vast number of fixes and improvements. 

Please see the following link for more information. You will need to download and install the latest image:
https://github.com/Fourdee/DietPi/issues/1355

🈯️ Confirmed install of Sonarr running fine on v6.7

I'll mark this as closed, please reopen if required.

closed 2018-04-29 by Fourdee