MichaIng/DietPi #1737: Chromium autostart fails to launch

AYapejian, 2018-04-29:
## Creating a bug report/issue:

#### Required Information:
* DietPi Version:
```shell
root@pi-display:~# cat /DietPi/dietpi/.version
6
7
```
* SBC: RPi 3
* PSU: 5v 3a noname
* SD Card: Sandisk evo 32gb
* uname: 
```shell
root@pi-display:~# uname -a
Linux pi-display 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux
```

#### Additional Information (if applicable):
- Software title: Chroumium
- Can this issue be replicated on a fresh installation of DietPi? Yes with Autostart set to Chromium

#### Expected behaviour:
Chromium should autostart

#### Actual behaviour:
X starts and crashes

#### Steps to reproduce:
Install Chromium and set to autostart, reboot

#### Did you submit a ```dietpi-bugreport```?
No

#### Extra details:
Issue seems to be that the /Dietpi/dietpi/login calls `xinit chromium` however the bin is `chromium-browser`

#### My Fix
Really just need to rename the binary in the `login` script  I believe but some of these switches / notes may be helpful to others.

* `touch /root/start-chrome.sh`
* `chmod +x /root/start-chrome.sh`
* Add contents to file and save:

```shell
#!/bin/sh
# If using the 7" Pi LCD set below in /DietPi/config.txt
#    framebuffer_width=1024
#    framebuffer_height=614
# 	 gpu_mem_256=128
# 	 gpu_mem_512=128
# 	 gpu_mem_1024=128

# Command line switches https://peter.sh/experiments/chromium-command-line-switches/
# 	--test-type gets rid of some of the chromium warnings that you may or may not care about in kiosk on a LAN
#   --pull-to-refresh=1 
#   --ash-host-window-bounds="400,300"

url=${CHROMIUM_URL:-"http://URL_IF_NOT_SET"}
chromium_opts="--kiosk --test-type --force-tablet-mode"
xinit /usr/bin/chromium-browser ${chromium_opts} ${url}

# Change /DietPi/dietpi/login
# `xinit chromium` to `/root/start-chrome.sh`
```

* Run with `/root/start-chrome.sh` and/or replace `xinit chromium` with `/root/start-chrome.sh` in `/Dietpi/dietpi/login` ( NOTE: New to dietpi, I'm guessing this would get overwritten during updates though )

symbios24, 2018-04-30:
try the sh start-chrome.sh

Fourdee, 2018-04-30:
@AYapejian 

Many thanks for the report and fix👍 

We'll try to replicate.

Fourdee, 2018-04-30:
Confirmed.

Notes:
- 🈯️ We can also remove the pre-req for desktop install.
- 🈯️ We'll implement auto GPU mem split increase in ```dietpi-software``` during Chromium install.
- We'll also implement a startup bash script for Kiosk and additional options, based on @AYapejian

MichaIng, 2025-11-15:
Please open a new issue about that. Add there: matchbox mode is a 3rd mode similar to kiosk, so we have kiosk, app, and matchbox modes? At some point, an additional submenu to select one of these 3 modes, with sane explanation about their differences, wouldn't be a bad idea.