MichaIng/DietPi #1740: DietPi-Drive_Manager | Filesystem check

Fourdee, 2018-04-30:
>

MichaIng, 2018-04-30:
@Fourdee 
Jep, I like the idea. Although I am not sure, if forced fix is always the best, as it can also lead to data loss. Although, the data is anyway lost, if fix cannot be applied, right?

Fourdee, 2018-04-30:
@MichaIng 

> Although, the data is anyway lost, if fix cannot be applied, right?

Pretty much, lost via known fixable means. Paid Data recovery business then required.

MichaIng, 2018-04-30:
@Fourdee
Yeah, the question is if then the fsck fix attempt and removal/clearing makes things worse.
Maybe there is some level of errors where we can leave choice to user, informing in the first place about possible results of a forced fix/clearing.

Fourdee, 2018-04-30:
@MichaIng 

> Maybe there is some level of errors where we can leave choice to user, informing in the first place about possible results of a forced fix/clearing.

Yep 👍 

Dry run 1st, then repair option?

Fourdee, 2018-04-30:
Finally fixed the incorrect scaling in G_WHIP while I was here 🤣 :
We now calculate character count per line against X:
[image]

Fourdee, 2018-04-30:
🈯️ Tested with EXT4 external drive

Fourdee, 2018-04-30:
Ok so only downside, as we define drive must be unmounted for a check/repair to run, this removes the ability to do this on:
- /boot (which has umount disabled to prevent issues with ramdisk etc)
- RootFS (cannot/should not be done on a active filesystem)

MichaIng, 2018-04-30:
Jep, I guess this is known that rootfs can only be scanned and repaired on reboot (boot arguments) or on other system.

For /boot, as long as it's within single script on our control, I think it's okay to umount + check + remount it? Only important thing is it being mounted again on reboot/shutdown for ramdisk?
Actually as far as I can remember FAT fs can be scanned while mounted as well, but still to be on secure side I would always umount.

MichaIng, 2018-05-09:
Testing:
- F2FS: `-f` for full check&repair, `-a` just for actually processed files? No dry-run possible!
```
root@DietPi:/var/log# fsck -h /dev/sdb1
fsck from util-linux 2.29.2
fsck.f2fs: invalid option -- 'h'
        Error: Unknown option ?

Usage: fsck.f2fs [options] device
[options]:
  -a check/fix potential corruption, reported by f2fs
  -d debug level [default:0]
  -f check/fix entire partition
  -p preen mode [default:0 the same as -a [0|1]]
  -t show directory tree [-d -1]
```
- NTFS: .... wait, messed up, as I removed ntfs-3g and format gone wrong... 🤣

Fourdee, 2018-05-10:
@MichaIng 

> We definitely need separate commands + options for supported FS types.

Great work 👍 

I'll implement this today and send commit.

MichaIng, 2018-05-10:
@Fourdee 
Waiiiit, I'm already done 😄! Currently testing with all FS and will push the commit to my opened PR.

Fourdee, 2018-05-10:
@MichaIng 

Great work on this PR https://github.com/Fourdee/DietPi/pull/1763, legend 👍 

PS. Parcel sent today, should be with you next week (NanoPi K1+ and a special item 😃 )

MichaIng, 2018-05-11:
ToDo:
- [ ] Select root/boot device for scan (& repair) on next reboot.
  - With https://github.com/Fourdee/DietPi/commit/0eb0db0209985ff91fcd425bbd4ca4a4da6626d8 possible just for rootfs right now.
  - Live Check & Repair disabled for root and boot
  - Enable /boot scan on reboot and/or live? The latter would be possible, as long as re control remount for proper DietPi-RAMdisk sync, kernel/firmware updates etc.

Methods:
- `> /forcefsck` deprecated but still works, `echo y > /forcefsck` does not work any more to force repair.
  - Temporary: File is removed after reboot!
- RPi only: Add `fsck.mode=force` to `cmdline.txt`, `fsck.repair=yes` to force repair. According to some web research, ext4 auto repair seems to be safe, as it just fixes errors, that are safe to repair and do not create additional damage.
  - Permanent, would need removal after reboot, at least if no already added manually by user.
- `tune2fs -c 1 /dev/sdXX`: Set automatic scan interval to every `1` boot.
  - Permanent, would need adjustment to default value after reboot.

Fourdee, 2018-05-11:
@MichaIng 

> v6.9

Yep 👍

MichaIng, 2018-06-03:
Move scan on reboot to v6.10
- Concentrate on testing many deep changes done already.

MichaIng, 2018-11-20:
Not sure where I found/reported this:
- On x86, fsck is done via initramfs and on Jessie, the created flag (don't remember location of this currently) prevents other fsck methods (on later boot stage) from being done. Nothing shows up in boot log then (dmesg/journalctl) as well then.
- On Stretch, there is generally systemd-fsck available, which does rootfs check on reboot, if requests by one of the above methods. On Jessie, systemd-fsck does not yet exist, thus above methods do not work in every case, especially if no initramfs is used (e.g. RPi).
- On RPi `fsck.mode=force` does it on earliest boot stage, but need to be actively removed to prevent this on every reboot.

So assuring rootfs fsck on reboot on all systems without further removal steps after reboot, only works on Stretch+, AFAIK, via `> /forcefsck`, done by either initramfs or systemd-fsck then.
- [ ] Check if the flag file forces it on all partitions then or only on the partition where it is placed. Then enable this feature only on Streth+-
- [ ] Check whether forcefsck works still on Buster, since it is already marked as deprecated for a while.

Fourdee, 2018-11-22:
@MichaIng 

The idea I had to cover all systems was:
- During boot at local-fs.target once rootfs is mounted
- Copy fsck binary somewhere off the rootfs (/tmp?)
- service which unmounts rootfs, runs ```/tmp/fsck```, then remounts
- disable service

However, don't believe you can unmount rootfs once its mounted in any case.

MichaIng, 2018-12-09:
PR up: https://github.com/Fourdee/DietPi/pull/2335

MichaIng, 2019-01-28:
Will mark as resolved.

closed 2019-01-28 by MichaIng