MichaIng/DietPi #1741: Webmin missing perl dependency during install

Fffrank, 2018-05-02:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 6.7
- SBC Device (EG: RPi 3)? Odroid C2
- Power supply used (EG: 5v 1A RAVpower)? 2.5A Amazon
- SD card used (EG: Sandisk ultra)? Sandisk Ultra
- Distro (EG: Jessie) | Jessie

#### Additional Information (if applicable):
- Software title? WebMin
- Can this issue be replicated on a fresh installation of DietPi? Yes.

#### Expected behaviour:
WebMin installs through dietpi-software

#### Actual behaviour:
Webmin Fails due to missing dependency (perl)

#### Steps to reproduce:
Install WebMin on fresh Dietpi

#### Did you submit a ```dietpi-bugreport```?
No.

MichaIng, 2018-05-04:
@Fffrank 
Thanks for your report.

Even worse on my Stretch VM:
```
dpkg: dependency problems prevent configuration of webmin:
 webmin depends on libnet-ssleay-perl; however:
  Package libnet-ssleay-perl is not installed.
 webmin depends on libauthen-pam-perl; however:
  Package libauthen-pam-perl is not installed.
 webmin depends on libio-pty-perl; however:
  Package libio-pty-perl is not installed.
 webmin depends on apt-show-versions; however:
  Package apt-show-versions is not installed.
```

We should use their dependency list: http://webmin.com/deb.html
`apt-get install perl libnet-ssleay-perl openssl libauthen-pam-perl libpam-runtime libio-pty-perl apt-show-versions python`

luisspuerto, 2018-05-18:
I think I have the same issue and I summited a bug report: c1086a6e-a26a-4ef4-ad97-44ebd15b46c7-0

I've just 
```sh
apt-get install perl libnet-ssleay-perl openssl libauthen-pam-perl libpam-runtime libio-pty-perl apt-show-versions python
```
And now works.

Fourdee, 2018-05-18:
@luisspuerto 

Thanks for the report 👍 

I'll move this to v6.9 and investigate this further.

Fourdee, 2018-05-18:
Ok, perl/python are required at the very least for G_AGF to resolve. We could manually add the required pre-reqs to install list, however, could change on distro version and .deb update of webmin.