MichaIng/DietPi #1742: Image for NanoPi Fire 3/M3/T3 with kernel 4.4?

swd120, 2018-05-02:
Have a NanoPi Fire 3 - I installed the existing M3/T3 image and it works fine but has the dated 3.x kernel.   Is it possible to get an updated image based on the 4.4 kernel version?

http://wiki.friendlyarm.com/wiki/index.php/NanoPi_Fire3#Get_Started => img s5p6818-friendly-core-xenial-4.4-arm64-YYYYMMDD.img.zip

Thanks!

MichaIng, 2018-05-02:
@swd120 
Thanks for your request. Can't say much about NanoPi but note, that the images you link are `Ubuntu` based, where DietPi needs `Debian` based images 😉.
Although I was experimenting a bid about installing DietPi on top of Ubuntu, but this is far away from being stable or even working with much hack around: https://github.com/Fourdee/DietPi/issues/1557

SuBLiNeR, 2018-05-02:
Hi there @swd120 i think you should start with this image from "armbian" (https://dl.armbian.com/nanopim3/archive/Armbian_5.38_Nanopim3_Debian_stretch_next_4.14.15.7z) and than use the Prep-Script as descirbed by @Fourdee. You'll have than a working DietPi on `Stretch` with `kernel 4.14.15` or higher if you do a firmware right after you installed the armbian image on first run -> than reboot and do the PREP-Script.

swd120, 2018-05-04:
Giving that a try - thanks!

closed 2018-05-04 by swd120