MichaIng/DietPi #1743: RPI high temperature when system idle

nyx-rattapoom, 2018-05-02:
#### Required Information:
- DietPi Version : 6.7
- SBC Device : RPi 1 B 
- Distro: Linux raspberry 4.14.37+ #1111 Thu Apr 26 13:33:13 BST 2018 armv6l GNU/Linux

I recent change OS from raspbian-lite to DietPI. After fresh install I saw cpu abnormal temperature.
In raspbian cpu temperature idle ~42-45 C but on DietPI it idle ~60 C.
I `sudo cpu` command it told min-max cpu frequency is the same (700mhz)
I think it issue about cpu scaling, so could I have any advice ?

Fourdee, 2018-05-02:
@nyx-rattapoom 

Thanks for the report, we'll run tests to see if we can replicate.

SuBLiNeR, 2018-05-02:
@Fourdee @nyx-rattapoom Just testing this on a old RPiBPlus.... nothing stranged happens there with latest DietPi in Idle mode. CPU is around 43 degrees while doing nothing.

@nyx-rattapoom What else have you installed what could causes the idle so high?

SuBLiNeR, 2018-05-02:
@Fourdee Nothing to thank me... It was me a pleasure! 

Normally this device runs a MinecraftPE Server and it almost is very low on temperature; around 56 degrees while my son is playing on it.
I did a clean install for only testing this... so it must be a hardware failure or something in the background causing this. 

I'm really looking forward on hearing about what else is maybe installed.

nyx-rattapoom, 2018-05-04:
@SuBLiNeR I'm not sure. It same model. Please run ```lscpu``` and ```sudo cpu``` for me.
```
dietpi@raspberry:~$ lscpu
Architecture:          armv6l
Byte Order:            Little Endian
CPU(s):                1
On-line CPU(s) list:   0
Thread(s) per core:    1
Core(s) per socket:    1
Socket(s):             1
Model:                 7
Model name:            ARMv6-compatible processor rev 7 (v6l)
CPU max MHz:           900.0000
CPU min MHz:           700.0000
BogoMIPS:              698.95
Flags:                 half thumb fastmult vfp edsp java tls
```

@Fourdee I reinstall dietpi without addition install result is the same.
```
dietpi@raspberry:~$ uptime
 05:44:50 up 1 day, 19:53,  1 user,  load average: 0.04, 0.03, 0.00
dietpi@raspberry:~$ sudo cpu
[  OK  ] Root access verified.

 ───────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ───────────────
 Architecture |     armv6l
 Temp         |     58'c : 136'f  | Running warm, but safe.
 Governor     |     ondemand
 Throttle up  |     50% CPU usage

                 Current Freq    Min Freq   Max Freq
 CPU0         |      900 MHz      700 MHz    900 MHz
```
I don't know why Min Freq is 700 MHz.
I have orange pi zero. It running with dietpi but Min freq is 240 Mhz maybe it cause this issue ?

[image]

Fourdee, 2018-05-04:
@nyx-rattapoom 

> I have orange pi zero. It running with dietpi but Min freq is 240 Mhz maybe it cause this issue ?

RPi 1 clock steps are 700MHz and 900MHz. Below that is possible, but unstable. Either way, the high CPU temp should not occur at 700MHz idle.

Does the CPU feel hot to the touch? Roughly the same temp?

#### Please paste results:
```
cat /sys/class/thermal/thermal_zone0/temp
vcgencmd measure_temp
```

SuBLiNeR, 2018-05-04:
@nyx-rattapoom this how it looks to me... while running a Nukkit-Server:

`root@RaspPiBPlus:~# lscpu
Architecture:          armv6l
Byte Order:            Little Endian
CPU(s):                1
On-line CPU(s) list:   0
Thread(s) per core:    1
Core(s) per socket:    1
Socket(s):             1
Model:                 7
Model name:            ARMv6-compatible processor rev 7 (v6l)
CPU max MHz:           900,0000
CPU min MHz:           700,0000
BogoMIPS:              898.66
Flags:                 half thumb fastmult vfp edsp java tls`

this is how `htop` looks alike 

[image]

Looks to me like the same.... only that the BogoMIPS is different...

SuBLiNeR, 2018-05-04:
@Fourdee the result of `cat `/sys/class/thermal/thermal_zone0/temp`
vcgencmd measure_temp`

SuBLiNeR, 2018-05-05:
@all So i did some testing here with the latest kernel 4.14.37.... it seems that this kernel is not stable after doing rpi-update at least. it crahses my whole home networking.

Just to inform, do not update as of now via rpi-update!

MichaIng, 2018-05-05:
@SuBLiNeR 
Just for the record: `rpi-update` is considered to be a developer tool and might always be unstable in certain environments. In production system one should never use it, as far as there are no certain issues with the latest stable release (APT packages raspberrypi-kernel/bootloader/bin/0) that got fixed with github master: https://github.com/Hexxeh/rpi-firmware

But yeah, I use it also 🤣, maybe 4.14.39 released an hour ago fixes your network issue with this?
`kernel: lan78xx: PHY DSP registers initialization to address EEE link drop issues with long cables` No idea, better stay with the APT packed firmware to avoid running into trouble by times.

SuBLiNeR, 2018-05-05:
@MichaIng I know.... but he was talking about `4.14.37+` which i hadn't... so i gave it a try. 

Offtopic:

Result was than as explained above. aaaannnndd... normally i'm using the dev kernels with no such problems since 2 years. It is now the second time that my whole network was being killed with this kernel. Just to let you know.... :1234:

SuBLiNeR, 2018-05-05:
@nyx-rattapoom btw. as i told you, start fresh without using rpi-update or kernel updating. System runs for me stable since 3 hours between 43 - 45 degrees while my son is playing on it.

If it isn't a harware failure it should be okay.

nyx-rattapoom, 2018-05-06:
@Fourdee 
```
dietpi@raspberry:~$ cat /sys/class/thermal/thermal_zone0/temp
59452
dietpi@raspberry:~$ vcgencmd measure_temp
temp=60.0'C
dietpi@raspberry:~$ 
```
@SuBLiNeR  OK, After that I will reinstall dietpi without rpi-update and post result again.

nyx-rattapoom, 2018-05-06:
So I installed raspbian-lite without addition.
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.34+ #1110 Mon Apr 16 14:51:42 BST 2018 armv6l GNU/Linux

pi@raspberrypi:~ $ vcgencmd version
Apr 16 2018 18:16:56 
Copyright (c) 2012 Broadcom
version af8084725947aa2c7314172068f79dad9be1c8b4 (clean) (release)

pi@raspberrypi:~ $ vcgencmd measure_temp
temp=46.1'C
```
from `vcgencmd version` output same as dietpi but temperature lower than 10'C. 
So I cann't find the reason why, And now I'm done.

Fourdee, 2018-05-06:
@nyx-rattapoom 

Interesting, thanks for posting the results.

Its possible a incorrect CPU clock is being applied based on the board revision number and detection.

Please can you paste results, we'd like to verify the RPi revision number:
```
grep -m1 'Revision' /proc/cpuinfo | awk '{print $3}'
```

And if you run DietPi again, please paste results:
```
cat /DietPi/config.txt | grep freq
```

Fourdee, 2018-05-11:
@nyx-rattapoom 

> So I cann't find the reason why, And now I'm done.

I'll mark this as closed. Please reopen if you are wiling to verify https://github.com/Fourdee/DietPi/issues/1743#issuecomment-386876775 and still require assistance.

closed 2018-05-11 by Fourdee