MichaIng/DietPi #1747: Dietpi-software: Raspimjpeg does not stream mjpeg with Lighttpd

fieldOfView, 2018-05-04:
## Creating a bug report/issue:
When installing raspimjpeg (package 59) with Lighttpd as the webserver, streaming mjpeg does not work. Getting jpeg stills in rapid succession (the "default stream" option) does work. Streaming mjpeg does work well when selecting nginx as the webserver preference.

#### Required Information:
- DietPi Version 6.7
- SBC: RPi ZeroW
- Power supply 5v 1A offbrand
- SD card used 8Gb class 10 offbrand
- Distro: Stretch (4.14.34+ #1110 Mon Apr 16 14:51:42 BST 2018 armv6l GNU/Linux)

#### Additional Information (if applicable):
- Software title raspimjpeg (59:   RPi Cam Control)
- This issue can be replicated on a fresh installation of DietPi

#### Steps to reproduce:
Open the rpicam webpage in a browser (and see the camera stream working)
Go to System and press "mjpeg stream"

#### Expected behaviour:
An mjpeg stream should be received by the client one frame at a time from a single http connection

#### Actual behaviour:
No data is arriving at the client, and after a while the client times out (if it is well-behaved)
!()[https://www.raspberrypi.org/forums/download/file.php?id=19933&sid=409607f05a0f8ec6ebd6fb15ede556f8]

#### Did you submit a ```dietpi-bugreport```?
No

#### Extra details:
This is an upstream issue between raspimjpeg and lighttpd, some more information can be had here:
https://www.raspberrypi.org/forums/viewtopic.php?t=63276&start=4250#p1210885

For DietPi, this is mostly a documentation issue. Since selecting the raspimjpeg package in the dietpi-software already pops up a dialog, that dialog could include a warning about the incompatibility and an advisory to use nginx (or apache?) instead.

Fourdee, 2018-05-05:
@fieldOfView 

Many thanks for the report 👍 

We'll open for investigations.

Fourdee, 2018-05-06:
Moved to 6.9 so we can focus on 6.8 wrap up and release.

Fourdee, 2018-05-24:
Notes:
- https://elinux.org/RPi-Cam-Web-Interface:
By default the system generates the live preview as a continuous set of captured images as this has maximum browser compatibility. MJPEG streaming may be selected but this may not work on all browsers.

fieldOfView, 2018-05-24:
Yes, repetitive jpeg “faux-streaming” works fine. I am using raspimjpeg as an alternative to mjpegstreamer for an OctoPrint instance, and that requires mjpeg streaming.

Fourdee, 2018-06-14:
https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_raspicam/README.md
https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=45178

🈯️ 
```
G_AGI subversion libraspberrypi-dev cmake libjpeg8-dev build-essential imagemagick libav-tools libv4l-dev
wget https://github.com/jacksonliam/mjpg-streamer/archive/master.zip -O package.zip
unzip package.zip -d /opt
cd /opt/mjpg-streamer-master/mjpg-streamer-experimental
export LD_LIBRARY_PATH=.
make -j $(nproc --all)
make install

./mjpg_streamer -o "output_http.so -w ./www" -i "input_raspicam.so -x 1280 -y 720 -fps 15"

```
[image]

Fourdee, 2018-07-16:
Moving this out of milestone for the following reasons:
- Low priority. Does not effect usage of RPi cam software currently.
- This is an additional feature request, which requires further investigations at a later date when time permits.

MichaIng, 2019-07-07:
@fieldOfView @Fourdee 
This should be retested with the new Raspbian Buster-based image: https://github.com/MichaIng/DietPi/issues/2935#issuecomment-506389899
It was not an issue with some older Jessie-based image (checking the linked RPi forum thread), so it is not a general issue with Lighttpd.

MichaIng, 2019-08-11:
@fieldOfView 
I just found by chance that the extended device tree has some mjpeg support.
Could be tested with:
```
G_CONFIG_INJECT 'start_x=' 'start_x=1' /DietPi/config.txt
reboot
```

MichaIng, 2021-02-18:
I have no RPi camera to test, but generally it should work with Lighttpd (now), as the web interface and it's own installer is actively developed and has a Lighttpd block.

I updated our install code to especially use the dedicated Stretch-compatible raspimjpeg binary on Stretch systems, but otherwise it looks good. Would be awesome if someone with an RPi camera module could test it, but I mark this issue as closed now. Please open a new issue if anything does not work or any feature is missing.

closed 2021-02-18 by MichaIng