MichaIng/DietPi #1748: DietPi-Drive_Manager | Incorrectly detect/assigns mounts

Fourdee, 2018-05-04:
Due to ```PTUUID``` being displayed in our results, lacks UUID etc :
```
root@DietPi:~# blkid
/dev/mmcblk1p1: SEC_TYPE="msdos" LABEL="BOOT" UUID="7451-4AA3" TYPE="vfat" PARTUUID="000f1766-01"
/dev/mmcblk1p2: LABEL="rootfs" UUID="af314fbe-6db6-4970-84b4-19fd22bd6bf7" TYPE="ext4" PARTUUID="000f1766-02"
/dev/sda1: UUID="757cbcab-ed15-4250-8029-f8c9bd3aac9c" TYPE="ext4" PARTUUID="258cf254-01"
/dev/mmcblk1: PTUUID="000f1766" PTTYPE="dos"

root@DietPi:~# blkid -o device
/dev/mmcblk1p1
/dev/mmcblk1p2
/dev/sda1
/dev/mmcblk1

root@DietPi:~# blkid -s UUID -o value
7451-4AA3
af314fbe-6db6-4970-84b4-19fd22bd6bf7
757cbcab-ed15-4250-8029-f8c9bd3aac9c

root@DietPi:~# blkid -s PARTUUID -o value
000f1766-01
000f1766-02
258cf254-01
```

SuBLiNeR, 2018-05-04:
Don't see a problem here until it is one if you describe how!?

[image]

Fourdee, 2018-05-05:
@SuBLiNeR 

> Don't see a problem here until it is one if you describe how!?

Thanks for confirming current install is fine. I experienced this using dev testing branch and in PREP.

Fourdee, 2018-05-05:
Testing C2 passed:
Resolved.

closed 2018-05-05 by Fourdee