MichaIng/DietPi #1749: Merging DietPi-Set_Core_Environment into DietPi-PREP?

MichaIng, 2018-05-05:
@Fourdee @k-plan
As DietPi-Set_Core_Environment resets a bid too much to use it for adopting changes on update, what do you think about merging it into the preparation script?

I am sometimes confused when tracking issues on preparation, switching between both scripts. Also some steps are doubled.

It would give a better overview and allows better red line within the process. And it would decrease the memory usage of DietPi a bid :wink:.

Only thing is that we would need to adjust the script calls within patch_file to manually adopt the changes.

Alternatively we could reduce the script to only implement our services and disable/mask the incompatible ones?
But to be true I still vote for moving our services into single files on github and bring them in place on dietpi-update by default. So they get updated automatically and are completely outside the memory afterwards. Now their content is kept in RAM together with dietpi-set_core_environment :wink:.

Fourdee, 2018-05-05:
@MichaIng 

The original reason for splitting these is to allow us to quickly run ```DietPi-Set_Core_Environment``` to patch critical DietPi 'stuff'/cron jobs/services etc on existing system, without having to manually patch using a copy from PREP.
The idea of it was to simplify patching for us, less chance of bad patch.

However, you make valid points. Yep, lets move this back to PREP.

MichaIng, 2018-05-06:
@Fourdee 

If we want to use it for the initial purpose, as far as I could see, we need to move at least dietpi user creation, cron and grub adjustments that would overwrite manual user additions/changes.

What do you think of the idea to create a separate /root folder tree on github that contains a folder structure for needed DietPi files outside of /boot/dietpi? This could me merged then into rootfs on dietpi-update (and by DietPi-PREP) and by this guarantee that our services and e.g. needed config files are always up-to-date without the need for defining their content within the scripts?

Fourdee, 2018-05-06:
@MichaIng 

> What do you think of the idea to create a separate /root folder tree on github that contains a folder structure for needed DietPi files outside of /boot/dietpi? 

We could try to use our existing ```/var/lib/dietpi``` folder for this?

> This could me merged then into rootfs on dietpi-update (and by DietPi-PREP) and by this guarantee that our services and e.g. needed config files are always up-to-date without the need for defining their content within the scripts?

I've probably not had enough coffee yet 🤣 , but i'am unsure what you mean? 😕

MichaIng, 2018-05-06:
@Fourdee
I mean our service files and properly configuration files we add: E.g. `dietpi-boot.service` `/etc/profile.d/99-dietpi*` etc. All those which need a certain location on system.
Storing them within the same folder structure on github as well, so that we can just copy this to `/` and are fine.

Fourdee, 2018-05-06:
@MichaIng 

> Storing them within the same folder structure on github as well, so that we can just copy this to / and are fine.

Makes sense 👍 Good idea. Yep agree, lets go ahead with this.

Fourdee, 2018-05-07:
Ok, best we wait until all pre-v6.2 images have been re-done, as v6.1 and below get patched with ```/DietPi/dietpi/func/dietpi-set_core_environment```. And, i'am not keen on copying and pasting it all in the patch file :)

MichaIng, 2018-05-30:
Tests:
🈯️ v6.0 after first run setup -> v6.9 via `dietpi-update 1`
🈯️ v6.-1 -> v6.9 via first run setup

Works as expected from first update on. I will mark this as closed, although further testing makes sense, as well as reordering/cleaning the DietPi-Set_Core_Environment parts within DietPi-PREP.

closed 2018-05-30 by MichaIng

MichaIng, 2018-05-30:
@Fourdee 
Jep, also all specific software title related content of `/DietPi/dietpi/conf` could be placed e.g. into `rootfs/var/lib/dietpi/dietpi-software/...` to reduce RAM load.
I was even thinking if it makes sense to not place them on the system at all, just download on demand/software installation. But this would mean a possible mismatch between DietPi version and software conf/service version.

Btw. do you know if it's possible to place symlinks into github? Should be possible if cloned from local repo, but at least I don't know how to create them within github web ui directly.

Fourdee, 2018-05-31:
@MichaIng 

Not sure, although we could symlink DietPi-RAMdisk to where we want it, during init:
```
ln -sf /var/lib/dietpi/dietpi-software/conf /DietPi/dietpi/conf
```

```
root@DietPi:~# ls -lha /DietPi/dietpi | grep conf
lrwxrwxrwx 1 root root   36 May 31 18:31 conf -> /var/lib/dietpi/dietpi-software/conf
```