MichaIng/DietPi #1750: plex is missing in dietpi-process_tool

bokiroki, 2018-05-05:
raspberry pi 3
latest 6.7 system
Hi i want to configure plex but it's missing from dietpi-process_tool
What can i do, all other programs are there..

Fourdee, 2018-05-05:
@bokiroki 

Thanks for the report 👍 

We'll take a look.

Fourdee, 2018-05-05:
Ok yep, as I thought.

Plex spawns new processes each time transcoding starts. Because of this, we would need to run ```dietpi-process_tool``` as a background process, that gets applied every say 60 seconds. 
The problem with this, is it will degrade system performance every time it runs, negating its purpose.

As such. I'll add the Plex service processes, however, transcoding will not be an option.

Fourdee, 2018-05-05:
Completed.

closed 2018-05-05 by Fourdee

bokiroki, 2018-05-06:
Thanks @Fourdee, i see that there is one more plex process: Plex EAE Service
On plex forums i saw that this is Easy Audio Encoder, i suppose it helps when audio is needed to be transcoded, don't know exactly
One question if you dont mind.. deluge is fixed in latest version. didn't had any problems yet, but i see that it's always using 0.7 or 1.3 percent of cpu when it's not used..other programs sits on zero when they not doing anything..not big deal, just what i noticed
[image]

bokiroki, 2018-05-06:
@Fourdee i installed transmission like you said. When i enter transmission config
/var/lib/transmission-daemon/info/settings.json
i see some options that are from older versions and some from newer versions..
With google i found this thread, someone mentioned that it because of script, i quess you are using script to inject optimised options for dietpi..
https://forum.transmissionbt.com/viewtopic.php?t=17018
take a look
Thanks

Fourdee, 2018-05-06:
@bokiroki 

> i see some options that are from older versions and some from newer versions..

In our attempt to cover both Jessie and Stretch configurations. I'll run some tests and see if a complete re-do of the config is required.

Thanks for the heads up 👍