MichaIng/DietPi #1751: Blue text on update

fedekrum, 2018-05-05:
#### Required Information:
- DietPi Version | v6.6
- SBC Device     | RPi 3 Model B (armv7l) (index=3)
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Update

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages will be upgraded:
  libraspberrypi-bin libraspberrypi0 perl-base raspberrypi-bootloader
  raspberrypi-kernel
5 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 38.9 MB of archives.
After this operation, 519 kB of additional disk space will be used.
Get:2 https://archive.raspberrypi.org/debian stretch/main armhf raspberrypi-kernel armhf 1.20180417-1 [33.0 MB]
Get:1 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf perl-base armhf 5.24.1-3+deb9u3 [1,196 kB]
Err:2 https://archive.raspberrypi.org/debian stretch/main armhf raspberrypi-kernel armhf 1.20180417-1
  GnuTLS recv error (-110): The TLS connection was non-properly terminated.
Get:3 https://archive.raspberrypi.org/debian stretch/main armhf libraspberrypi-bin armhf 1.20180417-1 [335 kB]
Err:3 https://archive.raspberrypi.org/debian stretch/main armhf libraspberrypi-bin armhf 1.20180417-1
  GnuTLS recv error (-110): The TLS connection was non-properly terminated.
Get:4 https://archive.raspberrypi.org/debian stretch/main armhf libraspberrypi0 armhf 1.20180417-1 [837 kB]
Get:5 https://archive.raspberrypi.org/debian stretch/main armhf raspberrypi-bootloader armhf 1.20180417-1 [3,527 kB]
Fetched 5,560 kB in 3min 15s (28.5 kB/s)
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20180417-1_armhf.deb  GnuTLS recv error (-110): The TLS connection was non-properly terminated.
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-bin_1.20180417-1_armhf.deb  GnuTLS recv error (-110): The TLS connection was non-properly terminated.
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

MichaIng, 2018-05-05:
@fedekrum 
Thanks for your report.

Can you verify that dirmngr is installed: `apt install dirmngr`
Also try to clean your APT cache:
```
apt clean
rm -R /var/cache/apt/archives/*
```

Fourdee, 2018-05-07:
Unable to replicate on v6.7 RPi 3 fresh install.

@fedekrum 

Please also verify the following package is installed:
```
apt-get install raspbian-archive-keyring
```

Fourdee, 2018-05-30:
Closing due to no response and unable to replicate in our tests.

closed 2018-05-30 by Fourdee