MichaIng/DietPi #1752: Feature Request: PPPOE Server with DHCP

bsidio, 2018-05-06:
It would be great if DIET PI can be used as a router.

MichaIng, 2018-05-06:
@bsidio
Thanks for your request.

Of course as modem  device you require special hardware, but you can already setup your device as DNS server and WiFi gateway, e.g.: https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1207#p1207

Is this what you are looking for?

bsidio, 2018-05-06:
No, not modem just a pppoe Server since I get a wan connection type pppoe. It won't require any special hardware.
Just need to install https://packages.debian.org/search?keywords=pppoe

MichaIng, 2018-05-06:
@bsidio
Ah okay, to be true this is not my profession, I think @k-plan might be able to help you best with what is possible there.

Not sure if the current DietPi network settings are compatible with this (without manual adjustments that could then break proper work of DietPi-Config).

But sounds interesting if its possible/compatible to implement into DietPi.

k-plan, 2018-05-07:
@bsidio 

> Feature Request: PPPOE Server with DHCP
> ....
> No, not modem just a pppoe Server since I get a wan connection type pppoe.

So I think, you need a PPPoE Client , not a PPPoE Server for your device.
And you need two different physical network interfaces, eg. `eth0` and `eth1` or `eth0` and `wlan0` or `wlan0` and `wlan1`, etc .

>  I think @k-plan might be able to help you best with what is possible there.

Yes possible, if you have to interface (`ethernet` or `WiFi`) : 

- https://blog.confirm.ch/using-pppoe-on-linux/

- https://wiki.debian.org/PPPoE

> Not sure if the current DietPi network settings are compatible with this (without manual adjustments that could then break proper work of DietPi-Config).

Yes, correct. After adding and edit `/etc/network/interfaces` for `pppoe client` use, you can never touch `dietpi-config`  -> `Network` at this moment. Sorry.

I personally will prefer, if I really need a real PPPoE Server or PPPoE Client for my SBC ,  especially if this has not two different  physical network interfaces, some cheap devices like this, powered by USB from the SBC:

https://mikrotik.com/product/RB941-2nD
https://mikrotik.com/product/RB931-2nD
https://mikrotik.com/product/RBmAPL-2nD
https://mikrotik.com/product/RBcAPL-2nD-307

Cost like a good supported WiFi USB or Ethernet USB stick and work much better for PPPoE Client and if you need it, as well as PPPoE Server for Clients (ROS Level 4 licence: until 200 PPPoE tunnels)

https://wiki.mikrotik.com/wiki/Manual:Interface/PPPoE

MichaIng, 2021-05-27:
`/etc/network/interfaces.d/pppoe.conf` could be used currently to setup the WAN-side interface, and e.g. `/etc/network/interfaces.d/lan.conf` to configure then LAN-side interface. Then `/etc/network/interfaces` should be cleared aside of the first directive which includes files from `/etc/network/interfaces.d/`. `dietpi-config` does not touch those drop-in configs, and will show Ethernet and WiFi both as disabled, which is fine.

More flexible NAT/hotspot setups are planned as part of the network rework project. Adding a PPPoE client is a consequent next step, so I'll add it to that project.