MichaIng/DietPi #1753: DietPi failed to install with Vmware image

dietpiuser, 2018-05-06:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | v6.4
- SBC Device     | Virtual Machine (x86_64) (index=20)
- Distro         | stretch (index=4)
- Command        | Connection test: https://deb.debian.org/debian/
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2018-03-11 18:09:58--  https://deb.debian.org/debian/
Resolving deb.debian.org (deb.debian.org)... 130.89.148.14, 128.31.0.62, 5.153.231.4, ...
Connecting to deb.debian.org (deb.debian.org)|130.89.148.14|:443... connected.
ERROR: The certificate of ‘deb.debian.org’ is not trusted.
ERROR: The certificate of ‘deb.debian.org’ is not yet activated.
The certificate has not yet been activated
```

#### Did you submit a ```dietpi-bugreport```?
   │ DietPi-Bugreport: Connection test: dietpi.com
            │  - exit_code = 5
            │  - VERSION:v6.4 | HW_MODEL:20 | HW_ARCH:10 | DISTRO:4
            │
            │ Log file contents:
            │ Spider mode enabled. Check if remote file exists.
            │ --2018-03-11 18:12:49--  http://dietpi.com/
            │ Resolving dietpi.com (dietpi.com)... 185.101.93.93
            │ Connecting to dietpi.com (dietpi.com)|185.101.93.93|:80... connected.
            │ HTTP request sent, awaiting response... 301 Moved Permanently
            │ Location: https://dietpi.com/ [following]
            │ Spider mode enabled. Check if remote file exists.
            │ --2018-03-11 18:12:50--  https://dietpi.com/
            │ Connecting to dietpi.com (dietpi.com)|185.101.93.93|:443... connected.
            │ ERROR: The certificate of ‘dietpi.com’ is not trusted.
            │ ERROR: The certificate of ‘dietpi.com’ is not yet activated.

#### Extra details:

Trying to deploy DietPi Vmware image and is failing with above certificate error.
[image]

MichaIng, 2018-05-06:
@dietpiuser
Thanks for your report. 

Hmm that's strange, I can't access https://deb.debian.org/debian directly here, but https://deb.debian.org/debian/dists/ works and then going back to parent folder. 

Perhaps something changed in the way the redirector works, at least for some then chosen mirrors. Don't know. Might be we need to change the way we check for repo availability.

Please try to change your APT mirror (e.g. to ftp.debian.org/debian) with dietpi-config or manually within /etc/apt/sources.list and rerun dietpi-software for first run setup.

dietpiuser, 2018-05-06:
Hi,

THanks for your response. I've updated the APT mirror list and it goes through but hit a new error. See attached.
[image]
[image]

MichaIng, 2018-05-06:
@dietpiuser
Ah yeah, the security mirror is not touched by dietpi-config as its URL structure is different on different main mirrors.

As it is related to certificate issues, please try to reinstall the debian keyring and ca-certificates, at least it's worth trying:
apt install --reinstall debian-archive-keyring ca-certificates

Otherwise, as the above most properly will not help, please manually edit `nano /etc/apt/sources.list` and replace the security/updates line with:
`deb http://security.debian.org/debian-security stretch/updates main contrib non-free`

I will check deb.debian.org mirrordirector at home later as well, but I guess it's a local issue with your close final mirror, pointed to.

Fourdee, 2018-05-06:
@MichaIng 

> Perhaps something changed in the way the redirector works,

Wouldnt be the first time:
https://github.com/Fourdee/DietPi/issues/669#issuecomment-269344844

Fourdee, 2018-05-06:
@MichaIng 

I'll setup a VM and try to replicate.

Fourdee, 2018-05-06:
Ok we can add ```--no-check-certificate``` to our wget G_ command.

Fourdee, 2018-05-06:
NTP sync is the issue:
```
root@DietPi:~# date
Sun 11 Mar 17:23:23 GMT 2018
```

Fourdee, 2018-05-06:
I'll recreate the image from scratch, issue should be resolved with latest version of DietPi + PREP.

Fourdee, 2018-05-06:
Nope, PC hard locking every 20-30 minutes. Going to try and start a Go Fund Me to replace it.

Fourdee, 2018-05-06:
#### Fundraiser:
https://www.gofundme.com/dietpi

dietpiuser, 2018-05-06:
Hi guys,

After manually adding the below lines in /etc/apt/sources.list as a workaround, it worked and i was able to continue the installation.

>>> deb http://security.debian.org/debian-security stretch/updates main contrib non-free

dietpiuser, 2018-05-07:
HI guys,

I'm trying to add an IPtable entry and it fails with the below error:

root@DietPi:~# /usr/bin/sudo /usr/share/bash-completion/completions/iptables -A INPUT -p tcp --destination-port 443 -j REJECT --reject-with tcp-reset
sudo: /usr/share/bash-completion/completions/iptables: command not found

Tried up update IPtable and I can the below error:

root@DietPi:~# apt-get update && apt-get install iptables
Err:1 http://security.debian.org/debian-security stretch/updates InRelease
  Temporary failure resolving 'security.debian.org'
Err:2 http://ftp.debian.org/debian stretch InRelease
  Temporary failure resolving 'ftp.debian.org'
Err:3 http://ftp.debian.org/debian stretch-updates InRelease
  Temporary failure resolving 'ftp.debian.org'
Err:4 http://ftp.debian.org/debian stretch-backports InRelease
  Temporary failure resolving 'ftp.debian.org'
Reading package lists... Done
W: Failed to fetch http://ftp.debian.org/debian/dists/stretch/InRelease  Temporary failure resolving 'ftp.debian.org'
W: Failed to fetch http://ftp.debian.org/debian/dists/stretch-updates/InRelease  Temporary failure resolving 'ftp.debian.org'
W: Failed to fetch http://security.debian.org/debian-security/dists/stretch/updates/InRelease  Temporary failure resolving 'security.debian.org'
W: Failed to fetch http://ftp.debian.org/debian/dists/stretch-backports/InRelease  Temporary failure resolving 'ftp.debian.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  libip6tc0 libiptc0 libxtables12
The following NEW packages will be installed:
  iptables libip6tc0 libiptc0 libxtables12
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 489 kB of archives.
After this operation, 1,910 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Err:1 http://ftp.debian.org/debian stretch/main amd64 libip6tc0 amd64 1.6.0+snapshot20161117-6
  Temporary failure resolving 'ftp.debian.org'
Err:2 http://ftp.debian.org/debian stretch/main amd64 libiptc0 amd64 1.6.0+snapshot20161117-6
  Temporary failure resolving 'ftp.debian.org'
Err:3 http://ftp.debian.org/debian stretch/main amd64 libxtables12 amd64 1.6.0+snapshot20161117-6
  Temporary failure resolving 'ftp.debian.org'
Err:4 http://ftp.debian.org/debian stretch/main amd64 iptables amd64 1.6.0+snapshot20161117-6
  Temporary failure resolving 'ftp.debian.org'
E: Failed to fetch http://ftp.debian.org/debian/pool/main/i/iptables/libip6tc0_1.6.0+snapshot20161117-6_amd64.deb  Temporary failure resolving 'ftp.debian.org'
E: Failed to fetch http://ftp.debian.org/debian/pool/main/i/iptables/libiptc0_1.6.0+snapshot20161117-6_amd64.deb  Temporary failure resolving 'ftp.debian.org'
E: Failed to fetch http://ftp.debian.org/debian/pool/main/i/iptables/libxtables12_1.6.0+snapshot20161117-6_amd64.deb  Temporary failure resolving 'ftp.debian.org'
E: Failed to fetch http://ftp.debian.org/debian/pool/main/i/iptables/iptables_1.6.0+snapshot20161117-6_amd64.deb  Temporary failure resolving 'ftp.debian.org'
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

Any clues?

MichaIng, 2018-05-07:
@Fourdee
Hmm I don't like the idea to not check certificates. I mean HTTPS is pretty standard for many repos and deb.debian.org is standard repo on Debian. Should work fine, even that indeed we see quite often issues. Of course correct system time needs to be set for correct cert check. So we could offer checking NTP status on APT errors and in case of NTP errors offer to adjust network and NTP settings via dietpi-config.
Maybe the error codes of APT and NTP status give some hint as well, if it's a network or SSL or DNS resolving issue, allowing us to add the right solution offer. Or just offer all we can think of.

For me, changing APT mirror instead of disabling SSL check is the better last resort offer. SSL loses it's benefits completely if we allow to accept a fake certificate, possibly by wrong resolving due to harmful DNS entry or something.

@dietpiuser
Jep `iptables` is by default not installed on DietPi images.

Is network generally up?
`ping 8.8.8.8`

Please check your DNS server:
`cat /etc/resolv.conf`
How did you setup your network? I guess bridged networking for VM as it is by default and DHCP within the VM itself?
Is the IP address correctly received?
`ip a`

In case of DHCP via router, the DNS server should be given by router. If this somehow does not work, but IP was recieved correctly (plus internet access is there via ping IP), you could try to switch to static IP via dietpi-config, copy current settings, but change the DNS server manually to an external one, e.g. `8.8.8.8` fur Google DNS.

Then check again if it was added to resolv.conf and then name resolving/APT works.

But generally with DHCP via router everything should work by default, so this would need separate investigation

Fourdee, 2018-05-07:
@MichaIng 

> For me, changing APT mirror instead of disabling SSL check is the better last resort offer. SSL loses it's benefits completely if we allow to accept a fake certificate, possibly by wrong resolving due to harmful DNS entry or something.

Yep, agree 👍 

I'am pretty sure an updated image will resolve it, NTP check seems off during the 1/30 checks (faster than it should be). Should have my new PC arriving Wednesday, so I should be able to redo the image this week 😃

k-plan, 2018-05-07:
@Fourdee 

> Should have my new PC arriving Wednesday, so ...

Wow ... so quick?  :+1:  :heart:   :wink: 

Thanks to all the supporter on https://www.gofundme.com/dietpi

Fourdee, 2018-05-07:
@k-plan 

> Wow ... so quick? 👍 ❤️ 😉

Yep, indeed. I woke up this morning and was shocked, surprised and blown away by the support. I honestly did not expect it.

> Thanks to all the supporter on https://www.gofundme.com/dietpi

Indeed, unbelievable support, I've thanked everyone personally via gofundme and twitter.

New PC!!! 😄 
https://twitter.com/DietPi_/status/993513382132305920

dietpiuser, 2018-05-07:
OK, the issue was I updated the upstream DNS on the box to cloudflare DNS and it could not resolve the FTP repo because i had also put in the IPv6 DNS entries which is not enabled on DietPi by default.

I changed it back to Google upstream DNS and it worked fine.

I was able to upgrade iptables for IPv4 and add the iptable entries.

I'm running Pi-hole on DietPi.

So it;s looking fine now.

I'll wait until a new update is available to change the repo back to HTTP.

Fourdee, 2018-05-09:
PC now arriving Friday https://twitter.com/DietPi_/status/994222442460663808 ☹️ 

Will get the VMware image done over the weekend.

MichaIng, 2018-05-09:
I'll set milestone to v6.9 with goal to extend our APT and NTP error handling.

Fourdee, 2018-05-09:
@MichaIng 

> Use the time until release for fixing issues with current code?

Yep, think everything is nearly completed? Only config inject outstanding (although its been working fine in all testing i've done)?
And https://github.com/Fourdee/DietPi/issues/1740

MichaIng, 2018-05-09:
@Fourdee 
Jep nothing urgent opened from my side. `G_CONFIG_INJECT` does now work secure (and output as expected) for my impression, if escaping rules are followed. I implemented error handling after `sed`, to make bug tracking easier for us, just in case (see my last commits). I just didn't find a way to check for error within `if` condition (grep) command, as error code $? is not preserved preserved. If e.g. `(` is not escaped, `grep -E` throws an error, complaining about mission `)`, but `sed -E` just takes it literally, if no closing parenthesis found. Really no bid issue. Would be just nice as well handle error and quit function, if grep throws error, instead of going on with `else`.

Hmm maybe it's possible to do save error code directly within condition? 🤔

About file system check:
- I tested as well on ext4, worked fine.
- Could have some improvement: Offer to automatically un- and remount the drive, as now user needs to to this manually. Scan on reboot for boot and root partition. But all this can be done in v6.9.
- Ahh, but just tested for 🈴 F2FS: Different options... No dry-run possible, it seams...
```
root@DietPi:/var/log# fsck -h /dev/sdb1
fsck from util-linux 2.29.2
fsck.f2fs: invalid option -- 'h'
        Error: Unknown option ?

Usage: fsck.f2fs [options] device
[options]:
  -a check/fix potential corruption, reported by f2fs
  -d debug level [default:0]
  -f check/fix entire partition
  -p preen mode [default:0 the same as -a [0|1]]
  -t show directory tree [-d -1]
```
- I will switch to other issue with this, to test all supported file systems.

Fourdee, 2018-05-17:
#### VMware image updated, NTPD sync and APT now resolved:
https://dietpi.com/downloads/images/DietPi_VMWare-x86_64-Stretch.7z

With many thanks to all the donators who helped fund my new PC, allowing me to continue VM support.
Its fast, freaky fast 😃 
https://twitter.com/DietPi_/status/996841556463292416
https://twitter.com/DietPi_/status/996847866273464322

adamotte, 2018-06-25:
Can you apply the same fix to VirtualBox image ?

MichaIng, 2018-06-25:
@adamotte 
You mean the NTP/APT issue faced by TO?

Actually as DietPi automatically performs updates on first run, everything should be working. APT generally should work, a mirror fix is applied with v6.6 (anyway just affected RPi) and NTP was fixed (we switched from NTP to systemd-timesyncd) with v6.9. Or did you face issues during/after first run of the VBox image?
- [ ] Will do some test myself
I will create some fresh VBox images soon anyway, at latest with v6.10 release, but the incremental first run patches need to work flawlessly anyway.
_______________
**RPi only:** Reviewing our [patch_file](https://github.com/Fourdee/DietPi/blob/testing/dietpi/patch_file) code I just could imagine the following:
- `dietpi-update` performs an `apt-get update` since v6.3 (https://github.com/Fourdee/DietPi/commit/c0205705202e45592dd3a95bf8283ca65d2081c7#diff-c613a85da508fb885b67c34f9661e243), before running the incremental patch file. I am not sure if `raspberrypi-sys-mods` needs `apt-get dist-upgrade`, but if not, then our default Raspbian mirror director that time (https://www.mirrorservice.org/sites/archive.raspbian.org/raspbian) would have been broken, at least if the raspberrypi devs did not fix their update script according to our comment: https://github.com/RPi-Distro/raspberrypi-sys-mods/commit/f2db61056a1b8abfc8ace68c39a56d584938bff1#commitcomment-28336410
- Our patch then on v6.5 -> v6.6 addresses this issue, but as it needs to assure the `raspberrypi-sys-mods` update is installed, it runs another `apt-get dist-upgrade`. This could then fail, if this is already in place and mirror broken by this. Or do broken APT sources just affect `apt-get update`?
- [ ] Test needed with older `raspberrypi-sys-mods` installed, or was this already done?

sarsw, 2018-08-01:
Did anyone see a new DietPi VirtualBox image? I downloaded 6.4 a couple of days ago from DietPi and it fails to update. Well, actually dietpi-update eventually said I'm on 6.12 but there were lots of errors to acknowledge along the way, like the screen shots above. "apt-get install screen" fails so I think the update is not right either.  I also tried the VMware DietPi image but VMware Player 12.x.x said it's the wrong vmx file format so I downgraded to VMware Player 12.0.0 (per the DietPi site recommendation) and it still said the image VMX file was wrong so I tried the VirtualBox image instead with VMWare Player... and that's where I am now. I tried & dropped VirtualBox because I wanted it to autostart my DietPiVM in Windows and that was proving problematic with the third party service I tried.

shred86, 2018-09-11:
Yeah, this is occurring on VirtualBox with v6.12 (latest version downloaded from the website). I had to change the NTP server to get the date/time to finally update in order to install dietpi.

Fourdee, 2018-09-11:
@shred86 

Thanks for the report 👍 

We'll reopen for investigations and local testing, see if we can replicate.

Fourdee, 2018-09-11:
Confirmed same issue with 1st run connection test fail, due to date + cert out in pre-v6.14 images. We'll update the images.

I'll mark this as closed in favor of https://github.com/Fourdee/DietPi/issues/2026

closed 2018-09-11 by Fourdee

shred86, 2018-09-11:
@Fourdee 

I'm using the VirtualBox image. I just tried re-downloading it from the site but it looks like it hasn't been updated yet. I'll definitely test it out tonight if you're able to update the image by then. Thanks for the work!

Fourdee, 2018-09-12:
@shred86 

> I'm using the VirtualBox image. 

Ah apologies, I did not see correctly.

The VirtualBox is yet to be updated, please see here for status:
https://github.com/Fourdee/DietPi/issues/2026#issuecomment-414383868