MichaIng/DietPi #1754: DietPi-Software | Transmission, review/clean configs across distros

Fourdee, 2018-05-06:
> https://github.com/Fourdee/DietPi/issues/1750#issuecomment-386908660

bokiroki, 2018-05-07:
@Fourdee i installed transmission daemon with apt-get install and i get same legacy options..so debian package manager decided to use them..and after i saw in transmission manual there are only few legacy options and they are not that important.. so it's not big deal.. you can close this issue if you want..

Fourdee, 2018-05-07:
@bokiroki 

> i installed transmission daemon with apt-get install and i get same legacy options..so debian package manager decided to use them..and after i saw in transmission manual there are only few legacy options and they are not that important.. so it's not big deal.. you can close this issue if you want..

Thanks for the heads up 👍 

No worries, I need to recheck our Transmission installation anyway and see if we can tweak it. I'll leave this open for now.

Fourdee, 2018-05-25:
```
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring Transmission: bittorrent server with web interface (c)

[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "cache-size-mb": 295,
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "download-queue-size": 3,
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "peer-limit-global": 40,
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "max-peers-global": 40,
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "upload-slots-per-torrent": 5,
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "port-forwarding-enabled": true,
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "encryption": 2,
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "idle-seeding-limit": 1,
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "idle-seeding-limit-enabled": true,
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "ratio-limit": 1.1,
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "ratio-limit-enabled": true,
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "trash-original-torrent-files": true,
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "download-dir": "/mnt/dietpi_userdata",
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "rpc-password": "dietpi",
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "rpc-username": "root",
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "rpc-whitelist-enabled": false,
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "message-level": 0,
[ SUB1 ] DietPi-Services > dietpi_controlled
[  OK  ] DietPi-Services | dietpi_controlled : transmission-daemon
[  OK  ] DietPi-Services | dietpi_controlled : cron
[  OK  ] DietPi-Software | DietPi-Software installation completed.
[ SUB1 ] DietPi-Services > start
[  OK  ] DietPi-Services | start : transmission-daemon
[  OK  ] DietPi-Services | start : cron
[ SUB2 ] DietPi-Process_tool > Apply
[  OK  ] DietPi-Process_tool | Completed
```

Fourdee, 2018-05-25:
Completed.

closed 2018-05-25 by Fourdee