MichaIng/DietPi #1755: Install directly on a macbook?

gieljnssns, 2018-05-07:
Is it possible to install dietpi directly on a macbook air.

MichaIng, 2018-05-07:
@gieljnssns
The question is if default Debian runs on MacBook. Found this: https://wiki.debian.org/MacBook#Debian_installation

It is not recommended but seems to work with the netboot installer, that we also use for our x86 images. 

You can simply try to use our x86 EFI image. Otherwise manually install Debian with the netboot installer and run our preparation script on it.

No guarantee, as I don't remember somebody reporting that it works, would need review/search within forum and github issues.

If it finally works fine, would be indeed nice to know and could be added as info to our x86 image and preparation script for users.

SuBLiNeR, 2018-05-07:
Just to be clear, it is not as easy as described at the link @MichaIng posted. Lot's of ppl say that there is still a overheating problem on older releases of MacBookAir's.

If you really like to do this project, i suggest you to carefully read this HowTo and than decide if you really want to go these steps: https://linuxnewbieguide.org/how-to-install-linux-on-a-macintosh-computer/

gieljnssns, 2018-05-07:
https://github.com/Fourdee/DietPi/issues/1171#issuecomment-336522021
This method isn't working, the iso isn't bootable (etcher)

MichaIng, 2018-05-09:
@gieljnssns 
Thanks for testing this 👍.

May I ask how you generally managed to install Debian on Mac?
You followed the method linked by @SuBLiNeR or you had already another method working before?

Would be quite interesting, if possible anyhow or without too much needed adjustments to our scripts, to produce some guide or even image from this experience, for Mac users.

gieljnssns, 2018-05-09:
> May I ask how you generally managed to install Debian on Mac?

I've installed Mac OS X Yosemite (no System integrity protection)
Followed this guide:
https://wiki.debian.org/MacBook#Debian_installation
I took the "debian-9.4.0-amd64-netinst.iso" from here https://cdimage.debian.org/debian-cd/current/amd64/iso-cd/
When there is an error in the installation about some network firmware; I found this http://commandlinefanatic.com/cgi-bin/showarticle.cgi?article=art048 and i took the "linux-firmware-nonfree_1.11_all.deb" from here http://www.filewatcher.com/m/linux-firmware-nonfree_1.11_all.deb.3073110-0.html
Instead of making 2 partitions, I made a lot off free space on my drive and took the option: Guided use the largest free space on disk.
I let install the GRUB bootloader and now Debian is the first OS that is starting.
I can still start macOS by holding "option" when starting up

gieljnssns, 2018-05-09:
Then I'm following this guide https://github.com/Fourdee/DietPi/issues/1285#issue-280771944
When you log in using ssh and your user isn't is not in the sudoers file ```sudo usermod -a -G sudo user```
log out and log back in
While running DietPi installer I took "generic device"
"Keep Wifi files"
"Stretch"
```sudo reboot```
Finnish DietPi installation...

MichaIng, 2018-05-09:
@gieljnssns 
Thanks for this testing. So DietPi after this steps finally works as expected?

Generic device is the most safe attempt, but native x86 (ID 21) should also work, I think. Could you post the resolut of `dpkg -l` here, so I can check for installed kernel/firmware packages? There might be something different then on native PCs.

Btw. If you create an own user (and do not use `root` or `dietpi`), then on current version (v6.7) the login scripts are missing. This is already implemented with v6.8, should should be released today or soon (@Fourdee ?).

If not, then you could copy `/home/dietpi/.bashrc` to `/home/<username>/` for now, but after v6.8 update, Do `sed -i '/DietPi/d' /home/<username>/.bashrc` to remove our entries again, as the related scripts are executed via `/etc/profile.d/dietpi-login` then, thus for all users with sudo permissions 🙂.

gieljnssns, 2018-05-09:
```
root@DietMini:~# dpkg -l
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                    Version                  Architecture             Description
+++-=======================================-========================-========================-===================================================================================
ii  adduser                                 3.115                    all                      add and remove users and groups
ii  apt                                     1.4.8                    amd64                    commandline package manager
ii  apt-transport-https                     1.4.8                    amd64                    https download transport for APT
ii  apt-utils                               1.4.8                    amd64                    package management related utility programs
ii  avahi-daemon                            0.6.32-2                 amd64                    Avahi mDNS/DNS-SD daemon
ii  base-files                              9.9+deb9u4               amd64                    Debian base system miscellaneous files
ii  base-passwd                             3.5.43                   amd64                    Debian base system master password and group files
ii  bash                                    4.4-5                    amd64                    GNU Bourne Again SHell
ii  bash-completion                         1:2.1-4.3                all                      programmable completion for the bash shell
ii  bc                                      1.06.95-9+b3             amd64                    GNU bc arbitrary precision calculator language
ii  bind9-host                              1:9.10.3.dfsg.P4-12.3+de amd64                    Version of 'host' bundled with BIND 9.X
ii  bsdutils                                1:2.29.2-1+deb9u1        amd64                    basic utilities from 4.4BSD-Lite
ii  bzip2                                   1.0.6-8.1                amd64                    high-quality block-sorting file compressor - utilities
ii  ca-certificates                         20161130+nmu1            all                      Common CA certificates
ii  console-setup                           1.164                    all                      console font and keymap setup program
ii  console-setup-linux                     1.164                    all                      Linux specific part of console-setup
ii  coreutils                               8.26-3                   amd64                    GNU core utilities
ii  cpio                                    2.11+dfsg-6              amd64                    GNU cpio -- a program to manage archives of files
ii  crda                                    3.18-1                   amd64                    wireless Central Regulatory Domain Agent
ii  cron                                    3.0pl1-128+deb9u1        amd64                    process scheduling daemon
ii  curl                                    7.52.1-5+deb9u5          amd64                    command line tool for transferring data with URL syntax
ii  dash                                    0.5.8-2.4                amd64                    POSIX-compliant shell
ii  dbus                                    1.10.26-0+deb9u1         amd64                    simple interprocess messaging system (daemon and utilities)
ii  debconf                                 1.5.61                   all                      Debian configuration management system
ii  debian-archive-keyring                  2017.5                   all                      GnuPG archive keys of the Debian archive
ii  debianutils                             4.8.1.1                  amd64                    Miscellaneous utilities specific to Debian
ii  diffutils                               1:3.5-3                  amd64                    File comparison utilities
ii  dirmngr                                 2.1.18-8~deb9u1          amd64                    GNU privacy guard - network certificate management service
ii  dmsetup                                 2:1.02.137-2             amd64                    Linux Kernel Device Mapper userspace library
ii  docker-ce                               18.04.0~ce~3-0~debian    amd64                    Docker: the open-source application container engine
ii  dosfstools                              4.1-1                    amd64                    utilities for making and checking MS-DOS FAT filesystems
ii  dpkg                                    1.18.24                  amd64                    Debian package management system
ii  dropbear-bin                            2016.74-5                amd64                    lightweight SSH2 server and client - command line tools
ii  dropbear-run                            2016.74-5                all                      lightweight SSH2 server and client - startup scripts
ii  e2fslibs:amd64                          1.43.4-2                 amd64                    ext2/ext3/ext4 file system libraries
ii  e2fsprogs                               1.43.4-2                 amd64                    ext2/ext3/ext4 file system utilities
ii  efibootmgr                              14-2                     amd64                    Interact with the EFI Boot Manager
[... cut ...]
ii  xkb-data                                2.19-1+deb9u1            all                      X Keyboard Extension (XKB) configuration data
ii  zip                                     3.0-11+b1                amd64                    Archiver for .zip files
ii  zlib1g:amd64                            1:1.2.8.dfsg-5           amd64                    compression library - runtime
ii  zlib1g-dev:amd64                        1:1.2.8.dfsg-5           amd64                    compression library - development
```

gieljnssns, 2018-05-09:
> Thanks for this testing. So DietPi after this steps finally works as expected?

I think so, I use this for installing [Hassio](https://www.home-assistant.io/hassio/)
https://community.home-assistant.io/t/ive-installed-hassio-on-an-old-mac-mini-no-vm/52856
My raspberry pi isn't fast enough.

MichaIng, 2018-05-09:
@gieljnssns
Nice, I don't see any difference to usual x86_64 systems, thus it should be possible to choose device ID 21 for preparation script as well. But should not result very differently, thus leave it as it is now. 

DietPi-Software also offers Home assistant automated installation by the way. It does not include much optimizations but is easier of course than doing all the steps oneself.

gieljnssns, 2018-05-09:
> DietPi-Software also offers Home assistant automated installation by the way.

I think this is just Home assistant not Hassio

MichaIng, 2018-05-09:
@gieljnssns 
Ah sorry, though "Hassio" is short form of **H**ome-**ASS**istant.**IO** 🤣.

MichaIng, 2018-05-12:
@gieljnssns
Okay, as far as I could retrace now, you have 3 partition on the drive: OS X, Debian OS and the EFI partition? 

As Debian boots by default and grub-efi-amd64 is installed, the bootloader seems to be the one from Debian, not OS X anymore?

The question is then why our x86 EFI image is not working, what differences are there. Might be some special firmware for Macs, that Debian installer skips on non-Macs, although the linux-firmware-nonfree package is by default installed.

gieljnssns, 2018-05-18:
> Okay, as far as I could retrace now, you have 3 partition on the drive: OS X, Debian OS and the EFI partition?

As Debian boots by default and grub-efi-amd64 is installed, the bootloader seems to be the one from Debian, not OS X anymore?

How can I check this?

> The question is then why our x86 EFI image is not working, what differences are there.

I burn my usb and sd with Etcher, and etcher says the image isn't bootable

But I've had another problem, when I did the update to v6.8 I've lost all my docker containers (Hassio)
Is this normal?

MichaIng, 2018-05-19:
@gieljnssns 
> I burn my usb and sd with Etcher, and etcher says the image isn't bootable

Ahh, sorry didn't read careful enough earlier, I thought the successful flashed image does not boot on Mac.

Yeah, of course, now I found the issue. It seems not possible to directly flash a bootable EFI image. Thus we offer our image bundles with Clonezilla to flash on e.g. a USB drive, boot it from there, which will then allow to install DietPi on the target drive. You already linked the instructions above: https://github.com/Fourdee/DietPi/issues/1171#issuecomment-336522021

No idea why Etcher claims it being unbootable... However the instructions tell you to use Rufus instead, which is btw. less than 1 MB compared to Etcher more than 50 MB and Rufus is even a bid more flexible, providing additional feature like bad block testing etc. No idea why giant Etcher is promoted everywhere, just because it offers beautiful material design UI? Other topic...

So if you are in mood, it would be interesting if following the instructions with the UEFI image precisely, using Rufus for flashing on USB, boot and from Clonezilla flash on final HDD, works on Mac.

gieljnssns, 2018-05-19:
I think rufus is Windows only?

Invictaz, 2018-05-19:
@gieljnssns You have also Piwriter for Mac: https://sourceforge.net/projects/piwriter/

Did you try that?

MichaIng, 2018-05-19:
@gieljnssns 
Ah yes, didn't think about that 😜.

@gieljnssns 
👍 

Finally Ether also allows to ignore and continue. Only thing I don't know if it automatically chooses needed GPT partition then?

gieljnssns, 2018-05-21:
I've tried with; 

- Ether also allows to ignore and continue
- https://sourceforge.net/projects/piwriter/
- applepi-baker
- sudo dd command
- ...

 I can not boot with the x86 EFI image.

MichaIng, 2018-05-21:
@gieljnssns 
Hmm okay, since it boots fine on other systems, seems that Mac needs some specials to boot the stick. Looks like the method via Debian wiki you used is the way to go for now.

MichaIng, 2019-11-06:
We have a new x86_64 UEFI installer image which could be tested on Mac. It is based on Clonezilla as before, but since a year has been gone and Clonezilla as well as Debian (now Buster) has gone through a bunch of updates, it is worth giving it a try.

MichaIng, 2020-02-05:
I mark this issue as closed. Basically on MAC our UEFI installer should work. If not then the MAC in question has some non-default x86_64 attributes, which at least me cannot test or work around, due to lac of MAC device. If someone wants to create a MAC image, I would be happy to provide it via our download page as community image with credits. Basically it means, installing Debian via mini.iso or debootstrap and running DietPi-PREP on it:
- https://github.com/MichaIng/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh
- To create a Clonezilla installer image (required for UEFI, since no direct image flashing is possible, IMO), DietPi-Imager can be used from a running DietPi system: https://github.com/MichaIng/DietPi/blob/master/.meta/dietpi-imager

closed 2020-02-05 by MichaIng