MichaIng/DietPi #1759: RPi Zero W | Serial Terminal via UART did not work

k-plan, 2018-05-09:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version``` => DietPi v6.7 
- SBC Device (EG: RPi 3)? => RPi Zero W v1.3
- Power supply used (EG: 5v 1A RAVpower)?  [5.1V/2,5A T5875DV Stontronics](https://docs-emea.rs-online.com/webdocs/14ba/0900766b814ba6b1.pdf)
- SD card used (EG: Sandisk ultra)? => SanDisk 8GB
- Distro (EG: Jessie) | ```uname -a```? => Debian Stretch

#### Additional Information (if applicable):
- Software title? 
```
#Serial Console: Set to 1 if you require a serial console during 1st run of DietPi.
CONFIG_SERIAL_CONSOLE_ENABLE=1

```
- Can this issue be replicated on a fresh installation of DietPi?

- Yes!

#### Expected behaviour:

- Connect any [3,3V USB-to-UART converter](https://eckstein-shop.de/USB-zu-UART) to [RPi Pi: 1-8-10](https://pinout.xyz/pinout/uart) , open serial Terminal  with 115200-8-none-1-none and power up your RPi

#### Actual behaviour:

- Device boot up, but no serial terminal output

#### Steps to reproduce:

- Take a RPi Zero W, RPi 3B or RPi 3B+ and check it out

#### Extra details:

- It take a while, to get it to work. 

- It is a missing parameter in `/boot/config.txt` : 

```

# Enable UART
enable_uart=1

```

This problem is discribled only for RPi 3 so far, so it take some days, to find it. It should happen on all devices with onboard BT.

https://www.raspberrypi.org/forums/viewtopic.php?t=141195

http://www.netzmafia.de/skripten/hardware/RasPi/RasPi_Serial.html

> Änderungen beim Raspberry Pi Modell 3
> Das BCM2837 auf dem Raspberry Pi3 hat 2 UARTs (wie auch seine Vorgänger), jedoch zur Unterstützung Die Bluetooth-4.2-Funktionalität des voll ausgestatteten PL011-UART wurde von den Header-Pins auf den Bluetooth-Chip verschoben und der Mini-UART auf den Header-Pins 8 und 10 verfügbar gemacht. Der Mini-UART ist eine kleine Variante des ursprünglichen UART, die neben einem geringen Durchsatz auch nicht mehr so stabil ist wie zuvor. Die Baud-Rate des Mini-UART wird vom System-Takt (Videocore IV) abgeleitet. Das bedeutet, dass die Baud-Rate nicht mehr stabil bleibt, sondern je nach CPU-Auslastung stark schwankt. Deshalb schlagen die Entwickler vor, den Core-Takt fest auf 250 MHz einzustellen (siehe unten). 
>...
>Durch das Ändern der Datei` /boot/config.txt `kann die Abhängigkeit vom System Clock durch Hinzufügen der folgende Zeile am Ende der Datei fest eingestellt werden:
>`core_freq=250`
>Dies behebt das Timing-Problem und scheint wenig Einfluss auf andere Dinge zu haben. Die SPI-Taktfrequenz und ARM Timer sind ebenfalls abhängig vom System Clock.
Aus irgendeinem seltsamen Grund ist ab dem `Kernel 4.4.9 "DISABLE UART"` die Standard-Voreinstellung - vermutlich, um Konflikte mit Bluetooth zu verhindern. Um die serielle Schnittstelle zu aktivieren, müssen Sie `enable_uart=1` in der Datei `/boot/config.txt` setzen. Dies behebt angeblich auch die core_freq-Problematik, so dass der obige Eintrag ggf. nicht mehr notwendig ist. Wer sicher gehen will, schreibt auf jeden Fall beides in die Datei` /boot/config.txt`:
>
>`core_freq=250`
>`enable_uart=1`
>Wenn Sie Bluetooth nicht verwenden, ist es möglich, die beiden seriellen Ports im Devicetree zu vertauschen oder Bluetooth mittels Devicetree-Overlay ganz abzuschalten. Zum Vertauchen fügen Sie am Ende der Datei `/boot/config.txt `die Zeile
>`dtoverlay=pi3-miniuart-bt`
>hinzu. Nach einem Neustart funktioniert die UART-Zuordnung wieder wie in den Vorgängerversionen. Mittels pi3-disable-bt kann man Bluetooth komplett abschalten. Dann muss aber auch noch das bluetooth-Modem deaktiviert werden, was über systemctl mit dem folgenden Kommando erreicht wird:
`sudo systemctl disable hciuart`
> 

Have cross check with a old RPi B+:

- Serial terminal and `login` work without `enable_uart=1` only  `CONFIG_SERIAL_CONSOLE_ENABLE=1` is needed
- but it works as well with this parameter

RPi Zero W:

- only:  `CONFIG_SERIAL_CONSOLE_ENABLE=1` => No serial output

-  `CONFIG_SERIAL_CONSOLE_ENABLE=0` and  `enable_uart=1` => serial output until `login`

- `CONFIG_SERIAL_CONSOLE_ENABLE=1` and  `enable_uart=1`  => serial output and serial ` login ` possible

Fourdee, 2018-05-09:
@k-plan 

Thanks 👍 

I'll take a look when I can.

Fourdee, 2018-05-09:
@k-plan 

Great writeup, research, makes this much easier to implement/test 👍 

Notes:
> This allegedly also fixes the core_freq problem, so that the above entry may no longer be necessary. If you want to be sure, be sure to write both in the file /boot/config.txt:

we'll try adding ```enable_uart=1``` for all systems, then I'll test:
NB: will require a new image, applied during initial boot stage.
- RPi 3
- RPi 3B+
- RPi Zero W

Fourdee, 2018-05-10:
New image completed, with updated boot code. Seems to be fine, however, need to verify with serial connection:
[image]
https://dietpi.com/downloads/testing/DietPi_RPi-ARMv6-Stretch.7z

k-plan, 2018-05-11:
@Fourdee 

Testing `DietPi_v6.7_RPi-ARMv6-Stretch.img` on RPi Zero W (always fresh install):

- 1.)
Settings:

```
#-------Serial/UART-----
enable_uart=0
```
and

```
#Serial Console: Set to 1 if you require a serial console during 1st run of DietPi.
CONFIG_SERIAL_CONSOLE_ENABLE=1
```
- first phase inital boot => filesystem resize => NO serial output

- second phase inital boot => intital setup => NO serial output => you have to do inital setup via ssh => auto reboot

- second boot (after finish inital setup) => serial output of boot messages and login is possible

```
. . . 
[  OK  ] Started DietPi-Boot.
[  OK  ] Started DietPi-PostBoot.
[  OK  ] Started rc.local backwards compatibility.
         Starting Terminate Plymouth Boot Screen...
         Starting Hold until boot process finishes up...
[  OK  ] NTPD: time sync | Completed
[  OK  ] Root access verified.

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: start

Raspbian GNU/Linux 9 RPi-Zero-W ttyS0

[  OK  ] DietPi-Services | start : cron
[ SUB1 ] DietPi-Process_tool > Apply
[  OK  ] DietPi-Process_tool | Cron (894) : Nice      0
[  OK  ] DietPi-Process_tool | Cron (894) : Affinity  0
[  OK  ] DietPi-Process_tool | Cron (894) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | DHCP Client (631) : Nice      0
[  OK  ] DietPi-Process_tool | DHCP Client (631) : Affinity  0
[  OK  ] DietPi-Process_tool | DHCP Client (631) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Dropbear (260) : Nice      0
[  OK  ] DietPi-Process_tool | Dropbear (260) : Affinity  0
[  OK  ] DietPi-Process_tool | Dropbear (260) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Completed

 ───────────────────────────────────────
 DietPi     | 16:43 | Fri 11/05/18
 ───────────────────────────────────────
 v6.7       | RPi Zero W (armv6l)
 ───────────────────────────────────────
 eth0       | 192.168.88.95
 ───────────────────────────────────────
 Default Login:
 Username = root
 Password = dietpi

 Please login to continue

```

------------------------------

- 2.)
Settings:

```
#-------Serial/UART-----
enable_uart=1
```
and

```
#Serial Console: Set to 1 if you require a serial console during 1st run of DietPi.
CONFIG_SERIAL_CONSOLE_ENABLE=1
```

- first phase inital boot => filesystem resize => serial output

- second phase inital boot => intital setup => serial output, BUT NO login prompt, so it is impossible to login via serial terminal => you have to do inital setup via ssh => auto reboot => now  serial output again

- second boot (after finish inital setup) => serial output of boot messages and login is possible

----------------------------

If you want to have a serial console during 1st run of DietPi, we must do it the other way around.

- 1.) enable serial console here: https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L1111-L1112

-2.) `enable_uart=1` in `config.txt` per default : https://github.com/Fourdee/DietPi/blob/testing/config.txt#L77-L78

- 3.) leave `CONFIG_SERIAL_CONSOLE_ENABLE=0` as default, so serial console will be switched of during inital boot

It should be the same for the other devices: https://github.com/Fourdee/DietPi/blob/dfa9db43cea447fe741f590b4bda3ffc5088228e/dietpi/func/dietpi-set_hardware#L1423-L1496

At the moment, I only can check RPi-Zero-W. No RPi 3B, no RPi 3B+ or Odroids.

Fourdee, 2018-05-11:
@k-plan 

Ok, yep, we can enable serial console by default, then disable as needed during 1st run 👍

k-plan, 2018-05-11:
@Fourdee 

> Ok, yep, we can enable serial console by default, then disable as needed during 1st run

Yep, this is, I agree with that. :+1: 

[image]

Fourdee, 2018-05-11:
Ok, just need to update the RPi image, then i'll post link again.

Fourdee, 2018-05-11:
New image uploaded:
https://dietpi.com/downloads/testing/DietPi_RPi-ARMv6-Stretch.7z

Fourdee, 2018-05-12:
NB, need to redo image before setting live. Did not run through dietpi-update patches, few things not removed (eg: .bashrc entries etc).

k-plan, 2018-05-13:
@Fourdee 

> New image ready for testing:

Have do some test with this image.

- 1.) enabel serial terminal

```
#-------Serial/UART-----
#NB: enabled for 1st run only, if you want to keep this setting, please set CONFIG_SERIAL_CONSOLE_ENABLE=1 in dietpi.txt
enable_uart=1

#Serial Console: Set to 1 if you require a serial console.
#	NB: Serial console is always enabled by default for 1st run setup, then disabled afterwards, unless set below.
CONFIG_SERIAL_CONSOLE_ENABLE=1
```
Work as expected. You see all messages, can login via serial terminal and finish inital setup.

[image]

👍 

--------

- 2.) (default)

```
#-------Serial/UART-----
#NB: enabled for 1st run only, if you want to keep this setting, please set CONFIG_SERIAL_CONSOLE_ENABLE=1 in dietpi.txt
enable_uart=1

#Serial Console: Set to 1 if you require a serial console.
#	NB: Serial console is always enabled by default for 1st run setup, then disabled afterwards, unless set below.
CONFIG_SERIAL_CONSOLE_ENABLE=0
```

We see some ` [FAILED]` and `[ TIME ]` messages in seral terminal, put I thnk, it's okay.
After the last message appears, no login prompt, no login possible via serial terminal.

```
...

 INFO ] DietPi-Drive_Manager | Checking RootFS R/W access. Please wait...
[  OK  ] DietPi-Drive_Manager | RootFS R/W access.

 DietPi-Set_Hardware
─────────────────────────────────────────────────────
 Mode: serialconsole (disable)

[ INFO ] DietPi-Set_Hardware | Disabling known serial-getty services, please wait...
[FAILED] Failed to start serial-getty@tty.service.
See 'systemctl status serial-getty@tty.service' for details.
[FAILED] Failed to start serial-getty@ttyAMA0.service.
See 'systemctl status serial-getty@ttyAMA0.service' for details.
[FAILED] Failed to start serial-getty@ttySAC0.service.
See 'systemctl status serial-getty@ttySAC0.service' for details.
[FAILED] Failed to start serial-getty@ttyS0.service.
See 'systemctl status serial-getty@ttyS0.service' for details.
[FAILED] Failed to start serial-getty@ttyAMA1.service.
See 'systemctl status serial-getty@ttyAMA1.service' for details.
[FAILED] Failed to start serial-getty@ttySAC1.service.
See 'systemctl status serial-getty@ttySAC1.service' for details.
[FAILED] Failed to start serial-getty@ttyS1.service.
See 'systemctl status serial-getty@ttyS1.service' for details.
[FAILED] Failed to start serial-getty@ttyAMA2.service.
See 'systemctl status serial-getty@ttyAMA2.service' for details.
[FAILED] Failed to start serial-getty@ttySAC2.service.
See 'systemctl status serial-getty@ttySAC2.service' for details.
[FAILED] Failed to start serial-getty@ttyS2.service.
See 'systemctl status serial-getty@ttyS2.service' for details.
[  OK  ] DietPi-Set_Hardware | Setting in /DietPi/config.txt adjusted: enable_uart=0
[  OK  ] serialconsole disable | Completed
[  OK  ] Root access verified.
[ INFO ] DietPi-Drive_Manager | Checking RootFS R/W access. Please wait...
[  OK  ] DietPi-Drive_Manager | RootFS R/W access.

 DietPi-Set_software
─────────────────────────────────────────────────────
 Mode: apt-mirror (http://mirror.netcologne.de/raspbian/raspbian/)

[  OK  ] apt-mirror http://mirror.netcologne.de/raspbian/raspbian/ | Completed
         Stopping Raise network interfaces...
[  OK  ] Stopped Raise network interfaces.
dhclient: no process found
Internet Systems Consortium DHCP Client 4.3.5
Copyright 2004-2016 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/eth0/00:e0:4c:53:44:58
Sending on   LPF/eth0/00:e0:4c:53:44:58
Sending on   Socket/fallback
DHCPREQUEST of 192.168.0.104 on eth0 to 255.255.255.255 port 67
DHCPREQUEST of 192.168.0.104 on eth0 to 255.255.255.255 port 67
[ TIME ] Timed out waiting for device dev-ttySAC2.device.
[ TIME ] Timed out waiting for device dev-tty.device.
[ TIME ] Timed out waiting for device dev-ttyAMA1.device.
[ TIME ] Timed out waiting for device dev-ttyS1.device.
[ TIME ] Timed out waiting for device dev-ttyS2.device.
[ TIME ] Timed out waiting for device dev-ttySAC0.device.
[ TIME ] Timed out waiting for device dev-ttyAMA2.device.
[ TIME ] Timed out waiting for device dev-ttySAC1.device.
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 3
```

Fourdee, 2018-05-13:
@k-plan 

Legend, thanks for testing.

> [FAILED]

I'll check this, see if we can resolve, then redo the image and set live.

Fourdee, 2018-05-15:
New image live, not had a chance to check out the reported serial errors yet. 

Notes:
RPi disable those? uses kernel cmd line options and ignore service?

Fourdee, 2018-05-25:
Couldn't replicate this in my testing:
```
[ TIME ] Timed out waiting for device dev-ttySAC2.device.

```

```
root@DietPi:~# /DietPi/dietpi/func/dietpi-set_hardware serialconsole disable
[  OK  ] Root access verified.
[  OK  ] DietPi-Drive_Manager | RootFS R/W access verified.

 DietPi-Set_Hardware
─────────────────────────────────────────────────────
 Mode: serialconsole (disable)

[ INFO ] DietPi-Set_Hardware | Disabling known serial-getty services, please wait...
[  OK  ] DietPi-Set_Hardware | Setting in /DietPi/config.txt adjusted: enable_uart=0
[  OK  ] serialconsole disable | Completed
```

Appears the issue is now resolved. I'll mark this as completed.

closed 2018-05-25 by Fourdee