MichaIng/DietPi #1760: Asus TB | Onboard WiFi not functional

Fourdee, 2018-05-09:


MichaIng, 2018-05-10:
@SuBLiNeR 
Okay, so it's a kernel issue somewhere between 4.4.16 and 4.4.119.

Is there a chance for soon Tinker-OS with kernel upgrade to >= 4.4.120 or current kernel sources from their side?

SuBLiNeR, 2018-05-10:
@MichaIng It looks like they merged sources onto theirs git with those fixes and will release a more lately kernel and OS higher than 2.0.5.

Atm it is best to transform an armbian install to dietpi after self-compiling an armbian-image.

FaserF, 2018-05-10:
Same issue with my TB.

SuBLiNeR, 2018-05-10:
@FaserF Just use this image (https://dl.armbian.com/tinkerboard/archive/Armbian_5.41_Tinkerboard_Debian_stretch_default_4.4.119.7z)  and transform it into DietPi with the Prep-Script while connected to ethernet - **before** using the PREP-Script, use armbian-config to upgrade it to latest 'firmware'. finally than you could run the script as decribed -> #1285

MichaIng, 2018-05-10:
Okay, hard to say when next Tinker-OS release will be and we do not release DietPi images based on ARMbian.

Following possibilities:
- Revert kernel to 4.4.16/Tinker-OS 1.8, at least as additional offer for users that require WiFi. This one is not available any more: https://github.com/Fourdee/DietPi/issues/1444#issuecomment-362385108
- Manually build development kernel from: https://github.com/TinkerBoard/debian_kernel/tree/develop
- Find a workaround:

@SuBLiNeR 
Does WiFi work on clean Tinker-OS, as I can't believe them serving a faulty kernel as stable release?
In this case we would need to find about the difference of their WiFi implementation.

Then hope for soon release of Tinker-OS 2.0.6 with required fix 😄.

SuBLiNeR, 2018-05-10:
@MichaIng the funny thing about it is, as soon as you using anything below 2.0.4, it works! 

Second thing is what i tried was following, 4days ago.

Install TOS 2.0.1, boot it up only while on eth0. go into terminal and run the PREP–script ... everything runs well than. 
After Dietpi is booting twice times up, you should be able to switch from eth0 to wlan0 and it should wotk after a reboot again.

I than chosed for myself to try it with armbian.... it works. the main difference was, that my kernel is 4.4.126 than. 

Again, tha base image from armbian with a self compiled kernel aka 4.4.120 and up, works oob.

FaserF, 2018-05-14:
Issue still exists for me on the newest image.
After a reboot it says connected with 0 Mbit and I have no internet. Before the reboot it worked well.
I've tried disabling network, forgetting the wifi network, nothing worked for me.

Fourdee, 2018-05-15:
@FaserF 

> After a reboot it says connected with 0 Mbit and I have no internet. Before the reboot it worked well.
I've tried disabling network, forgetting the wifi network, nothing worked for me.

Strange, could be a range issue, have you tried moving the device closer to the WiFi router? What was the speed/dB levels before the reboot?

FaserF, 2018-05-15:
@Fourdee sorry I dont remember that. I am now using my TB with LAN and its working well.
It's just strange that this happened even with your new fix.

Fourdee, 2018-05-15:
@FaserF 

Confirmed bug, able to replicate, i'll reopen for investigations:
```
root@DietPi:~# dmesg | grep wlan
[    2.176996] usbcore: registered new interface driver rndis_wlan
[    6.182743] [WLAN_RFKILL]: Enter rfkill_wlan_init
[    6.182906] [WLAN_RFKILL]: Enter rfkill_wlan_probe
[    6.182932] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi_chip_type = ap6212
[    6.182937] [WLAN_RFKILL]: wlan_platdata_parse_dt: enable wifi power control.
[    6.182942] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi power controled by gpio.
[    6.182948] of_get_named_gpiod_flags: can't parse 'WIFI,poweren_gpio' property of node '/wireless-wlan[0]'
[    6.182954] of_get_named_gpiod_flags: can't parse 'WIFI,reset_gpio' property of node '/wireless-wlan[0]'
[    6.182976] of_get_named_gpiod_flags: parsed 'WIFI,host_wake_irq' property of node '/wireless-wlan[0]' - status (0)
[    6.182982] [WLAN_RFKILL]: wlan_platdata_parse_dt: get property: WIFI,host_wake_irq = 150, flags = 0.
[    6.182987] [WLAN_RFKILL]: rfkill_wlan_probe: init gpio
[    6.182993] [WLAN_RFKILL]: Exit rfkill_wlan_probe
[    8.498510] RTL871X: rtw_ndev_init(wlan0) if1 mac_addr=f0:03:8c:05:d7:05
[    9.884527] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   11.347086] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[   11.376316] RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
[   11.376424] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```

Fourdee, 2018-06-28:
Notes:
- Issue still exists with ASUS TB S

Fourdee, 2018-07-06:
Solution: ```G_CONFIG_INJECT '^8723bs' '8723bs' /etc/modules```

We'll patch for v6.12

Hax4us, 2018-07-11:
Thnx @fourdee ..I were waiting for this :)

Hax4us, 2018-07-11:
But it is not working for me @Fourdee I copy pasted your command in terminal but same results

Fourdee, 2018-07-11:
@Hax4us 

Strange, lets check it got applied. Please paste results:
```
lsmod
dmesg | grep wlan
cat /etc/modules
```

Hax4us, 2018-07-11:
```
root@DietPi:~# lsmod
Module                  Size            Used by
8723bs.              1232896.       0

root@DietPi:~# dmesg | grep wlan
usbcore: registered new interface driver rndis_wlan
[WLAN_RFKILL]:   Enter rfkill_wlan_init
[WLAN_RFKILL]:   Enter rfkill_wlan_prob
[WLAN_RFKILL]:   wlan_platdata_parse_dt:  wifi_chip_type=ap6212
[WLAN_RFKILL]:   wlan_platdata_parse_dt:  enable wifi power control
[WLAN_RFKILL]:   wlan_platdata_parse_dt:  wifi power controlled by gpio
of_get_named_gpiod_flags:  can't parse 'WIFI,poweren_gpio'  property of node '/wireless-wlan[0]'
of_get_named_gpiod_flags:  can't parse 'WIFI,reset_gpio'  property of node '/wireless-wlan[0]'
of_get_named_gpiod_flags: parsed 'WIFI,host_wake_irq'  property of node '/wireless-wlan[0]'. -  status (0)
[WLAN_RFKILL]:  wlan_platdata_parse_dt:  get property:  WIFI,host_wake_irq=150, flags=0.
[WLAN_RFKILL]: rfkill_wlan_probe: init gpio
[WLAN_RFKILL]: Exit rfkill_wlan_probe
RTL871X: rtw_ndev_init(wlan0) if1 mac _add=f0:03:8c:2d:aa:6d
IPv6:  ADDRCONF(NETDEV_UP): wlan0:  link is not ready
RTL871X:  rtw_set_802_11_connect(wlan0)  few_state=0x00000008
RTL871X: 
 rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
IPv6: ADDRCONF(NETDEV_CHANGE):  wlan0: link becomes ready 
RTL871X:  OnDeAuth(wlan0) reason=3,  ta=9e:df:b1:00:81:a7, ignore=0
RTL871X:  OnDeAuth(wlan0) reason=3,  ta=9e:df:b1:00:81:a7, ignore=0

I am posting only uncommented line 
root@DietPi:~# cat /etc/modules
8723bs
```

Hax4us, 2018-07-15:
Any progress here ?

MichaIng, 2018-07-17:
@Fourdee 
Just to not forget, here seems to be still something done. Hmm `G_CONFIG_INJECT '^8723bs' '8723bs' /etc/modules` would just change something, if the current line is `8723bs<and_something_else_here>`, is this the case/intention?

Fourdee, 2018-07-17:
@MichaIng 

> would just change something, if the current line is 8723bs<and_something_else_here>, is this the case/intention?

On the ASUS TB, the modules file is blank, so we just need to add this by default:
```
root@DietPi:~# cat /etc/modules
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

8723bs
```

Then our blacklist overrides it, when WiFi is disabled in DietPi:
```
root@DietPi:~# cat /etc/modprobe.d/disable_wifi.conf
blacklist bcmdhd
blacklist 8723bs
blacklist ap6211
blacklist brcmutil
blacklist brcmfmac
blacklist cfg80211
```

@Hax4us 
Apologies for the delay, is this still an issue you are experiencing?

Fourdee, 2018-07-17:
🈯️ Fresh install + WiFi test after reboot:
```
root@DietPi:~# lsmod
Module                  Size  Used by
8723bs               1232896  0

root@DietPi:~# dmesg | grep wlan
[    2.176142] usbcore: registered new interface driver rndis_wlan
[    5.762855] [WLAN_RFKILL]: Enter rfkill_wlan_init
[    5.763466] [WLAN_RFKILL]: Enter rfkill_wlan_probe
[    5.763554] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi_chip_type = ap6212
[    5.763581] [WLAN_RFKILL]: wlan_platdata_parse_dt: enable wifi power control.
[    5.763606] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi power controled by gpio.
[    5.763638] of_get_named_gpiod_flags: can't parse 'WIFI,poweren_gpio' property of node '/wireless-wlan[0]'
[    5.763669] of_get_named_gpiod_flags: can't parse 'WIFI,reset_gpio' property of node '/wireless-wlan[0]'
[    5.763759] of_get_named_gpiod_flags: parsed 'WIFI,host_wake_irq' property of node '/wireless-wlan[0]' - status (0)
[    5.763788] [WLAN_RFKILL]: wlan_platdata_parse_dt: get property: WIFI,host_wake_irq = 150, flags = 0.
[    5.763811] [WLAN_RFKILL]: rfkill_wlan_probe: init gpio
[    5.763839] [WLAN_RFKILL]: Exit rfkill_wlan_probe
[    7.467751] RTL871X: rtw_ndev_init(wlan0) if1 mac_addr=80:c5:f2:46:e1:81
[    9.076056] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   10.645049] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[   10.747431] RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
[   10.747520] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

root@DietPi:~# cat /etc/modules
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

8723bs

root@DietPi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1
    link/sit 0.0.0.0 brd 0.0.0.0
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 88:d7:f6:c2:f0:42 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.22/24 brd 192.168.0.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::8ad7:f6ff:fec2:f042/64 scope link
       valid_lft forever preferred_lft forever
4: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 80:c5:f2:46:e1:81 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.23/24 brd 192.168.0.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::82c5:f2ff:fe46:e181/64 scope link
       valid_lft forever preferred_lft forever
```

Fourdee, 2018-07-17:
Notes:
- Connects fine, but this appears to be the issue for @Hax4us  ```OnDeAuth```
```
IPv6: ADDRCONF(NETDEV_CHANGE):  wlan0: link becomes ready 
RTL871X:  OnDeAuth(wlan0) reason=3,  ta=9e:df:b1:00:81:a7, ignore=0
RTL871X:  OnDeAuth(wlan0) reason=3,  ta=9e:df:b1:00:81:a7, ign
```

> https://bbs.archlinux.org/viewtopic.php?id=164279

> https://superuser.com/questions/974017/ubuntu-wlan0-authenticates-and-then-drops

@Hax4us 
Try setting the following, then reboot system.
```G_CONFIG_INJECT 'REGDOMAIN=' 'REGDOMAIN=GB' /etc/default/crda```

Hax4us, 2018-07-18:
@Fourdee still not working , maybe it will work after fresh installation ? Where is latest image ?

Hax4us, 2018-07-18:
Still old `SET failed blah blah.` error

Hax4us, 2018-07-21:
@fourdee I can confirm that same problem is exist in my TB DietPi 6.12

Hax4us, 2018-07-21:
It worked only first time after flashing new image and I did some things like installing some tools and LXDE ,and in each reboot WiFi was working but after shutdown same problem ......😑😑

Hax4us, 2018-07-21:
What the hell is going on in my TB now after reboot it works but idk why and how,  well I will report back in case of same problem ..

Hax4us, 2018-07-21:
Again same problem. It is very unstable for me :(

Fourdee, 2018-07-23:
@Hax4us 

> It worked only first time after flashing new image and I did some things like installing some tools and LXDE ,and in each reboot WiFi was working but after shutdown same problem ......😑😑

Apologies for the delay. Thanks for confirming issue still exists. 

Apologies, however, i'am out of ideas as to why this is occurring on your device. I am unable to replicate the issue here with TB DietPi 6.12, and, everything appears in order on your system (excluding https://github.com/Fourdee/DietPi/issues/1760#issuecomment-404186197).

I'am unsure how to debug this further (https://github.com/Fourdee/DietPi/issues/1760#issuecomment-404186197), and, it may be some time before either myself, or comminuty support can find a solution.

The only thing I can suggest, is to try the official TinkerOS image (if not done already), and, just verify if the loss of WiFi on reboot issue exists with that (eg: could be a hardware issue?).

Hax4us, 2018-07-27:
@fourdee thnx for your all work and now I have armbian in my tinker and there is no WiFi issue

Fourdee, 2018-07-28:
@Hax4us 

> can abnormal (sudden) shutdown be a culprit ?

Is possible. A hard power cycle will most likely cause filesystem corruptions, which could affect various areas of the systems software (and therefore hardware).

So just to confirm:
- Download https://dietpi.com/downloads/images/DietPi_ASUSTB-ARMv7-Stretch.7z
- Install it
- Allow it to update successfully to v6.12
- Make no changes to any software etc
- use ```dietpi-config``` to setup WiFi
- reboot
- WiFi issues appear?

I've done the same above test multiple times and i'am unable to replicate on my local devices. The ```/etc/modules``` fix was successful at least my end.

Fourdee, 2018-08-13:
No response, unable to replicate, marking as closed.

closed 2018-08-13 by Fourdee