MichaIng/DietPi #1762: Sickrage does not start refuses the connection

lucario1001, 2018-05-09:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```6.7```
- SBC Device  RPi3
- Power supply used 2.1amp power brick
- SD card used Sandisk ultra
- Distro (EG: Jessie) | ```Linux DietPi 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux```?

#### Additional Information (if applicable):
- Software title Sickrage
- Can this issue be replicated on a fresh installation of DietPi: Yes

#### Expected behaviour:

Show gui when you goto the link of your pi
#### Actual behaviour:
Does not startup gui

#### Steps to reproduce:
Install sickrage
profit!?!?
#### Did you submit a ```Bug report```?

9e598df7-d17c-48e9-a59d-733b0c8ca780-0
#### Extra details:

Fourdee, 2018-05-10:
@lucario1001 

Thanks for the report 👍 

We'll try to replicate in local testing.

Fourdee, 2018-05-10:
Unable to replicate on fresh install. Will check the bug report.

Fourdee, 2018-05-10:
Process exists, but service is in fail state. Process hung?
```
#LOG > 2018-05-09 23:43:51 INFO     MAIN :: Signal 15 caught, saving and exiting...
#SRV > sickrage	deactivating (final-sigterm) (Result: timeout)
```

@lucario1001 
Please try killing the process:
```
dietpi-services stop
killall -s SIGKILL -w python
```

If the command finishes successfully, please try reinstalling SickRage:
```
dietpi-software uninstall 116
dietpi-software install 116
```

lucario1001, 2018-05-10:
sickrage fails i just made a new install did the commands you said here is another reference code ```32b5125b-7661-4fbd-9d23-c7fedd83e949-0```

Fourdee, 2018-05-11:
@lucario1001 

Thanks.

We'll need to debug the binary, please run the following commands:
```
killall -s SIGKILL -w python
python /etc/sickrage/SickBeard.py
```

Check the onscreen print and paste it here please.

lucario1001, 2018-05-11:
```Unable to find /etc/sickrage/config.ini, all settings will be default!
17:54:45 INFO::MAIN :: Backing up database before upgrade
17:54:45 INFO::MAIN :: Proceeding with upgrade
17:54:45 INFO::MAIN :: Add minor version numbers to database
17:54:45 INFO::MAIN :: Updated to: 43.1
17:54:45 INFO::MAIN :: Backing up database before upgrade
17:54:45 INFO::MAIN :: Proceeding with upgrade
17:54:45 INFO::MAIN :: Starting SickRage [] using '/etc/sickrage/config.ini'
17:54:45 INFO::MAIN :: Generating New API key
17:54:45 INFO::TORNADO :: Starting SickRage on http://0.0.0.0:8081/
17:54:45 INFO::CHECKVERSION :: Checking for updates using SOURCE
17:54:45 INFO::MAIN :: Checking for scene exception updates from http://sickrage.github.io/scene_exceptions/scene_exceptions.json
17:54:46 INFO::MAIN :: Checking for XEM scene exception updates for theTVDB
17:54:46 INFO::MAIN :: Checking for scene exception updates for AniDB
```
now it works o-0

lucario1001, 2018-05-11:
never mind it wont launch now on reboot

Fourdee, 2018-05-18:
If I remember correctly, the recommended service for sickrage has failed restart enabled by default. I'll see about adding this back in to our service, may provide a "solution".

Fourdee, 2018-05-20:
Hmm, I believe this is just be a simple case of SystemD timing out the service before Sickrage has had enough time to init.

I'll update the service for v6.9 to never timeout, lets see if that resolves the issue you reported.

Fourdee, 2018-05-20:
Patch applied to v6.9, service will update automatically. I'll mark this as preliminary resolved for v6.9 (ETA: 1 week release).

@lucario1001 

If you want to apply this fix now, copy and paste all into SSH:
```
				cat << _EOF_ > /etc/systemd/system/sickrage.service
[Unit]
Description=SickRage
After=dietpi-service.service

[Service]
User=root
Group=root
Type=forking
GuessMainPID=no
TimeoutSec=infinity
RestartSec=20
TimeoutStopSec=20
Restart=always
ExecStart=/usr/bin/python /etc/sickrage/SickBeard.py -q --daemon --nolaunch --datadir=$G_FP_DIETPI_USERDATA/sickrage

[Install]
WantedBy=multi-user.target
_EOF_

systemctl daemon-reload
sync
reboot
```

closed 2018-05-20 by Fourdee