MichaIng/DietPi #1764: Fresh installation with subsonic on RPi 3 results in 503 error

SevronCarl, 2018-05-11:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
"6
7"
using DietPi_v6.6_RPi-ARMv6-Stretch.img image
- SBC Device (EG: RPi 3)?
RaspberryPi 3 B+
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
dane-elec, 16gb
- Distro (EG: Jessie) | ```uname -a```?
Linux DietPi 4.14.34-v7+

#### Additional Information (if applicable):
- Software title?
Subsonic
- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Expected behaviour:

It should work after first install
#### Actual behaviour:

[image]

#### Steps to reproduce:

Followed guide here https://dietpi.com/phpbb/viewtopic.php?f=8&t=9#p9
Address set to 192.168.1.200
Used DietPi-Software to install subsonic
Waited until finished.
Attempted to visit 192.168.1.200:4040 for subsonic
503 Error.

#### Did you submit a ```dietpi-bugreport```?

f2fe16e9-e1bf-41db-a744-662b763712e7-0

#### Extra details:

subsonic_sh.log;
[image]

Fourdee, 2018-05-12:
@SevronCarl 

Many thanks for the report 👍 

We'll try to replicate and resolve.

Fourdee, 2018-05-12:
@SevronCarl 

This is a known issue with jetty, please see the online doc for the fix:
https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=30

---

I'll see if we can apply this to install by default.

SevronCarl, 2018-05-12:
And now I've restarted it and can't connect to subsonic at all, think I'm gonna put it to bed for a while...
Try something else.

Fourdee, 2018-05-12:
Confirmed bug with all systems.

Java issue?
Online doc states Java7: http://www.subsonic.org/pages/installation.jsp#debian
Try with Oracle JRE?

Fourdee, 2018-05-12:
Interesting AirSonic is also failing with the same Java errors, looks like a Java specific issue, recent Java update breaking these programs?

MichaIng, 2018-05-12:
@Fourdee
`Requirements: Java 7 or later`
Should work with java 8, since java 10 is already released...

Will also do some tests here. I would prefer to stay with OpenJava, if possible...

MichaIng, 2018-05-12:
Got it running after manually installing the Buster version of `openjdk-8-jre-headless` over the Stretch version, even that both share nearly the same version numbers: https://packages.debian.org/de/buster/openjdk-8-jre-headless

On my Stretch VM, after installing AirSonic via dietpi-software, including OpenJDK 8 etc:
```
systemctl stop airsonic
wget http://ftp.de.debian.org/debian/pool/main/o/openjdk-8/openjdk-8-jre-headless_8u171-b11-1_amd64.deb
dpkg -i openjdk-8-jre-headless_8u171-b11-1_amd64.deb
systemctl start airsonic
```
Wait for a while until everything is leaded, takes a while, can be watched via `htop` Java initiates `airsonic.war` with 100% CPU usage, filling RAM up to around 400M, then CPU usage goes down. Only after this, `http://<localIP>:8080/airsonic` can be accessed... Puuh Java is crazy overhead, vote against using it for smaller otherwise independent software... 400M RAM, are you sure?? Or is this not normal and might be still case of an issue here?

Fourdee, 2018-05-13:
@MichaIng 

Great work and research 👍 

I'll move this to v6.9, will focus on final testing for v6.8 and release today.

Fourdee, 2018-05-21:
Re-tests:
SubSonic:
- RPi 3 🈯️ 
- VM 🈯️ 

AirSonic:
- RPi 3 🈯️ took 3 minutes before web interface was accessable (long init/startup time)
- VM 🈯️

Fourdee, 2018-05-21:
Completed. Patched reinstall for v6.9.

closed 2018-05-21 by Fourdee