MichaIng/DietPi #1765: DietPi-Drive_Manager | Why is it not supported on VM?

MichaIng, 2018-05-12:
Why is DietPi-Drive_Manager not supported on VM? Works great with all it's features, at least on my testing VMs. Was there some risk or issue observed?

Fourdee, 2018-05-12:
@MichaIng 

> Was there some risk or issue observed?

It was not compatible pre-v6.x, however, with all the Init and detection improvements to ```dietpi-drive_manager``` it should be compatible now.

If you can verify testing, we can enable. My PC has been delayed until next week ☹️

MichaIng, 2018-05-12:
@Fourdee
As all my testing and development of the fsck implementation was done on VM, including many times formatting and using the USB stick with all file system types, I think it can be assumed compatible.

But just to be sure, I will redo some tests with virtual drives instead of real ones, add to VM via USB.

closed 2018-05-12 by MichaIng

Fourdee, 2018-05-13:
@MichaIng 

Legend, many thanks 👍 

> nice if we could implement partition table and initial file system creation as well. (=> 6.9)

Yep, I've been meaning to do this for a long time 👍, required feature I believe.