MichaIng/DietPi #1766: FeautreRequest: /var/log to SD/USB-Drive

WTFMaster, 2018-05-12:
I would really apprechiate the feature, to shift the /var/log to USB/SD-Drive. Like already implemented for the UserData.

MichaIng, 2018-05-12:
@WTFMaster 
Thanks for your request.

Great idea, was already thinking the same by times. Combines benefits from persistent logging (available crash logs) and RAMlog (reduced SD writes) and gives even more flexibility (read logs from external system, in case of broken boot).

Generally would be possible via symlink or mountpoint. We just need to assure that all disk logging is started after ALL drives are mounted. But I think this should be already the case, as syslog/rsyslog/journalctl services start after fstab is handled, or at least write the early boot logs (before even rootfs is mounted) to disk afterwards? Needs testing.

Is quite larger tasks all together I think, would be good then to combine this with DietPi-RAMlog or merge it into a new system where RAM as logging location is just one option.

FlybyKnights, 2018-05-23:
Simple workaround:

Grab the code from [here](https://unix.stackexchange.com/questions/228015/move-a-file-and-replace-it-by-a-symlink) (the one with the 'while' statement) and add it to` nano /bin/mvln` then make it executable `chmod +x  /bin/mvln`

That gives you the ability to move files and leave behind symbolic links using `mvln original_path target_path`

Now its just a matter of running `mvln /var/log/what_ever_file_you_want /usb_disk_path/.configs`

I've used it to move my configurations to the USB Drive. If my SD corrupts / blow up, i jsut re-install all the apps. Delete their folders and re-symlink them. No issues

Fourdee, 2018-07-09:
I see no reason for this currently, as the majority of our users have RAMlog installed:
https://dietpi.com/survey/

I'll mark as closed, however, please reopen if required.

closed 2018-07-09 by Fourdee