MichaIng/DietPi #1767: Mask systemd-logind?

MichaIng, 2018-05-12:
@Fourdee 
Just recognized again, when `dbus` is installed, because of desktop or `systemd-timesyncd` usage, the static service `systemd-logind` is starting up again, allocating/reserving 6 gettys by default.
- We suppressed this in the past via
```
G_CONFIG_INJECT 'ReserveVT=' 'ReserveVT=0' /etc/systemd/logind.conf '\[Login\]'
G_CONFIG_INJECT 'NAutoVTs=' 'NAutoVTs=0' /etc/systemd/logind.conf '\[Login\]'
```
- This was replaced by `systemctl mask getty-static` after `dbus` (thus logind) removal. But when installing `dbus` according to above reasons, `logind` starts, enabling the additional gettys by itself, while `getty-static` is skipped.
- Btw. the presence/allocation of additional gettys can be tested by switching to e.g. tty2 via `<Alt>+<F2>` and see if login prompt appears there or not. After switching, the getty services do also show up within `htop`. When settings above are applied, no login prompt appears at `<Alt>+<F2>` and no getty service is started.
- We could just apply the settings above to ensure just tty1 is present by default, regardless of logind is starting or not. But I am thinking if it isn't better then to just mask `systemd-logind` itself. As already discussed before, without `libpam-systemd` (`pam_systemd.so`), besides dynamic getty creation, other logind feature are unavailable anyway: https://www.freedesktop.org/software/systemd/man/systemd-logind.service.html

So the question if these features are useful in some situations or not. Looks to me that this is the case if several (real) users use one machine in parallel via hot seat or different physical interfaces. SSH connections do not create/need additional gettys as already should be clear 😉. Not sure if, when using a desktop, switching users while all of them stay logged in, need additional gettys then?
`loginctl` btw. is the tool to check, of user sessions are handled via logind or not. When `libpam-systemd` is present, `loginctl` contains one or several entries (when switching to other gettys), without the package, no entries are present, thus logind seems useless besides automated getty allocation.

We could also add a dietpi.txt setting for single/multi user machine and mask logind + purge `libpam-systemd` or unmask logind + install `libpam-systemd` conditionally, at least in combination with an installed desktop.

Fourdee, 2018-05-30:
@MichaIng 

> mask systemd-logind
+
G_CONFIG_INJECT 'ReserveVT=' 'ReserveVT=0' /etc/systemd/logind.conf '\[Login\]'
G_CONFIG_INJECT 'NAutoVTs=' 'NAutoVTs=0' /etc/systemd/logind.conf '\[Login\]'

> We could also add a dietpi.txt setting for single/multi user machine and mask logind + purge libpam-systemd or unmask logind + install libpam-systemd conditionally, at least in combination with an installed desktop.

Yep good idea 👍. Just to clarify:
- mask ```systemd-logind``` and remove ```libpam-systemd```, if no desktop is installed?
- add options to ```dietpi.txt``` to control ```systemd-logind``` masking (+ install ```libpam-systemd``` when enabled)?
- leave ```getty-static``` as is, already controlled by serial enable etc?

MichaIng, 2018-05-31:
@Fourdee 
I am not sure in which environments several TTYs and logind features are useful/needed but jep, I would by default mask `systemd-logind` and leave `libpam-systemd` uninstalled as it is now, independently from the use of desktop or not. That we never saw an issue/bug report about missing logind features shows how rarely it is used.

But yeah `dietpi.txt` makes sense for power users and those who know, that they need it.
Then we should leave default `/etc/systemd/logind.conf` (as it is now), to have automated TTY spawning enabled together with `systemd-logind` directly. `dbus` also needs to be installed as pre-req, just to not forget. I think for now it's enough to implement this for first-run setup only. By times it would be great to have all `dietpi.txt` options available via `dietpi-config` as well, but that's another larger task 😉.

`getty-static` is currently just masked and not controlled after PREP, AFAIK. But that also makes sense. If one needs additional gettys by times, logind seems the modern dynamic/automated spawning solution. `getty-static` I guess is more like an ancient fallback and should stay masked.

Only question if/how we patch:
- Masking logind for all, as this is our default?
- At least checking for `libpam-systemd`, as this would have been installed afterwards and enables logind features, thus user or other software seem to require them?
- Keep logind as well for desktop users?
- Not touch it at all on existing systems to stay safe?

Fourdee, 2018-06-01:
@MichaIng 

> Masking logind for all, as this is our default?

Yep, lets make this the new default, unless  ```libpam-systemd``` is installed (EG: desktop)

> At least checking for libpam-systemd, as this would have been installed afterwards and enables logind features, thus user or other software seem to require them?

Yep, I'd go with that, check for ```libpam-systemd```, disable ```systemd-logind``` if not installed. Could probably run this at the end of dietpi-software installs, to auto enable if ```libpam-systemd``` is installed?

Fourdee, 2018-06-05:
I'll nudge this to v6.10, vast number of core changes in v6.9 already 😃

MichaIng, 2018-07-29:
PR is up: https://github.com/Fourdee/DietPi/pull/1976

Fourdee, 2018-08-06:
Ok, some testing on native PC with LXDE:
- dietpi-update, killed X server once ```systemd-logind.service ``` was stopped and masked
- PID error returned when loading LXDE

We need to unmask and allow ```systemd-logind.service ``` when desktops are installed.

MichaIng, 2018-08-06:
@Fourdee 
Hmm, `libpam-systemd` was not installed on the system with lxde? Thought there was some dependency. Or did the check not work as expected?

Fourdee, 2018-08-06:
@MichaIng 

```libpam-systemd``` is installed

```
root@DietPi:~# dpkg-query -s 'libpam-systemd' &> /dev/null; echo $?
0
root@DietPi:~# dpkg-query -s 'libpam-systemd'
Package: libpam-systemd
Status: install ok installed
Priority: standard
Section: admin
Installed-Size: 369
```

MichaIng, 2018-08-06:
@Fourdee 
Whoopsie, my fault, forgot `!` within check. `0 => yes`, `1 => no` and within arithmetic the other way round, I guess I mixed it up 🤣.

Fixed: https://github.com/Fourdee/DietPi/commit/a4bc048a8bd65787d8df7fc4d186e973fb38f23c

Fourdee, 2018-08-06:
@MichaIng 

> nd within arithmetic the other way round, I guess I mixed it up 🤣.

I've done the same thing a few times 👍 😄

MichaIng, 2018-08-06:
@Fourdee 
I rechecked, if libpam-systemd is installed with all desktops: `policykit-1` depends for it, but we do not install it xfce4: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L3412-L3474

I am not sure, if systemd-logind is still required for xfce then? At least "libpam-systemd" got never installed, thus no logind feature available. But not sure how/why xserver still needs it to be present?
Better retest with XFCE.

closed 2018-08-13 by MichaIng