MichaIng/DietPi #1768: DietPi-Software list

Fourdee, 2018-05-13:
> https://dietpi.com/dietpi-software.html

Think it would look great, started this last year and never got a chance to finish it.

MichaIng, 2018-05-13:
@Fourdee 
Yey would be indeed beautiful and allow interested ones a closer look without going one step further into the forum. But quite some work as well 😅.

Fourdee, 2018-05-13:
Notes:
- software_0.jpg = mini box (640x400)
- software_1.jpg = Large box (640x400, would be nice bigger?)

Fourdee, 2018-05-14:
Made a start: https://dietpi.com/dietpi-software.html
Really want this available to end users.

Fourdee, 2018-05-14:
mmm, Might simplify this a little, instead of having the large box with info, simply link to forum info when box selected? Less duplicate info + maintenance.

Fourdee, 2018-05-19:
Ok, in regards to having all the usage info on the new page. Made the decision this will be based on how much info there is per item.

EG: All desktop info will be on the new site
EG: VNC info will still link to forum, as we cannot cleanly code sections, and, its too long.

To achieve full move from forum to the new page, i'll either need to spend weeks improving my HTML/CSS knowledge, or, someone is willing to help us improve the HTML/CSS so we can make this possible.

Fourdee, 2018-05-26:
Note to setf: Downloads completed: https://dietpi.com/dietpi-software.html

DJCrashdummy, 2018-05-30:
as this site is/should be dedicated to end users beside the (already included :ballot_box_with_check: ) hint that some software may be disabled due to their hardware, IMHO it should be mentioned at the introduction, that this is only the little selection of highly optimized or/and pre-configured software but **still numerous other software can be installed** via apt, aptitude, synaptic, etc. like usual under debian (because dietpi is based on it).

btw: mentioning the base - at least as i did now - may also be reasonable there, since it is nowhere else mentioned at the homepage. it seems to me like after the transition from wheezy to jessie all easily recognizable references to debian vanished from the homepage.

MichaIng, 2018-05-30:
@DJCrashdummy 
This is an important point, thanks for mentioning. The main advantage of DietPi, being based on Debian, is the flexibility of using APT and being able to use Debian (respectively Raspbian) related guides, as well as Ubuntu guides in most cases. This should be outlined at the places where DietPi-Software is promoted.

We could go further and add some information on a dedicated page, about where DietPi behaves different than standard Debian/Raspbian, what you have to take care of when adjusting the system, e.g.
- Software services being controlled by DietPi-Services, instead of SystemD => systemctl disabled means DietPi controlled
- Using DietPi-Software and DietPi-Config where possible, as manual installations/adjustments might break the behaviour of similar DietPi settings and installations.
- By default disabled kernel modules and removed packages, that you would expect on default Debian, as we provide a vastly reduced image in terms of RAM and disk usage:
   - no NTP since v6.9 => moved to systemd-timesyncd, but without systemd-time**date**d (`timedatectl`)
   - no dbus, as long as not desktop (or time sync daemon mode) is selected
   - ... and things like that.

@Fourdee 
I think for such things, also to make user contribution easier and sort github issues, a Wiki would be useful? Would you mind to add one to DietPi repo, I didn't find a way to do this myself? We could move DietPi-PREP information, image creation procedure and things like that as well, so they will not be forgotten at the last page of open issues 😛.

Fourdee, 2018-05-30:
@MichaIng 

>  if one has some spare time

I've written 'spare time' on my xmas list to Santa 😉, although, i'll probably end up with Coal again 🥂 

I'll enable the WiKi section now, but may be some time before I can even begin to make a start on it.

@DJCrashdummy 

Thanks for your feedback, we'll take it on-board 👍

Fourdee, 2018-06-01:
Note to self: NextCloud completed: https://dietpi.com/dietpi-software.html

DJCrashdummy, 2018-06-04:
@MichaIng exactly! mentioning the base is surly useful for every user (beside pointing out that there is more than the listed software for complete newbies), but the further information may just be interesting for more techy users and should only be linked.
good idea with the wiki! so the information can be collected collaborative and extended any time, like a comparison to other distributions...
yes, in longterm this is also a better place for user-guides, how-tos, etc. :+1: 

---
> I've written 'spare time' on my xmas list to Santa :wink:, although, i'll probably end up with Coal again :clinking_glasses:

same here... :disappointed: every year!  but @Fourdee made my day! :stuck_out_tongue_winking_eye:

Fourdee, 2018-06-15:
ref: https://github.com/Fourdee/DietPi/issues/1845#issuecomment-397447909

Fourdee, 2018-07-16:
Note:
- From Pydio onwards:
-- _0.jpg
-- URL link + header name

Fourdee, 2018-07-23:
To Do:
- Go through all items, add/check pros/cons + info + pictures
- Add pending additions https://github.com/Fourdee/DietPi/issues/1768#issuecomment-397526254

Fourdee, 2018-08-01:
I've set this live. Its around 85% completed:
https://dietpi.com/dietpi-software.html

However, all software logos are now in, and links to their full info.

I'll finish off the rest, as/when I can.

MichaIng, 2020-11-11:
This is now handled on the dedicated website repository: https://github.com/MichaIng/DietPi-Website
Marking as closed here.

closed 2020-11-11 by MichaIng